pyvisa/pyvisa #523: [COM] Communication issue with Tektronix MDO3014 using scope.query('WFMOutpre:XZEro?')

samtalki, 2020-07-14:
Instrument details
------------------
* Model: Tektronix MDO3014
* Communication: TCPIP, GPIB, .... : TCPIP
* Link to the documentation (if available): 
[MDO4000-B-MSO-DPO4000B-and-MDO3000-Oscilloscope-Programmer-Manual-Rev-A.pdf](https://github.com/pyvisa/pyvisa/files/4919729/MDO4000-B-MSO-DPO4000B-and-MDO3000-Oscilloscope-Programmer-Manual-Rev-A.pdf)

Output of `pyvisa-info` 
-----------------------

Does not give any output in Anaconda Prompt or Spyder command prompt. This command has never worked since I first installed pyvisa.

Hello,

I am experiencing a communication issue with the Tektronix MDO3014 that is not described elsewhere in previous issues online.

The following errors are thrown when attempting to run scope.query('WFMOutpre:XZEro?') which is a supported command for this oscilloscope described in the documentation attached.

As required by the scope programmers manual, the data source is properly set as demonstrated via this command:
```

In[38]: scope.query('DATa:SOURCE?')
Out[38]: 'CH1\n'

```

When running scope.query('WFMOutpre:XZEro?')  the following error is yielded:

```

In: float(scope.query('WFMOutpre:XZEro?'))
Traceback (most recent call last):

  File "<ipython-input-34-bf6ebef1a925>", line 1, in <module>
    float(scope.query('WFMOutpre:XZEro?'))

  File "C:\Users\sat01\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 613, in query
    return self.read()

  File "C:\Users\sat01\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 427, in read
    message = self._read_raw().decode(enco)

  File "C:\Users\sat01\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 400, in _read_raw
    chunk, status = self.visalib.read(self.session, size)

  File "C:\Users\sat01\Anaconda3\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1584, in read
    ret = library.viRead(session, buffer, count, byref(return_count))

  File "C:\Users\sat01\Anaconda3\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 193, in _return_handler
    raise errors.VisaIOError(ret_value)

VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

```

The source code that I used to setup the scope object is as follows. This runs with no issues. Importantly, the '*IDN' query yields expected reply.

```

import pyvisa

from tkinter import Tk
import numpy as np
from struct import unpack
from tkinter import filedialog
import pylab
    
rm = pyvisa.ResourceManager('@ni')

#Choose the IP address of Scope to connect

scope1 = '192.168.3.155'
#Check that TCPIP0 is not actually TCPIP
scope = rm.get_instrument('TCPIP0::'+scope1)
print(scope.query("*IDN?"))

```

I'm at a loss here. I've followed the documentation for pyvisa and my instrument explicitly. Things aren't working as described.

MatthieuDartiailh, 2020-07-15:
Reading the documentation, you are supposed to set both the source and the encoding for the data transfer. Furthermore in most example given it appears that DATA:START and DATA:STOP are set before calling any of the WFMO commands. I would suggest you try to follow the example from page 2-100 and 2-101 and see if you can get thing to work by using commands in order.

samtalki, 2020-07-16:
Hi Matthieu, 

I have followed those commands in order. Unfortunately it seems that the same issue I described in my original post prevents most of those commands from working. For instance, if I attempt to set the x increment with scope.query('WFMOutpre:XINcr? 1e-6') and then attempt to check whether this command was successful with scope.query('WFMOutpre:XINcr?'), the output is the default "4e-9" despite the first command being executed without error.

Essentially, the library doesn't seem to be executing basic commands on the scope despite the fact that a proper connection is established, the proper data source is set, and the example order spelled out by 2-100 is followed. Additionally, I have confirmed that the scope itself is set to properly communicate over TCPIP.

Sorry for the long winded response. I am sorry for taking your time that I'm sure is very valuable. Let me know if I can provide anymore information to help resolve this issue.

Many thanks
Sam

MatthieuDartiailh, 2020-07-17:
Without seeing your whole script/notebook/session it is a bit hard to figure out all the issues, if you could share your whole code I may be able to help more. 
I have a comment though.

When setting a value the instrument will in general not answer and the command string should not contain the ?. For example to set the X increment you would do:
```python
scope.query('WFMOutpre:XINcr 1e-6')
scope.query('WFMOutpre:XINcr?')
```
Similarly if you want to lock the instrument do:
```python
scope.write("LOCk ALL")
scope.query('LOCk?')  # if you want to check the result
scope.write("UNLock")
scope.query('LOCk?')  # if you want to check the result
```
Hopefully this will help.

IntegralWorks, 2020-12-27:
@samtalki can you please try running this code: https://github.com/IntegralWorks/work_automation/blob/master/VISA_AUTOMATION/setLabelsAndScale.py and show me your output; sometimes I've had issues with whatever "connection software" (NI-MAX or Tektronix VISA or whatever) I want to see if you have either "dead entries" or a lack of entries. Also make sure you are using TCIP.