pyvisa/pyvisa #524: [COM] Communication issue with RS NRP-Z81 and Z86 using

spongio, 2020-07-18:
Instrument details
------------------
* Model: NRP-Z81 (but I have the same problem with Z86)
* Communication: USB
* Link to the documentation (if available):

Output of `pyvisa-info`
-----------------------

rm.open_resource("RSNRP::0x0095::104003::INSTR")

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/touko/.local/lib/python3.6/site-packages/pyvisa/highlevel.py", line 1753, in open_resource
    resource_pyclass = self._resource_classes[(info.interface_type, info.resource_class)]
AttributeError: 'int' object has no attribute 'interface_type'

I tried with both backends but without success. I am using an Ubuntu 18.04 x86_64 machine

Machine Details:
   Platform ID:    Linux-5.3.0-53-generic-x86_64-with-Ubuntu-18.04-bionic
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.6.9
   Compiler:       GCC 8.4.0
   Bits:           64bit
   Build:          Apr 18 2020 01:56:04 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'

MatthieuDartiailh, 2020-07-19:
Your instrument requires an additional VISA component known as a passport since RSNRP is not a usual interface type (it is a specific protocol built on top of USB). According to the manual of the passport found on the R&S website (https://scdn.rohde-schwarz.com/ur/pws/dl_downloads/dl_software/nrp_ni_visa_passport/Rs_ViNrp_Software_Manual_2_7_3.pdf) it runs only with NI VISA. Furthermore I could only find installer for Windows. So I am not you will be able to achieve what you are trying to.
Since the protocol is proprietary there is no easy way to duplicate over libusb like we do for USBTMC devices.

If you can find workaround for those issues, you will need to register a resource class with the interface type RSNRP for PyVISA to work. This is very simple so if you figure out the above it won't be a problem.

spongio, 2020-07-19:
Perfect, I was afraid of this.

I was really hoping to be doing something wrong. I will post if I find a workaround

Thank you very much

MatthieuDartiailh, 2020-07-19:
Sorry I could not be more positive @spongio .

I will keep the issue open for a couple of weeks and then close since there is not much that can be done on PyVISA side. Do not hesitate to report any finding even once the PR is closed.