pyvisa/pyvisa #526: VI_WARN_UNKNOWN_STATUS (1073676421) and sending bytes to device

NykeMant, 2020-07-22:
Hello everyone, 
as i'm quite new to this, i don't know how to describe my problem clearly. At first i just want to establish a connection to my device, for this my code looks like this:
import visa
rm = visa.ResourceManager()
inst = rm.open_resource(rm.list_resources()[0])

and as a result my terminal delivers this output:
"Path"\AppData\Local\Programs\Python38-32\lib\site-packages\pyvisa\ctwrapper\functions.py:1213: VisaIOWarning: VI_WARN_UNKNOWN_STATUS (1073676421): The status code passed to the operation could not be interpreted.       
  ret = library.viOpen(session, resource_name, access_mode, open_timeout, byref(out_session))

When i have established a connection i just want to send bytes to my device, is there another command to write, as when i tried as a test to write something with this code:
import visa
rm = visa.ResourceManager()
inst = rm.open_resource(rm.list_resources()[0])
inst.write("test", "3a")

the output is as follows:
"Path"\AppData\Local\Programs\Python38-32\lib\site-packages\pyvisa\ctwrapper\functions.py:1213: VisaIOWarning: VI_WARN_UNKNOWN_STATUS (1073676421): The status code passed to the operation could not be interpreted.       
  ret = library.viOpen(session, resource_name, access_mode, open_timeout, byref(out_session))
Settings address not valid!

The output from python -m visa info:
[image]

Best regards!

MatthieuDartiailh, 2020-07-22:
I am not sure what is going on here. Could you post the output of:
```python
import pyvisa
rm = .pyvisa.ResourceManager()
print(rm.list_resources())
```
Sometimes serial ports which do not have an instrument attached are reported. It might cause such a weird issue.

Could you let me know what instrument do you want to communicate with ? 

Finally `write` takes a single string that you want to send to the instrument, here you are specifying to append `"3a"` as a termination character which likely not what you meant to do. And `"Test 3a"` seems like a fairly uncommon message, did you get from the instrument manual ?

NykeMant, 2020-07-23:
Hello Matthieu
here is the the output:
('USB0::0x0D7D::0x0100::SPAD-LINE-2003033::INSTR',)

In the write function i wanted to test if there is already an established connection and to see what is sent. 
Best regards.

MatthieuDartiailh, 2020-07-23:
Can you share the model of your instrument ? I would like to have a look at the manual. If it is an option, can you try installing a 64 bits version of Python. In the past some users have reported issues on 32 bit windows (while other have no issue) and we never managed to get to the bottom of it.

Also if you need to send exact byte values, you will have to use `write_raw`, not `write`.

NykeMant, 2020-07-24:
Hello, 
concerning the model of my device, i myself am not sure about it and my colleagues told me, that the usb interface uses xilinx communication interface, i hope this information is useful.
As for changing to a 64-bit python version, i did this and tried to communicate with my device and i got some output which are inside my expectations. So thanks for this advice, but the warning is still present.
Best regards.

MatthieuDartiailh, 2020-07-24:
Once again not sure what is going on here. VISA is meant to be used with instrument supporting that specification. It is possible that your device use the USBTMC protocol (or some close variant) which allows you to send bytes to it but provides answer that do not lie within the VISA specification hence the warning. Without more details about your harware I don't think I can provide any additional feedback.

I will close for now, feel free to reopen if anything new comes up.

closed 2020-07-24 by MatthieuDartiailh