pyvisa/pyvisa #528: ValueError: could not convert string to float. Tektronix TDS714L

tzechong94, 2020-07-28:
Hello all, I am attempting to use Pyvisa to get readings voltage readings from an oscilloscope (Tektronix TDS714L) My ultimate goal is to collect the average voltage reading at each position when my stages move. (microscope related project) 

`import pyvisa
import numpy as np
pyvisa.log_to_screen()

rm = pyvisa.ResourceManager('C:\Windows\System32\\visa64.dll')
inst = rm.open_resource("GPIB::1::INSTR")
values = inst.query_ascii_values('CURV?')
`

I used the above code to figure out what data I'm reading from the oscilloscope, and there is what I get. 

`2020-07-28 10:21:44,477 - pyvisa - DEBUG - Library signatures: 90 ok, 0 failed
2020-07-28 10:21:44,477 - pyvisa - DEBUG - Created library wrapper for C:\Windows\System32\visa64.dll
2020-07-28 10:21:44,515 - pyvisa - DEBUG - viOpenDefaultRM('<ViObject object at 0x000001BF2AB3EA48>',) -> 0
2020-07-28 10:21:44,515 - pyvisa - DEBUG - Created ResourceManager with session 4097
2020-07-28 10:21:44,516 - pyvisa - DEBUG - viParseRsrcEx(4097, 'GPIB::1::INSTR', 'c_ushort(1)', 'c_ushort(0)', <ctypes.c_char_Array_256 object at 0x000001BF2AB3EBC8>, <ctypes.c_char_Array_256 object at 0x000001BF2AB3EC48>, <ctypes.c_char_Array_256 object at 0x000001BF2AB3ECC8>) -> 0
2020-07-28 10:21:44,516 - pyvisa - DEBUG - GPIB::1::INSTR - opening ...
2020-07-28 10:21:44,578 - pyvisa - DEBUG - viOpen(4097, 'GPIB::1::INSTR', <AccessModes.no_lock: 0>, 0, '<ViObject object at 0x000001BF2AB3EC48>') -> 0
2020-07-28 10:21:44,578 - pyvisa - DEBUG - GPIB::1::INSTR - is open with session 1
2020-07-28 10:21:44,581 - pyvisa - DEBUG - viWrite(1, b'CURV?\r\n', 7, 'c_ulong(7)') -> 0
2020-07-28 10:21:44,581 - pyvisa - DEBUG - GPIB::1::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2020-07-28 10:21:44,594 - pyvisa - DEBUG - viRead(1, <ctypes.c_char_Array_20480 object at 0x000001BF2AB3EC48>, 20480, 'c_ulong(1007)') -> 0
Traceback (most recent call last):
  File "C:\Users\NUSOSEC\Desktop\Oscilloscope\Oscilloscope.py", line 7, in <module>
    values = inst.query_ascii_values('CURV?')
  File "C:\Users\NUSOSEC\anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 672, in query_ascii_values
    return self.read_ascii_values(converter, separator, container)
  File "C:\Users\NUSOSEC\anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 462, in read_ascii_values
    return util.from_ascii_block(block, converter, separator, container)
  File "C:\Users\NUSOSEC\anaconda3\lib\site-packages\pyvisa\util.py", line 224, in from_ascii_block
    return container([converter(raw_value) for raw_value in data])
  File "C:\Users\NUSOSEC\anaconda3\lib\site-packages\pyvisa\util.py", line 224, in <listcomp>
    return container([converter(raw_value) for raw_value in data])
ValueError: could not convert string to float: '#41000\x01\x00\x02\x00\x00\x00\x01\x00\x02\x00\x01\x00\x02\x00\x01\x00\x01\x00\x02\x00\x00\x00\x02\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x00\x00\x01\x00\x01\x00\x02\x00\x00\x00\x02\x00\x01\x00\x02\x00\x00\x00\x02\x00\x02\x00\x01\x00\x01\x00\x02\x00\x01\x00\x01\x00\x00\x00\x01\x00\x01\x00\x00\x00\x01\x00\x01\x00\x02\x00\x01\x00\x02\x00\x00\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x02\x00\x02\x00\x01\x00\x01\x00\x02\x00\x02\x00\x01\x00\x00\x00\x01\x00\x01\x00\x02\x00\x02\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x01\x00\x01\x00\x00\x00\x01\x00\x00\x00\x01\x00\x02\x00\x01\x00\x01\x00\x01\x00\x00\x00\x01\x00\x02\x00\x01\x00\x02\x00\x01\x00\x00\x00\x01\x00\x02\x00\x00\x00\x01\x00\x00\x00\x01\x00\x01\x00\x01\x00\x01\x00\x02\x00\x01\x00\x01\x00\x01\x00\x00\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x00\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x02\x00\x01\x00\x01\x00\x00\x00\x00\x00\x00\x00\x01\x00\x02\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x00\x00\x02\x00\x01\x00\x01\x00\x00\x00\x02\x00\x00\x00\x01\x00\x01\x00\x02\x00\x01\x00\x01\x00\x01\x00\x02\x00\x01\x00\x01\x00\x01\x00\x02\x00\x01\x00\x01\x00\x02\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x00\x00\x00\x00\x02\x00\x02\x00\x02\x00\x01\x00\x01\x00\x00\x00\x02\x00\x01\x00\x01\x00\x00\x00\x00\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x00\x00\x01\x00\x01\x00\x01\x00\x00\x00\x01\x00\x02\x00\x02\x00\x01\x00\x01\x00\x01\x00\x00\x00\x01\x00\x01\x00\x01\x00\x01\x00\x02\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x00\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x02\x00\x02\x00\x01\x00\x02\x00\x00\x00\x01\x00\x00\x00\x01\x00\x01\x00\x01\x00\x01\x00\x02\x00\x02\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x00\x00\x01\x00\x01\x00\x02\x00\x00\x00\x00\x00\x02\x00\x00\x00\x01\x00\x00\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x02\x00\x02\x00\x02\x00\x01\x00\x00\x00\x00\x00\x01\x00\x02\x00\x01\x00\x01\x00\x01\x00\x01\x00\x00\x00\x02\x00\x02\x00\x01\x00\x01\x00\x01\x00\x02\x00\x02\x00\x02\x00\x00\x00\x01\x00\x01\x00\x01\x00\x00\x00\x01\x00\x02\x00\x02\x00\x01\x00\x01\x00\x02\x00\x02\x00\x00\x00\x00\x00\x00\x00\x01\x00\x01\x00\x01\x00\x02\x00\x02\x00\x01\x00\x01\x00\x01\x00\x00\x00\x02\x00\x02\x00\x01\x00\x02\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x00\x00\x01\x00\x02\x00\x01\x00\x02\x00\x02\x00\x00\x00\x02\x00\x01\x00\x00\x00\x01\x00\x01\x00\x01\x00\x02\x00\x01\x00\x02\x00\x02\x00\x01\x00\x01\x00\x00\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x02\x00\x01\x00\x01\x00\x00\x00\x01\x00\x01\x00\x00\x00\x01\x00\x01\x00\x02\x00\x02\x00\x02\x00\x01\x00\x02\x00\x01\x00\x02\x00\x01\x00\x01\x00\x01\x00\x02\x00\x01\x00\x01\x00\x02\x00\x02\x00\x00\x00\x00\x00\x01\x00\x00\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x02\x00\x01\x00\x02\x00\x02\x00\x02\x00\x01\x00\x01\x00\x01\x00\x02\x00\x01\x00\x01\x00\x02\x00\x00\x00\x01\x00\x01\x00\x02\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x02\x00\x00\x00\x01\x00\x00\x00\x00\x00\x02\x00\x01\x00\x00\x00\x00\x00\x01\x00\x00\x00\x02\x00\x02\x00\x01\x00\x01\x00\x01\x00\x02\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x02\x00\x01\x00\x01\x00\x00\x00\x00\x00\x02\x00\x00\x00\x02\x00\x01\x00\x02\x00\x01\x00\x02\x00\x01\x00\x01\x00\x02\x00\x01\x00\x01\x00\x01\x00\x02\x00\x02\x00\x01\x00\x02\x00\x01\x00\x00\x00\x00\x00\x01\x00\x01\x00\x01\x00\x02\x00\x01\x00\x01\x00\x02\x00\x01\x00\x00\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x02\x00\x01\x00\x01\x00\x02\x00\x01\x00\x01\x00\x01\x00\x02\x00\x02\x00\x01\x00\x00\x00\x02\x00\x00\x00\x01\x00\x01\x00\x00\x00\x02\x00\x00\x00\x00\x00\x01\x00\x00\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x02\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x02\x00\x00\x00\x01\x00\x02\x00\x01\x00\x01\x00\x01\x00\x02\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x01\x00\x02\x00\x01\x00\x02\x00\x01\x00\x02\x00\x01\x00\n'
2020-07-28 10:21:44,604 - pyvisa - DEBUG - GPIB::1::INSTR - closing
2020-07-28 10:21:44,604 - pyvisa - DEBUG - viDisableEvent(1, 1073709055, 65535) -> 1073676291
2020-07-28 10:21:44,604 - pyvisa - DEBUG - viDiscardEvents(1, 1073709055, 65535) -> 1073676292
2020-07-28 10:21:44,605 - pyvisa - DEBUG - viClose(1,) -> 0
2020-07-28 10:21:44,606 - pyvisa - DEBUG - GPIB::1::INSTR - is closed
2020-07-28 10:21:44,612 - pyvisa - DEBUG - Closing ResourceManager (session: 4097)
2020-07-28 10:21:44,614 - pyvisa - DEBUG - viClose(4097,) -> 0
Press any key to continue . . .`

Can someone point me in the right direction please? Do let me know what other information you need. Thank you!

MatthieuDartiailh, 2020-07-28:
From the message your instrument returned it appears it is configured to transfer data as binary not ascii. You can try to use query_binary_values. You will need to know what kind of number you expect. Oscilloscope often returns integer and you need to know the current settings of the instrument to convert that to an actual voltage. I suggest you give a look at your instrument manual that may contain some example. I have not worked with this model but I can send to you an example for a different one.

tzechong94, 2020-07-28:
Hi Matthieu,

Thank you so much for the reply! The error is gone after using query_binary_values. The data below is what I get, but like what you said, they are not the voltage values. 

```[-1.9250867343322097e-301, -3.116150495490394e-306, -2.0422021687836712e-301, -2.042200388724585e-301, -2.1588565609783996e-301, -2.042202161857477e-301, -2.0417446935984682e-301, -2.0422003817713343e-301, -1.2619234956837807e-296, -1.925086734359266e-301, -2.0417446935712006e-301, -3.116150484880415e-306, -2.041744693598362e-301, -1.3383752801750898e-296, -1.925540649399346e-301, -2.0417446866451124e-301, -1.9259999115698296e-301, -1.9250867343593716e-301, -2.042200388697529e-301, -2.0422003887245844e-301, -1.9255442234242308e-301, -3.116147779335944e-306, -2.0426560838509128e-301, -2.041744686645112e-301, -1.925542443365144e-301, -3.294156404170969e-306, -1.3383764513114946e-296, -1.2619234911268292e-296, -3.116147779253052e-306, -3.1161505060589274e-306, -1.92554065632554e-301, -2.0422003887245844e-301, -2.0421986086656037e-301, -2.042200381744173e-301, -3.1161504848805766e-306, -2.04220038872469e-301, -2.0421986086656037e-301, -2.0422003956780463e-301, -1.338376442197662e-296, -2.041744693598362e-301, -1.925542429431271e-301, -2.041744686645112e-301, -2.0422003817713347e-301, -1.9255442095175187e-301, -1.3383752847320416e-296, -3.116150495490556e-306, -2.0426542968112025e-301, -1.9250867412854598e-301, -2.0417446935984682e-301, -2.938141860003943e-306, -2.158860121069412e-301, -1.3383764421798613e-296, -3.294156404170969e-306, -2.0421986086656037e-301, -2.9381445762000005e-306, -2.041744693598468e-301, -2.0421986086656037e-301, -1.9255406563525952e-301, -2.0422021687566158e-301, -3.1161504954905556e-306, -2.041746466704093e-301, -3.116150495531839e-306]```

https://sg.tek.com/oscilloscope/tds794d-manual/tds500d-tds600c-tds700d-tds714l-service-manual I have seen the manual but I'm not too sure what is the conversion to change these readings to an average voltage reading. 

Could you send me the example for another model for reference please? Thank you so much :)

MatthieuDartiailh, 2020-07-28:
Here is the example (had to recompose it a bit so hopefully it still works), let me know if you run into any issue:
```python
import numpy
import pyvisa
rm = pyvisa.ResourceManager()
inst = rm.open_resource("GPIB::1::INSTR")

# Select the channel to transfer and select the format (binary, 2 byte per point, little endian)
inst.write(":DATA:SOU CH%d;:WFMP:ENC BIN;:WFMP:BYT_N 2;:WFMP:BYT_O MSB'"% channel)  

# Query range and offset
(gain, ioffset, offset) = inst.query_ascii_values(":WFMP:YMU?;:WFMP:YOF?;:WFMP:YZE?;", separator=";")

# Get data as i16, convert to numpy array
raw_values = inst.query_binary_values('CURV?', datatype="h", container=np.ndarray)

# Convert to actual voltage
voltages = gain*(raw_values - ioffset) + offset)
```

tzechong94, 2020-07-29:
Hi Matthieu, I faced a runtime error. 

```  File "C:\Users\NUSOSEC\Desktop\Oscilloscope\Oscilloscope.py", line 21, in <module>
    (gain, ioffset, offset) = inst.query_ascii_values(":WFMP:YMU?;:WFMP:YOF?;:WFMP:YZE?;", separator=";")
  File "C:\Users\NUSOSEC\anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 672, in query_ascii_values
    return self.read_ascii_values(converter, separator, container)
  File "C:\Users\NUSOSEC\anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 460, in read_ascii_values
    block = self.read()
  File "C:\Users\NUSOSEC\anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 427, in read
    message = self._read_raw().decode(enco)
  File "C:\Users\NUSOSEC\anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 400, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\NUSOSEC\anaconda3\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1584, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\NUSOSEC\anaconda3\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 193, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
Press any key to continue . . .```

Any idea what could be the problem? Thank you.

MatthieuDartiailh, 2020-07-29:
As I said it was not the same model. You can try command one at a time to check and see if the oscilloscope  complains about anything. Also some Tek ise WFMP and some WFMO so you can try that too.

tzechong94, 2020-07-29:
Thank you Matthieu!

closed 2020-07-29 by tzechong94

MatthieuDartiailh, 2020-07-29:
Did you get it to work ?

tzechong94, 2020-07-29:
Not yet, but I will read the instrument manual and try again. WFMO gives me the same timeout error. :(

tzechong94, 2020-07-29:
Hi Matthieu,

Thanks to your guidance, I got it to work. After reading the programmer manual of the oscilloscope, I realised that for this model, the problem was that I have to specify the waveform channel for the query form of YMUL, YOF and YZE. Also, I changed the datatype to 'b'.

Thank you so much! 

```
import numpy as np
import pyvisa
rm = pyvisa.ResourceManager()
inst = rm.open_resource("GPIB::1::INSTR")
print(inst.query('*IDN?'))

channel = 1
# Select the channel to transfer and select the format (binary, 2 byte per point, little endian)
inst.write(":DATA:SOU CH1;:WFMP:ENC BIN;:WFMP:BYT_N 1;:WFMP:BYT_O MSB")  

# Query range and offset
(gain, ioffset, offset) = inst.query_ascii_values(":WFMP:CH1:YMU?;:WFMP:CH1:YOF?;:WFMP:CH1:YZE?;", separator=";")

# Get data as i16, convert to numpy array
raw_values = inst.query_binary_values('CURV?', datatype="b", container=np.ndarray)
# Convert to actual voltage
voltages = gain*(raw_values - ioffset) + offset
print(voltages)
```

MatthieuDartiailh, 2020-07-29:
Happy you got it to work.