pyvisa/pyvisa #532: write_binary_values() doesn't convert bytes objects correctly

ABillBlakely, 2020-08-08:
I am having trouble working with the write_binary_values, function to write a file to the instrument.

I can transfer files, including text and images, from the instrument to my local PC just fine with the following code block:

 ```python
with open('screen.png', 'wb') as local_file:
    # datatype='s' tells pyvisa to transform IEEE format binary data to python bytes object.
    data = instrument.query_binary_values('MMEMory:DATA? "screen.png"', datatype='s')
    # results in a list of one item so access that item:
    local_file.write(data[0])
```

But if I try to mimic this the other direction:
```python
with open('data.txt', 'rb') as local_file:
    data = local_file.read()
    instrument.write_binary_values('MMEMory:DATA "test.txt",', data, datatype='s')
print(instrument.query("SYStem:ERRor?"))
```
I get an error:
```
...pyvisa\util.py", line 496, in to_binary_block
    return header + struct.pack(fullfmt, *iterable)
struct.error: pack expected 1 items for packing (got 14)
```

If I place `data` in a list: i.e. `data = [local_file.read()]` It no longer throws an error, but only the first letter of my text file is written. 

The whole issue can be reproduced without any instrument with just the following:

```python
with open('input.txt', 'rb') as local_file:
    data = [local_file.read()]
    data_encoded = pyvisa.util.to_ieee_block(data, datatype='s')
with open('output.txt', 'wb') as local_file:
    data_decoded = pyvisa.util.from_ieee_block(data_encoded, datatype='s')
    local_file.write(data_decoded[0])
```
My input file contains "Hello World!" and the output just contains "H"

ABillBlakely, 2020-08-08:
I played around with the last line of to_binary_block() and came up with:

```python
    if isinstance(iterable, bytes):
        return header + struct.pack(fullfmt, iterable)
    else:
        return header + struct.pack(fullfmt, *iterable)
```
which seems to work in at least this one case. There appears to be an issue expanding bytes object with the * operator. I don't understand enough around the structs/bytes to know if I'm causing another problem here though.

MatthieuDartiailh, 2020-08-09:
The s format is special in that it outputs a single string when unpacking and expect a single bytes object when packing.
We could special case the behavior of the utility functions for the s format, rather than special casing on bytes which is more likely to cause issues.

ABillBlakely, 2020-08-10:
Thank you for the comment Matthieu.

I'm still trying to understand what is wrong here. After reading the file as binary, I have a single bytes object that is passed in to the function.

`to_ieee_block(b'Hello', 's')` results in the same error, but if I'm understanding you this should work.

MatthieuDartiailh, 2020-08-10:
No I do not think that currently the "s" parameter can work (honestly I am surprised it kind of works when reading). For this to work we need to treat it differently (along with "p" for the safe of completeness) in `to_binary_block`. 
Something like:
```python
def to_binary_block(
    iterable: Sequence[Union[int, float]],
    header: Union[str, bytes],
    datatype: BINARY_DATATYPES,
    is_big_endian: bool,
) -> bytes:
    """Convert an iterable of numbers into a block of data with a given header.

    Parameters
    ----------
    iterable : Sequence[Union[int, float]]
        Sequence of numbers to pack into a block.
    header : Union[str, bytes]
        Header which should prefix the binary block
    datatype : BINARY_DATATYPES
        Format string for a single element. See struct module.
    is_big_endian : bool
        Are the data in big or little endian order.

    Returns
    -------
    bytes
        Binary block of data preceded by the specified header

    """
    array_length = len(iterable)
    endianess = ">" if is_big_endian else "<"
    fullfmt = "%s%d%s" % (endianess, array_length, datatype)

    if isinstance(header, str):
        header = bytes(header, "ascii")

    if datatype in ("s", "p"):
        block = struct.pack(datatype, iterable)

    else:
        block = struct.pack(fullfmt, *iterable)

    return header + block
```
Note that you can also avoid creating a temporary list by passing `container=bytes` to `query_binary_values`.

If this works feel free to open a PR.

MatthieuDartiailh, 2020-08-20:
The solution was a bit more involved. Please have a look at #536 and let me know if it works for you.

commit a61f6b7b (2020-12-07): Merge #536

536: pyvisa.utils: add support for "s" and "p" format r=MatthieuDartiailh a=MatthieuDartiailh

- [x] Closes #532 
- [x] Executed ``black . && isort -c . && flake8`` with no errors
- [x] The change is fully covered by automated unit tests
- [ ] Documented in docs/ as appropriate
- [ ] Added an entry to the CHANGES file