pyvisa/pyvisa #533: Very slow read with instruments connected by GPIB

predator41, 2020-08-10:
I want to configure the Model 2400 for maximum rate with the following commands ,but it takes about 22 Seconds to get the data, I test the same commands with NI VISA Test Panel, it's much faster
Any way to speed up?

```
import visa
import time

visa_address = 'GPIB0::16::INSTR'

rm = visa.ResourceManager()
smu = rm.open_resource(visa_address)
smu.timeout = 3000 # ms
smu.write_termination = '\n'
smu.read_termination = '\n'
smu.send_end = True

print(smu.query("*IDN?"))

cmds = [
        "*RST",
        ":SOUR:FUNC VOLT",
        ":SOUR:VOLT:RANG 20",
        ":SENSE:FUNCTION:CONC OFF",
        ":SENS:FUNC \'CURR:DC\'",
        ":SOUR:VOLT 0",
        ":SOUR:DEL 0",
        ":DISP:DIG 4",
        "TRIG:DEL 0",
        "TRIG:COUNT 2100",
        ":SENSE:CURR:NPLC 0.01",
        ":SENS:CURR:PROT 10E-3",
        ":SENS:CURR:RANG 10E-3",
        ":SENSE:AVER:STAT OFF",
        "SYST:AZER:STAT OFF",
        "DISP:ENAB OFF",
        "*CLS",
        ":TRAC:POIN 2100",
        ":TRAC:CLE",
        ":TRAC:FEED:CONT NEXT",
        "STAT:MEAS:ENAB 512",
        "*SRE 1"
       ]

for cmd in cmds:
    smu.write(cmd)

smu.query("*OPC?")
smu.write(":OUTP ON")
smu.write("INIT")
smu.write("DISP:ENAB ON")
smu.write(":OUTP OFF")

start_time = time.time()
raw_data = smu.query("TRAC:DATA?")
#print(raw_data)
elapsed_time = time.time() - start_time
print("Elapsed Time:{}\r\n".format(elapsed_time))

smu.close()
rm.close()
```

Instrument details
------------------
* Model:2400

* Communication: GPIB

* Development Environment: Python 3.7, PyVisa 1.10, NI-VISA 20.0, Windows 10

Output of `pyvisa-info`
-----------------------
KEITHLEY INSTRUMENTS INC.,MODEL 2410,xxxx,C34 Sep 21 2016 15:30:00/A02  /J/K

Elapsed Time:22.975309371948242

MatthieuDartiailh, 2020-08-10:
I will try to reproduce locally during the week. However I have a couple of questions:
- looking at your code, you do not wait for the acquisition to be complete before trying to retrieve the data. Given your NPLC settings and your number of points I assume the acquisition should take ideally less than 4 seconds. Is that so ?
- can you give me an estimate of how long it takes using the NI VISA test panel ?
- are you transferring data in ASCII or binary format ? I assume ASCII since you use a simply query, however if speed is a concern using the binary format would give better results.

predator41, 2020-08-11:
Thank you for the comment Matthieu.

I think I find the reason why the time difference happened, the default bytes to read of NI VISA Test Panel is 1024 which is much less than the bytes of 2100 readings

- The maximum speed of the 2400 is about 2000 readings per second. it takes about 1.3 seconds for the 2400 to compete the acquisition. Do you suggest me I should add a time sleep after the INIT command?

-  I am new to instruments control, I assume that I have to transfer the binary format data to current values 、time values if using the binary format,  but I don't how to implement it. Do you konw what else besides the binary format can improve the 
data transfer speed？

closed 2020-08-11 by predator41

MatthieuDartiailh, 2020-08-13:
Here is what you can try:
```python
import numpy as np
import pyvisa
import time

visa_address = 'GPIB0::16::INSTR'

rm = pyvisa.ResourceManager()
smu = rm.open_resource(visa_address)
smu.timeout = 3000 # ms
smu.write_termination = '\n'
smu.read_termination = '\n'
smu.send_end = True

print(smu.query("*IDN?"))

cmds = [
        "*RST",
        ":SOUR:FUNC VOLT",
        ":SOUR:VOLT:RANG 20",
        ":SENSE:FUNCTION:CONC OFF",
        ":SENS:FUNC \'CURR:DC\'",
        ":SOUR:VOLT 0",
        ":SOUR:DEL 0",
        ":DISP:DIG 4",
        "TRIG:DEL 0",
        "TRIG:COUNT 2100",
        ":SENSE:CURR:NPLC 0.01",
        ":SENS:CURR:PROT 10E-3",
        ":SENS:CURR:RANG 10E-3",
        ":SENSE:AVER:STAT OFF",
        "SYST:AZER:STAT OFF",
        "DISP:ENAB OFF",
        "*CLS",
        ":TRAC:POIN 2100",
        ":TRAC:CLE",
        ":TRAC:FEED:CONT NEXT",
        "STAT:MEAS:ENAB 512",
        "*SRE 1",
        ":FORM:DATA SREAL",  # request transfer in binary format
        ":FORM:ELEM VOLT,CURR,TIME",  # request voltage, current and time
        ":FORM:BORD NORM",  # request bytes in the normal order which I think is big endian (if the number seem all wrong change to SWAP)
       ]

for cmd in cmds:
    smu.write(cmd)

smu.query("*OPC?")
smu.write(":OUTP ON")
smu.write("INIT")
smu.query("*OPC?")
smu.write("DISP:ENAB ON")
smu.write(":OUTP OFF")

start_time = time.time()
raw_data = smu.query_binary_values("TRAC:DATA?",  datatype="f", is_big_endian=True, data_points=2100, container=np.ndarray)
# Reshape the array so that the first line is the voltage, the second the current and the third the time
data = raw_data.reshape((-1, 3).T
#print(raw_data)
elapsed_time = time.time() - start_time
print("Elapsed Time:{}\r\n".format(elapsed_time))

smu.close()
rm.close()
```
Let me know if it works and do not hesitate to ask if something goes wrong I could not test this code and I may have made some mistakes.

predator41, 2020-08-18:
Thanks Matthieu, It works！

MatthieuDartiailh, 2020-08-18:
Great ! Happy to have been helpful.