pyvisa/pyvisa #534: [TCPIP] Communication issue with Tektronix MDO3014 using PyVisa--communication works but instrument not listed in Resource Manager

IntegralWorks, 2020-08-10:
```
import pyvisa

rm = pyvisa.ResourceManager()

rm.list_resources()

('ASRL3::INSTR', 'ASRL10::INSTR')

inst = rm.open_resource('TCPIP::169.254.8.227::INSTR') #note: I manually found this ip address

inst.query('*IDN?')

'TEKTRONIX,MDO3014,C047688,CF:91.1CT FV:v1.26\n'
```

Also (fortunately), `inst.write("ACQ:STATE OFF")` and `inst.write("ACQ:STATE ON")` work. 
Why isn't `'TCPIP::169.254.8.227::INSTR'` listed in the Resource Manager?

Instrument details
------------------
* Model: Tektronix MDO3014
* Communication: TCPIP
* Link to the documentation (if available): https://github.com/pyvisa/pyvisa/files/4919729/MDO4000-B-MSO-DPO4000B-and-MDO3000-Oscilloscope-Programmer-Manual-Rev-A.pdf

Output of `pyvisa-info`
-----------------------

pyvisa-info
Machine Details:
   Platform ID:    Windows-10-10.0.15063-SP0
   Processor:      Intel64 Family 6 Model 58 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\jsalazar\AppData\Local\Programs\Python\Python38\python.exe
   Version:        3.8.2
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Feb 25 2020 23:03:10 (#tags/v3.8.2:7b3ab59)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 20971520
         Spec. Version: 5244928
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 20971520
         Spec. Version: 5244928

MatthieuDartiailh, 2020-08-13:
In some cases I have seen the VISA library does not scan all available TCPIP ports (which can be a good idea on a large network, typically Keysight VISA does this). In the case of NI-VISA, you can try starting NI-MAX and add manually the resource. After that the resource may be listed. But basically is NI-VISA does not report the resource there is not much PyVISA can do.

IntegralWorks, 2020-08-13:
`>>> rm.list_resources()
('TCPIP0::169.254.8.227::inst0::INSTR', 'TCPIP0::169.254.8.228::inst0::INSTR', 'ASRL3::INSTR', 'ASRL10::INSTR')
>>>`

https://imgur.com/a/ouTZoRq

Thank you for introducing me to a new tool.

closed 2020-08-13 by IntegralWorks

MatthieuDartiailh, 2020-08-13:
Happy you got it working.