pyvisa/pyvisa #535: Flushing the data in the output buffer of the device and timing issues for reading the output buffer

NykeMant, 2020-08-14:
Hello all,
i got some problems concerning reading the data, which my device sends to my PC. open_resource is connected to ('USB0::0x0D7D::0x0100::SPAD-LINE-2003033::INSTR',). My code looks like this:
`import xlsxwriter
import pyvisa
import time

# Create a workbook and add a worksheet.
workbook = xlsxwriter.Workbook('test_2.xlsx')
worksheet = workbook.add_worksheet()

'''Start Resource Manager'''
rm = pyvisa.ResourceManager()

test = rm.open_resource(rm.list_resources()[0])

pyvisa.log_to_screen()              # operator in use, so that first call of script result in a outcome
# time.sleep = 100

'''Reset Sensor'''
test.write_raw(b'\x3A')
time.sleep = 100

'''Set measurement size'''
test.write_raw(b'\x30\x00\x10')
time.sleep = 100

'''Start Measurement'''
test.write_raw(b'\x38')     
time.sleep = 100

bl_s = 100                  # defining blocksize to read

'''Read from device'''
read_test = test.read_raw(bl_s)

# Start from the first cell. Rows and columns are zero indexed.
row = 0
col = 0

# Iterate over the data and write it out row by row.
if int(read_test[3]) == 128:
    for smth in read_test:
        for item in read_test:
            worksheet.write(row, col, item)
            col += 1
            if col == 772:
                row += 1
                col = 0
        read_test = test.read_raw(bl_s)
else:
    print("fehler")
    test.flush(pyvisa.constants.VI_READ_BUF)
    # test.flush(pyvisa.constants.VI_WRITE_BUF_DISCARD)
    
workbook.close()

'''Close connection'''
test.clear()
test.before_close()
test.close()
rm.close()`

Using this code i want to document the measurement values into a excel table. At first i got the problem, that executing the code without `pyvisa.log_to_screen() ` i got an empty file and executing it another time the table contents are the expecte values and the expected values of the measurement form the first time executing the script. Then executing the code with `pyvisa.log_to_screen() ` i get the expected values everytime. Does anybody know another way around this problem?
For my second problem, i sometimes have the problem that i have defined a number which is too high and the transmitted values which are written to my my excel sheet aren't complete and are staying in the output buffer, meaning when i execute the code again first the missing values are transmitted and then the next values. Can someone help me optimize my for loop.
Another problem of mine is, even when my second problem can't be solved, i like to flush the buffer, so that for everytime i execute my code the measurement values i want to transmit to excel always start new.
I hope i have explained my problems understandable.
Best regards! 

[image]

MatthieuDartiailh, 2020-08-15:
From your edit I understand that your first problem is fixed (it was probably coming from an unflushed excel file or something similar).
Regarding your second issue, the code you have should do mostly what you want assuming `test.clear()` is actually called. You could try the following that may behave more consistently, however as I mention inline `read_raw` may not do what you believe, and you may need to use `read_bytes`:
```python
import xlsxwriter
import pyvisa
import time

# Create a workbook and add a worksheet.
workbook = xlsxwriter.Workbook('test_2.xlsx')
worksheet = workbook.add_worksheet()

'''Start Resource Manager'''
rm = pyvisa.ResourceManager()

# Use a context manager to ensure the resource is properly closed
with  rm.open_resource(rm.list_resources()[0]) as test:
    # Start by clearing the device, which should discard the output device of the buffer
    test.clear()
    # Alternatively you could try
    # test.flush(pyvisa.constants.BufferOperation.discard_read_buffer)
    # If that does not work (maybe your instrument does not implement all VISA standard command)
    # try:
    #     test.timeout = 10  # Set a short timeout
    #     test.read_raw()
    # except pyvisa.errors.VisaIOError:
    #     pass
    # test.timeout = 1000  # go back to a timeout of 1 s

    '''Reset Sensor'''
    test.write_raw(b'\x3A')
    time.sleep(0.1)

    '''Set measurement size'''
    test.write_raw(b'\x30\x00\x10')
    time.sleep(0.1)

    '''Start Measurement'''
    test.write_raw(b'\x38')     
    time.sleep(10)  # Whatever time make sense so that your system is done measuring

    bl_s = 100                  # defining blocksize to read

    '''Read from device'''
    # read_raw will read the device output till the instrument signal it is done, 
    # the size you specify is simply indicative of the typical block of data 
    # VISA will try to retrieve at once. If you need to read a given number of bytes 
    # use read_bytes, but note that the call will error if it cannot read the requested 
    # number of characters.
    read_test = test.read_raw(bl_s)

    # Start from the first cell. Rows and columns are zero indexed.
    row = 0
    col = 0

    # Iterate over the data and write it out row by row.
    if int(read_test[3]) == 128:
        for smth in read_test:
            for item in read_test:
                worksheet.write(row, col, item)
                col += 1
                if col == 772:
                    row += 1
                    col = 0
            read_test = test.read_raw(bl_s)
    else:
        print("fehler")
    
workbook.close()

'''Close connection'''
rm.close()
```

If you need more help could please consider sharing the manual or at least the model of the instrument you are communicating with.

vasilisniaouris, 2023-09-07:
I know this is an old issue, but I am encountering the same problem. I am have a USB device and which opens with the `ResourceManager` as a `SerialInstrument`. I have tried `.clear()`, `.flush(pyvisa.constants.BufferOperation.discard_read_buffer)` and neither of them work. Is there an update on how to deal with such issues? Is there an implementation of a 'read_all'-type function? If not, I think I will use a while loop and flag the `VisaIOError` to exit it.