pyvisa/pyvisa #538: Pyvisa unable to create ResourceManager - fails at viOpenDefaultRM()

kjohnson-automation, 2020-08-25:
I am unable to create a resource manager using pyvisa.

To Reproduce
--------------
Steps to reproduce the behavior:
python
import pyvisa
pyvisa.log_to_screen()
rm = pyvisa.ResourceManager()

log_to_screen output:
--------------
rm = pyvisa.ResourceManager()                                                                                                                                                                                                                                                                                                                                                                                                      2020-08-24 22:46:00,991 - pyvisa - DEBUG - No visa library specified, trying to find alternatives.
2020-08-24 22:46:00,991 - pyvisa - DEBUG - Environment variable PYVISA_LIBRARY is unset.
2020-08-24 22:46:00,994 - pyvisa - DEBUG - Automatically found library files: [None, 'C:\\Windows\\system32\\visa32.dll', 'C:\\Windows\\system32\\visa32.dll', 'C:\\Windows\\system32\\visa64.dll', 'C:\\Windows\\system32\\visa64.dll']
2020-08-24 22:46:00,997 - pyvisa - DEBUG - No user defined library files
2020-08-24 22:46:00,997 - pyvisa - DEBUG - The NI implementation available
2020-08-24 22:46:00,999 - pyvisa - DEBUG - Automatically found library files: [None, 'C:\\Windows\\system32\\visa32.dll', 'C:\\Windows\\system32\\visa32.dll', 'C:\\Windows\\system32\\visa64.dll', 'C:\\Windows\\system32\\visa64.dll']
2020-08-24 22:46:01,000 - pyvisa - DEBUG - No user defined library files
2020-08-24 22:46:01,002 - pyvisa - DEBUG - Library signatures: 90 ok, 0 failed
2020-08-24 22:46:01,002 - pyvisa - DEBUG - Created library wrapper for C:\Windows\system32\visa32.dll
2020-08-24 22:46:01,005 - pyvisa - DEBUG - viOpenDefaultRM('<ViObject object at 0x000002B6CA4658C8>',) -> -1073807360

Traceback:
--------------
VisaIOError
Traceback (most recent call last)
<ipython-input-3-a27db25a2dfc> in <module>
----> 1 rm = pyvisa.ResourceManager()
c:\users\kmj44\appdata\local\programs\python\python37\lib\site-packages\pyvisa\highlevel.py in __new__(cls, visa_library)                                                                                                                                                                                                                                                                                                                     1560         obj = super(ResourceManager, cls).__new__(cls)
1561
-> 1562         obj.session, err = visa_library.open_default_resource_manager()
1563
1564         obj.visalib = visa_library
c:\users\kmj44\appdata\local\programs\python\python37\lib\site-packages\pyvisa\ctwrapper\functions.py in open_default_resource_manager(library)
1225     """
1226     session = ViSession()
-> 1227     ret = library.viOpenDefaultRM(byref(session))
1228     return session.value, ret
1229
c:\users\kmj44\appdata\local\programs\python\python37\lib\site-packages\pyvisa\ctwrapper\highlevel.py in _return_handler(self, ret_value, func, arguments)
191
192         if ret_value < 0:
--> 193             raise errors.VisaIOError(ret_value)
194
195         if ret_value in self.issue_warning_on:
VisaIOError: VI_ERROR_SYSTEM_ERROR (-1073807360): Unknown system error (miscellaneous error).

**Output of `pyvisa-info`**
--------------
pyvisa-info
Machine Details:
   Platform ID:    Windows-10-10.0.18362-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 13, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\users\kmj44\appdata\local\programs\python\python37\python.exe
   Version:        3.7.4
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Jul  8 2019 20:34:20 (#tags/v3.7.4:e09359112e)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      pyvisa/pyvisa-py#1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Could not get more info:
            VI_ERROR_SYSTEM_ERROR (-1073807360): Unknown system error (miscellaneous error).
      pyvisa/pyvisa-py#2: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Could not get more info:
            VI_ERROR_SYSTEM_ERROR (-1073807360): Unknown system error (miscellaneous error).
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'

***General Notes***
I can cleanly communicate with GPIB devices that are hosted on a visa server remotely through the interactive panel from NI-MAX (Version 19.5), granted I am not even at that point here yet as it cannot successfully create a resource manager for me to try and open a connect with.

MatthieuDartiailh, 2020-08-25:
Do you have multiple version of VISA installed ? 

I have seen similar issues in the past and I kind of remember there were linked to bad VISA installs (with old dll lying around but not pointing to the right implementation). Try to see if you find a visa.dll under the NI folder where NI-MAX is located and see if it works better by passing its pass to pyvisa. Otherwise I would suggest to fully uninstall NI-VISA, fully re-install and restart the computer and see if it fixes things.

kjohnson-automation, 2020-08-25:
I have had multiple versions of Visa installed - at one point in time my NI-MAX visa install wasn't working at all so I had tor removed it, and even then it didn't work so after reinstall I had to set the config to default and reload and let it rebuild it again. I will look for the visa.dll.
Yesterday I had removed everything and reinstalled it (including package manager) each with a reboot in between.
Would it also help to remove pyvisa and reinstall that?

Thanks for the fast reply - this Lib has been so useful to me over the years and honestly its my first problem I've had with it on any OS.

MatthieuDartiailh, 2020-08-25:
I do not expect re-installing pyvisa to change anything no. However if nothing works you can also consider re-installing Python in case something is corrupted in ctypes.

kjohnson-automation, 2020-08-25:
I did not find any dlls in the NI-MAX folder that seem to do with VISA, only the ones in system32 and syswow64 and on 1 in each. I will look at reinstalling python.

kjohnson-automation, 2020-08-25:
Reinstalling python worked.
Thank you.

MatthieuDartiailh, 2020-08-25:
Happy you got it to work. Would you mind adding an entry in the documentation about this issue (under faq/faq.rst) ?

kjohnson-automation, 2020-08-25:
Sure.

MatthieuDartiailh, 2020-08-25:
Closed by #539

closed 2020-08-25 by MatthieuDartiailh

MatthieuDartiailh, 2022-07-21:
Please open a new issue with the output of pyvisa-info and the full reaceback of the error you are seeing.