pyvisa/pyvisa #541: VI_ERROR_INV_OBJECT multiple instruments from different processes

jennyan, 2020-08-31:
Hello,

I'm having some trouble with connecting to more than one instrument using the resource manger. The first instrument works fine. 

rm = visa.ResourceManager()
inst = rm.open_resource( TCPIP instrument )

The rm session id is 256 which seems a bit odd but there are no error when opening the default resource manager.
The inst session id is 1, which also seems a bit odd.

The second instrument is in another process. The rm is instantiated in the same way and yields the same session id (256). But when I try to open another TCPIP instrument I get VI_ERROR_INV_OBJECT. 

I'm using pyvisa 1.10.1, python 3.5, on a linux system. The backend is ni. I don't close the instruments inbetween trying to open a new.

I can't seem to find the issue here, some help?

Many thanks,
Jenny

IntegralWorks, 2020-08-31:
Do you have NI-MAX installed, and do you have something resembling this on your screen: https://imgur.com/a/2rV5Utp

I had a very similar problem to yours quite recently and I resolved the issue by "manually" adding my Ethernet connected devices in NI-MAX.
It probably is the _exact same issue_ if you had a working system before and one day due to installing some other software/an update/cosmic rays your before working VISA setup suddenly just broke.

MatthieuDartiailh, 2020-08-31:
My other question question is how do you create your second process ? If you forked it from your main Python process using multiprocessing, can you try spawning instead in case we are running in some weird issue with the state of the library on Linux and forking...

MatthieuDartiailh, 2020-08-31:
If the fix from @IntegralWorks works, could one of you contribute an entry to the FAQ about it ?

IntegralWorks, 2020-08-31:
> If the fix from @IntegralWorks works, could one of you contribute an entry to the FAQ about it ?

Yeah, I was about to say that installing NI-MAX might not be obvious to some, so if the issue (and fix) is reproduced, I'll add an entry.

MatthieuDartiailh, 2020-08-31:
Also @jennyan could you post the output of `pyvisa-info`. I would like to know what VISA implementation you are using... I am wondering if you are not using pyvisa-py since we fallback to it if we don't find a VISA implementation. Also minimal version of the script running in each process would help.

jennyan, 2020-09-02:
I haven't had the possibility to try the NI-MAX suggestion, so I can't confirm that. 

I did try to change to spawning the process instead but run into other issues with my system, but the problem was due to the process handling somehow. I made a workaround where I instantiate both instruments from the same process and then the resource manager works fine with both instrument.

Machine Details:
   Platform ID:    Linux-4.15.0-112-generic-x86_64-with-debian-buster-sid
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /home/usr/anaconda3/envs/python35/bin/python
   Version:        3.5.4
   Compiler:       GCC 7.2.0
   Bits:           64bit
   Build:          Feb 19 2018 10:59:04 (#default)
   Unicode:        UCS4 

PyVISA Version: 1.10.1

 Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libiovisa.so:
         found by: auto
         bitness: n/a
         Vendor: Keysight Technologies
         Impl. Version: 1376071104
         Spec. Version: 5244928

MatthieuDartiailh, 2020-09-02:
Since you are running Keysight VISA, do you have Connection Expert which is the equivalent of NI-MAX ?