pyvisa/pyvisa #542: [COM] Communication issue with Tektronix MDO4054C using for loop

mimirotimi, 2020-09-03:
Instrument details
------------------
* Model: Tektronix MDO4054C Mixed Domain Oscilloscope
* Communication: TCPIP
* Link to the documentation (if available):

Output of `pyvisa-info`
-----------------------

C:\Users\englocal>pyvisa-info
Machine Details:
   Platform ID:    Windows-10-10.0.10586-SP0
   Processor:      Intel64 Family 6 Model 58 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\users\englocal\appdata\local\programs\python\python38-32\python.exe
   Version:        3.8.5
   Compiler:       MSC v.1926 32 bit (Intel)
   Bits:           32bit
   Build:          Jul 20 2020 15:43:08 (#tags/v3.8.5:580fbb0)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      #1: C:\windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: Tektronix
         Impl. Version: 4194816
         Spec. Version: 4194304

Hi, I am trying to write a program that will take the measurement of the current every minute over 48 hours. However, after a certain amount of time (anywhere between 4 to 60 minutes) the program stops and I get a miscellaneous error. In the code you will notice that I open and close the resource every loop, this is because when I didn't I would get an I/O error. I appreciate any help I can get. 
The code:

import pyvisa
import time
import pandas as pd
visa
visa.ResourceManager
from visa import ResourceManager
rm = visa.ResourceManager()
ip_address = "TCPIP0::169.254.4.23::inst0::instr"
test_time = 1
for i in range(172800):
	tek = rm.get_instrument(ip_address)
	current = tek.query('measurement:meas1:value?')
	data = {
		'Time': [test_time],
		'Current': [current]
		}
	time.sleep(60)
	test_time = test_time + 1
	results = pd.DataFrame(data)
	results
	results.to_csv(r'C:\Users\englocal\Documents\LUNAR\Results.csv', index = False, header = False, mode = 'a')
	tek.close

I would like to reiterate that the code works fine but only for a limited amount of time. I should note I have tried modifying the read and write terminations in the get_instrument command but it did not seem to help.

MatthieuDartiailh, 2020-09-03:
Could you please report the exact error you are seeing in both the situations you describe ? Also in the script you posted you do not close the instrument since the close method is not called only accessed. If you want to ensure the communication is closed each time I would recommend using a context manager as follows:
```python
import pyvisa
import time
import pandas as pd

rm = pyvisa.ResourceManager()
ip_address = "TCPIP0::169.254.4.23::inst0::instr"
test_time = 1
for i in range(172800):
        with rm.open_resource(ip_address) as tek:
             # This gives you a string you may want to convert to a float before storing it.
	     current = tek.query('measurement:meas1:value?')
	data = {'Time': [test_time],  'Current': [current]}
	time.sleep(60)
	test_time = test_time + 1
	results = pd.DataFrame(data)
	results.to_csv(r'C:\Users\englocal\Documents\LUNAR\Results.csv', index = False, header = False, mode = 'a')
```

mimirotimi, 2020-09-03:
I do not have full copies of the errors and re-obtaining them would mean possibly waiting up to two hours for them. The error codes however are -1073807298 for the I/O error and the error code for the unknown error is -1073807360.  Just out of curiosity, what is the difference between calling and accessing the close instrument method. How would I call it explicitly?
Thanks for getting back to me so quickly.

MatthieuDartiailh, 2020-09-03:
Close does not require any argument so you would do `tek.close()`. Accessing it just give you a bound method object, that you could decide to pass to another function which is why access does not imply call. 
Intermittent errors are the worst. I do not have a magic solution since it may be an issue on the instrument side. When the error occurs can you try to check the instrument log (using its front panel to see if the instrument complains about something), if there is nothing there consider using a try/except block and attempt multiple time the operation with possibly a call to `clear` between two attempts.

mimirotimi, 2020-09-03:
Alright thanks. I did try using a try/except block but I'm not too familiar with python. I had the code in the try block and attempting to reopen the connection in the excpet block. Does that sound right? What does the clear command do?

mimirotimi, 2020-09-03:
Hi Matthieu, so I ran the code as you suggested and the new error I got  is now  -1073807342 . Invalid resource reference specified. Parsing error. I'm assuming this would happen if my reference was incorrect however the code ran for 25 loops.

MatthieuDartiailh, 2020-09-03:
The clear command empty all the buffers (in theory) which can be useful in case something got stuck somewhere. Getting an invalid resource reference on the 25 loop is very weird, in particular since you do not change the address I assume. Next time please post the full error traceback, the VISA code does not say anything else than this is an invalid resource reference. Also please post your updated code (except if you copy-pasted mine).

For the try/except you could do something like that:
```python
import pyvisa
import time
import pandas as pd
rm = pyvisa.ResourceManager()
ip_address = "TCPIP0::169.254.4.23::inst0::instr"
test_time = 1
tek = rm.open_resource(ip_address)

for i in range(172800):
    # Try 10 times before giving up
    for j in range(10)
        try:
            current = tek.query('measurement:meas1:value?')
            break
        except pyvisa.VisaIOError:
            if j == 9:
                raise
            else:
                tek.close()
                tek = rm.open_resource(ip_address)
                tek.clear()
    
    data = {'Time': [test_time], 'Current': [current]}
    time.sleep(60)
    test_time = test_time + 1
    results = pd.DataFrame(data)
    results.to_csv(r'C:\Users\englocal\Documents\LUNAR\Results.csv', index = False, header = False, mode = 'a')
```

IntegralWorks, 2020-09-04:
I'll test this code out when I get to work (no virtual instrument at home)

```
import sys
import pyvisa as visa
import time
import pandas as pd
from datetime import datetime, date, time, timedelta

now = datetime.now()
hours = float(f'{sys.argv[1]}')/24 #2/24 --> 1/12
target = datetime.timedelta(hours) #https://www.kite.com/python/answers/how-to-add-hours-to-the-current-time-in-python
target = now + target

rm = visa.ResourceManager()
ip_address = 'whatever'
scope = rm.open_resource(ip_address)

flag = False
data = []
while flag != True:
    if datetime.strftime(datetime.now(),'%X')[0:5] == target[0:5]:
        flag = True
    try:
        moment = datetime.now()
        current = scope.query('measurement:meas1:value?')
        time.sleep(5) #this might be faster than 5 secs, adjust accordingly
        data.append([moment,current])
    except visa.VisaIOError:
        scope.close()
        scope = rm.open_resource(ip_address)
        scope.clear()

df = pd.DataFrame(data,columns=['timestamp','current'])
df.to_csv()
```

mimirotimi, 2020-09-04:
> The clear command empty all the buffers (in theory) which can be useful in case something got stuck somewhere. Getting an invalid resource reference on the 25 loop is very weird, in particular since [...]

mimirotimi, 2020-09-04:
I have copied the code but added calling the results so that they print out on the Python IDLE. This is the full error I got for that:
Traceback (most recent call last):
  File "<pyshell#22>", line 1, in <module>
    tek = rm.get_instrument(ip_address)
  File "C:\Users\englocal\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\highlevel.py", line 1771, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Users\englocal\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\resources\resource.py", line 218, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "C:\Users\englocal\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\highlevel.py", line 1725, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "C:\Users\englocal\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1213, in open
    ret = library.viOpen(session, resource_name, access_mode, open_timeout, byref(out_session))
  File "C:\Users\englocal\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 193, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_INV_RSRC_NAME (-1073807342): Invalid resource reference specified. Parsing error.

Then with the try/except, I copied the code and called results again. These are the errors I got:
Traceback (most recent call last):
  File "<pyshell#9>", line 4, in <module>
    current = tek.query('measurement:meas1:value?')
  File "C:\Users\englocal\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\resources\messagebased.py", line 607, in query
    self.write(message)
  File "C:\Users\englocal\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\resources\messagebased.py", line 228, in write
    count = self.write_raw(message.encode(enco))
  File "C:\Users\englocal\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\resources\messagebased.py", line 202, in write_raw
    return self.visalib.write(self.session, message)
  File "C:\Users\englocal\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1866, in write
    ret = library.viWrite(session, data, len(data), byref(return_count))
  File "C:\Users\englocal\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 193, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<pyshell#9>", line 11, in <module>
    tek = rm.open_resource(ip_address)
  File "C:\Users\englocal\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\highlevel.py", line 1771, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Users\englocal\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\resources\resource.py", line 218, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "C:\Users\englocal\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\highlevel.py", line 1725, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "C:\Users\englocal\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1213, in open
    ret = library.viOpen(session, resource_name, access_mode, open_timeout, byref(out_session))
  File "C:\Users\englocal\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 193, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_INV_RSRC_NAME (-1073807342): Invalid resource reference specified. Parsing error.

MatthieuDartiailh, 2020-09-04:
Can you add a print before calls to `rm.open_resource` so that we see the value of ip_address ? If you do not modify ip_address you have no reason to get that error since the library crashes on parsing the address not even looking for the instrument. I am wondering if we are hitting a weird behavior of running in IDLE. Can you also try running in a command prompt ? 

Finally I just noticed that you are using a fairly old version of VISA (both in term of implementation and specs), may be worth updating. Note that you can use any vendor, it does not have to be Tek.

mimirotimi, 2020-09-04:
> Can you add a print before calls to `rm.open_resource` so that we see the value of ip_address ? If you do not modify ip_address you have no reason to get that error since the library crashes on [...]

I ran the code in the Python Command Prompt and although it ran longer I got the same error as before. I am unsure what you mean by vendors, I just did pip install in the computer's command prompt to install pyvisa. Is that wrong?

mimirotimi, 2020-09-04:
Okay this is strange, I turned the oscilloscope on and off to try again and the IP address that came onto the screen is now different.

MatthieuDartiailh, 2020-09-04:
Indeed...

To ensure your previous question, PyVISA provides access to the VISA library that needs to be installed separately. In your case as indicated in the ouput of `pyvisa-info` you are using a VISA library from Tektronik implementing the VISA specification. From the reported number your version is quite old (the specification reached 5. something).

IntegralWorks, 2020-09-05:
> 
> 
> Okay this is strange, I turned the oscilloscope on and off to try again and the IP address that came onto the screen is now different.

Install NI-Max and go to "network devices" and "manually" register your scope. When I get back to work this week I'll write a guide about how to do this. 

@MatthieuDartiailh I don't think it's a bug with PyVISA. Tektronix's "auto-IP" is kind of buggy and my setup consists of two MDO scopes connected to a hacked-router-converted-to-switch connected to a laptop that uses company wifi. the MDO scopes use their connections in two ways--to make available a feature called "e-scope" where you type in the ip address of scope in a browser to access a GUI console, and to make available VISA. However, sometimes you have to manually register the scope in either NI-MAX or Tektronix OpenChoice software. Again, I don't think this is a problem with PyVISA, this issue might even have to do with the implementation of DHCP or whatever on the MDO series.

MatthieuDartiailh, 2020-09-05:
Thanks for sharing @IntegralWorks . Honestly I never use DHCP with instruments only a local network. But indeed a randomly changing IP could make sense in the context of DHCP.

mimirotimi, 2020-09-08:
@IntegralWorks @MatthieuDartiailh hi all, right before leaving work on friday I set the IP address to a permanent address and let it run over the long weekend, and it worked! Thanks for the help, @IntegralWorks I think we caught it at about the same time.

closed 2020-10-13 by MatthieuDartiailh