pyvisa/pyvisa #544: TypeError in __del__ of resource

robsan00, 2020-09-21:
With the newest version of pyvisa, on Ubuntu I get a TypeError when the `Resource` object is cleaned up.

To Reproduce
--------------
Script:
```
import pyvisa

resource_manager = pyvisa.ResourceManager()
instrument = resource_manager.open_resource("USB0::0x14EB::0x0090::461816::INSTR")
```
Execution of this script results in the error:
```
python pyvisa_test.py
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/home/buibot00/venv_py3/test_ci/lib/python3.6/site-packages/pyvisa/highlevel.py", line 3086, in close
    resource.close()
  File "/home/buibot00/venv_py3/test_ci/lib/python3.6/site-packages/pyvisa/resources/resource.py", line 334, in close
    self.visalib.close(self.session)
  File "/home/buibot00/venv_py3/test_ci/lib/python3.6/site-packages/pyvisa_py/highlevel.py", line 439, in close
    return self.handle_return_value(session, sess.close())
  File "/home/buibot00/venv_py3/test_ci/lib/python3.6/site-packages/pyvisa/highlevel.py", line 248, in handle_return_value
    if rv < 0:
TypeError: '<' not supported between instances of 'NoneType' and 'int'
Exception ignored in: <bound method Resource.__del__ of <'USBInstrument'('USB0::0x14EB::0x0090::461816::0::INSTR')>>
Traceback (most recent call last):
  File "/home/buibot00/venv_py3/test_ci/lib/python3.6/site-packages/pyvisa/resources/resource.py", line 199, in __del__
  File "/home/buibot00/venv_py3/test_ci/lib/python3.6/site-packages/pyvisa/resources/resource.py", line 334, in close
  File "/home/buibot00/venv_py3/test_ci/lib/python3.6/site-packages/pyvisa_py/highlevel.py", line 439, in close
  File "/home/buibot00/venv_py3/test_ci/lib/python3.6/site-packages/pyvisa/highlevel.py", line 248, in handle_return_value
TypeError: '<' not supported between instances of 'NoneType' and 'int'
```

**Output of `pyvisa-info`**
```
pyvisa-info
Machine Details:
   Platform ID:    Linux-4.15.0-46-generic-x86_64-with-Ubuntu-18.04-bionic
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /home/buibot00/venv_py3/test_ci/bin/python3
   Version:        3.6.7
   Compiler:       GCC 8.2.0
   Bits:           64bit
   Build:          Oct 22 2018 11:32:17 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.0

Backends:
   ivi:
      Version: 1.11.0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.0
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.1.0). Backend: libusb1
      USB RAW: Available via PyUSB (1.1.0). Backend: libusb1
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
```

After reverting to the previous version of pyvisa (1.10.1), the error goes away, even when keeping pyvisa-py at 0.5.0, so I assume the issue is due to pyvisa, not pyvisa-py.

Please note that on a Windows machine, this problem does not occur. Here the `pyvisa-info` output of that machine:
```
pyvisa-info
Machine Details:
   Platform ID:    Windows-10-10.0.17763-SP0
   Processor:      Intel64 Family 6 Model 79 Stepping 1, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\work\venv\python3\scripts\python.exe
   Version:        3.6.4
   Compiler:       MSC v.1900 64 bit (AMD64)
   Bits:           64bit
   Build:          Dec 19 2017 06:54:40 (#v3.6.4:d48eceb)
   Unicode:        UCS4

PyVISA Version: 1.11.0

Backends:
   ivi:
      Version: 1.11.0 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
   py:
      Version: 0.5.0
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.1.0). Backend: N/A
      USB RAW: Available via PyUSB (1.1.0). Backend: N/A
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
```

MatthieuDartiailh, 2020-09-21:
Thanks for your report. I am honestly not sure how pyvisa-py 0.5.0 works with pyvisa<1.11... Anyway I believe the error is in PyVISA-py since the close method on usb resources do not return a StatusCode as it should. Please test https://github.com/pyvisa/pyvisa-py/pull/265 and let me know if it fixes your issue. If it does I will try to make a bugfix release later this week.

robsan00, 2020-09-22:
Thanks for the quick reply!

Yes, with your fix in pyvisa-py adding the missing return (referencing SHA 7f133c1ab9675efd45707562fa0fd37a5396925d), it works fine.

MatthieuDartiailh, 2020-09-23:
Good ! I will merge soon. I may try to bunch of couple of extra changes before the release if that is fine with you. Do not hesitate to ping me if the release take too long.

closed 2020-09-23 by bors[bot]