pyvisa/pyvisa #548: [COM] Communication issue with Keithley 2400 and 2182a using GPIB connection

brunoalderete, 2020-09-24:
Currently I am working on the acquisition of electrical contact resistance, for which I will be using a Keithley 2400 sourcemeter and a Keithley 2182a nanovoltmeter. Both of the instruments are connected to a windows computer via USB with a Keithley KUSB-488b adapter.

I am working with Python, the Pyvisa module, and the National Instruments backend (NI VISA) to create a script that will allow me to conduct the measurements needed. However I am running into trouble communicating with the instruments. I have run a couple of simple scripts, based on the official documentation provided by the developers of Pyvisa, but am unable to connect with either instrument. The following simple block of code may aid my explanation:

(1) >>> import pyvisa
(2) >>> rm = pyvisa.ResourceManager()
(3) >>> rm.list_resources()
(4)  ('ASRL3::INSTR', ASRL4::INSTR', 'ASRL8::INSTR', 'ASRL14::INSTR')
(5) >>> inst = rm.open_resource(' # Instrument address here # ')
(6) >>> print(inst.query("*IDN?")

The result in line (4) should give me a list of the addresses of my instruments. However, none of these are correct, hence I get a timeout error (VI_ERROR_TMO) when I try the four addresses provided by line (4). The expected output in line (4) of this simple code block would be the GPIB addresses of both instruments, as per the documentation.
I tried using the KI-488 Diagnostics Tool and N IMAX to communicate with both devices. Both of which I successfully communicated with. The SMU's address is GPIB::24 and the nanovoltmeter is GPIB::7.

I also tried passing the GPIB addresses that I obtained from the diagnostic tools in more complex scripts (current sweep). However this resulted in pyvisa.errors.VisaIOError: VI_ERROR_LIBRARY_NFOUND (-1073807202). I searched online for similar errors, most of which are an outcome of missing dll or incorrect bitness. However this is not my case, since a simple python -m visa info proves that the backends implemented are correct. I also tried passing the ResourceManager with the specific path of the visa dll, but this too resulted in an error.

Additionally I tried installing Keithley I/O Layer, but I was unsuccessful nonetheless. Also, in the Keithley communicator both devices do in fact appear with their corresponding GPIB address.

I would really appreciate any help that you may provide.
Thank you in advanced!

Instrument details
------------------
* Model: Keithley 2400 SourceMeter and Keithley 2182a Nanovoltmeter
* Communication: GPIB::24 for SourceMeter and GPIB::7 for Nanovoltmeter

Output of `pyvisa-info`
-----------------------
Machine Details:
Platform ID: Windows-7-6.1.7601-SP1
Processor: Intel64 Family 6 Model 94 Stepping 3, GenuineIntel

Python:
Implementation: CPython
Executable: C:\ProgramData\Anaconda3\python.exe
Version: 3.8.3
Compiler: MSC v.1916 64 bit (AMD64)
Bits: 64bit
Build: Jul 2 2020 17:30:36 (#default)
Unicode: UCS4

PyVISA Version: 1.11.0

Backends:
ivi:
Version: 1.11.0 (bundled with PyVISA)
#1: C:\windows\system32\visa32.dll:
found by: auto
bitness: 64
Vendor: National Instruments
Impl. Version: National Instruments
Spec. Version: National Instruments
#2: C:\windows\system32\visa64.dll
found by: auto
bitness: 64
Vendor: National Instruments
Impl. Version: National Instruments
Spec. Version: National Instruments

MatthieuDartiailh, 2020-09-28:
USB-GPIB adapters tend to be tricky and usually only supported by the VISA implementation of the manufacturer. However since NI-MAX works the problem must be different. 

I note you are running Python 3.8. This version of Python introduced some restriction on the locations from which dll can be loaded and may interfere with your process (as indicated by the error message pointing to a problem locating the library not the instrument). A quick way to check if that is likely to be the issue would be to downgrade to Python 3.7. Alternatively try to locate the Keithley dll handling gpib communication (likely under Program Files) and use `os.add_dll_directory` to add allow Python to load library from this location. If this works you can use a configuration file to make this setting permanent (see the documentation).

HideharuYamauchi, 2020-10-14:
I am facing a very similar issue.

I try to communicate with FLUKE RPM4 using GPIB-USB on CentOS8. Here is the output of PyVISA. 
[image]
The line `>>>rm.list_resources()` gives me a wrong address. I expect it gives me `GPIB0::1:INSTR` but it doesn't. 
It seems that PyVISA doesn't recognize GPIB.

The following is the output of `pyvisa-info` .
[image]

MatthieuDartiailh, 2020-10-14:
As mentioned above USB-GPIB only works with Visa library from the same manufacturer in general. What is the manufacturer of your adapter ?

HideharuYamauchi, 2020-10-15:
Thank you for your reply.
I'm using the GPIB-USB-HS of National Instruments. The backend of PyVISA is NI-VISA as above. They are the same manufacturer. Do you have any idea to solve this?

brunoalderete, 2020-10-15:
Hi Mattheiu and Hideharu,

I was able to solve this problem by using the pyvisa backend pyvisa-py ( rm = ResourceManager('@py') ) and it's working fine. I get the correct GPIB address and I am able to communicate with both instruments. Maybe that will work for you as well. I still don't understand why Ni-Visa is not working though.

HideharuYamauchi, 2020-10-15:
Thank you for your advice. I will give pyvisa-py a try.