pyvisa/pyvisa #549: instrument parameters missing? (like `io_protocol`)

greyltc, 2020-09-27:
[The docs say](https://pyvisa.readthedocs.io/en/latest/api/resources.html#pyvisa.resources.GPIBInstrument.io_protocol) I should be able to set an `io_protocol` paramater for my GPIB instrument.

But that doesn't work today with the following error:
```
pyvisa.errors.VisaIOError: VI_ERROR_NSUP_ATTR (-1073807331): The specified attribute is not defined or supported by the referenced object.
```
Maybe relatedly, I've noticed that a other parameters that used to work no longer do, `allow_dma` doesn't work on my serial instruments either for example.

**Output of `pyvisa-info`**
```
$ pyvisa-info
Machine Details:
   Platform ID:    Linux-5.8.11-arch1-1-x86_64-with-glibc2.2.5
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        3.8.5
   Compiler:       GCC 10.2.0
   Bits:           64bit
   Build:          Sep  5 2020 10:50:12 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.0

Backends:
   ivi:
      Version: 1.11.0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.0
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.1.0). Backend: libusb1
      USB RAW: Available via PyUSB (1.1.0). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR: Available via Linux GPIB (b'4.3.3 r[1916]')
      GPIB INTFC: Available via Linux GPIB (b'4.3.3 r[1916]')

```

MatthieuDartiailh, 2020-09-28:
PyVISA-py has support for a limited number of attributes and the ones you mention are actually not supported. I am not sure what was happening before (my changes should not have changed anything since I simply avoided monkeypatching the classes with attributes) but it was most likely a total no-op. In a sense the new behavior is more correct. Please open separate issues in PyVISA-py if you need support for those attributes. As a gap stopper we could also try to distinguish between valid but unsupported and invalid attributes. PR welcome.

MatthieuDartiailh, 2020-09-28:
Actually PyVISA-py already log the error causing an VI_ERROR_NSUP_ATTR when it results from it not being implemented. So if you log you should see that this happens because of an incomplete implementation on PyVISA-py side but it may be a bit cryptic.