pyvisa/pyvisa #551: Serial instrument not correctly recognised

gerritzen, 2020-09-29:
I was trying to set up a TTi MX-180TP via USB (which shows up as /dev/ttyACMx). 

`In [1]: import pyvisa                                                           

In [2]: rm = pyvisa.ResourceManager()                                           

In [3]: rm.list_resources()                                                     
Out[3]: ('ASRL/dev/ttyS4::INSTR', 'ASRL/dev/ttyACM1::INSTR')

In [4]: psu = rm.open_resource('ASRL/dev/ttyACM1::INSTR')                       

In [5]: psu                                                                     
Out[5]: <'Resource'('ASRL/dev/ttyACM1::INSTR')>

In [6]: psu.query('*IDN?')                                                      
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-6-51bfc98901b4> in <module>
----> 1 psu.query('*IDN?')

AttributeError: 'Resource' object has no attribute 'query'
`

The expected output would have been

`In [5]: psu
Out[5]: <'SerialInstrument'('ASRL/dev/ttyACM1::INSTR')>
`

This wrong behaviour could be reproduced in Fedora 32, CentOS 7 and Raspbian 10 for the combination of pyvisa 1.11.0 and pyvisa-py 0.5.0. We could not reproduce it with pyvisa 1.10.1 or 1.9.1 and with pyvisa-py 0.3.1 or 0.4.1

# Instrument Details

AIM-TTIi MX180TP via USB

# Output of pyvisa-info
## Raspbian

Machine Details:
   Platform ID:    Linux-5.4.65-v7+-armv7l-with-debian-10.4
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.7.3
   Compiler:       GCC 8.3.0
   Bits:           32bit
   Build:          Jul 25 2020 13:03:44 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.0

Backends:
   ivi:
      Version: 1.11.0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.0
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.1.0). Backend: libusb1
      USB RAW: Available via PyUSB (1.1.0). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'

## Fedora
Machine Details:
   Platform ID:    Linux-5.8.6-201.fc32.x86_64-x86_64-with-glibc2.2.5
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.8.5
   Compiler:       GCC 10.2.1 20200723 (Red Hat 10.2.1-1)
   Bits:           64bit
   Build:          Aug 12 2020 00:00:00 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.0

Backends:
   ivi:
      Version: 1.11.0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.0
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.1.0). Backend: libusb1
      USB RAW: Available via PyUSB (1.1.0). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'

MatthieuDartiailh, 2020-09-29:
Thanks for your detailed reports. Those issues have actually already been identified and fixes will be included  in the bugfix releases of pyvisa and pyvisa-py that are expected to be published before the end of the week. For pyvisa the PR has already been merged for pyvisa-py it will be merged shortly.

MatthieuDartiailh, 2020-09-29:
FYI:
PyVISA PR: https://github.com/pyvisa/pyvisa/pull/547
PyVISA-py PR: https://github.com/pyvisa/pyvisa-py/pull/277

BDuons, 2020-10-01:
Just tested the latest version (1.11.1pyvisa 0.5.1pyvisa-py), serial is working on windows.

MatthieuDartiailh, 2020-10-01:
Thanks @BDuons

closed 2020-10-01 by MatthieuDartiailh