pyvisa/pyvisa #553: Incompatibility with pyvisa-sim since release 1.11

Arkh42, 2020-10-06:
I work on a project built on top of PyVISA (see [rtestbench](https://github.com/Arkh42/rtestbench)).
For automated testing, I use pyvisa-sim >= 0.3.
My automated tests suddenly started to fail (requirements: pyvisa >= 1.9), until I limited the version of pyvisa to <= 1.10.1.

I saw in your documentation on PyPI:

>  __1.11 (16-09-2020)__
>
> PyVISA 1.11 introduces a small backward incompatibility in the handling of the arguments to handlers for VISA events.

However, I have no clue on how to fix this issue.

To Reproduce
--------------
Steps to reproduce the behavior:
```python
import pyvisa as visa
manager = visa.ResourceManager(visa_library='@sim')
```

**Output of `pyvisa-info`**
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
/opt/hostedtoolcache/Python/3.8.5/x64/lib/python3.8/site-packages/pyvisa/highlevel.py:3015: in __new__
    visa_library = open_visa_library(visa_library)
/opt/hostedtoolcache/Python/3.8.5/x64/lib/python3.8/site-packages/pyvisa/highlevel.py:2929: in open_visa_library
    return cls(argument)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

cls = <class 'pyvisa-sim.highlevel.SimVisaLibrary'>, library_path = ''

    def __new__(
        cls: Type[T], library_path: Union[str, LibraryPath] = ""
    ) -> "VisaLibraryBase":
        """Create a new VISA library from the specified path.
    
        If a library was already created using the same path and class this
        library object is returned instead.
    
        Parameters
        ----------
        library_path : str | LibraryPath
            Path to the VISA library to use in the backend.
    
        Raises
        ------
        OSError
            Raised if the VISA library object could not be created.
    
        """
        if library_path == "":
            errs = []
            for path in cls.get_library_paths():
                try:
                    return cls(path)
                except OSError as e:
                    logger.debug("Could not open VISA library %s: %s", path, str(e))
                    errs.append(str(e))
                except Exception as e:
                    errs.append(str(e))
            else:
                raise OSError("Could not open VISA library:\n" + "\n".join(errs))
E               OSError: Could not open VISA library:

/opt/hostedtoolcache/Python/3.8.5/x64/lib/python3.8/site-packages/pyvisa/highlevel.py:175: OSError

MatthieuDartiailh, 2020-10-06:
The issue has been reported under Pyvisa-sim and is actually something I overlooked because I do not actively maintain pyvisa-sim (I am not a fan of the chosen approach and my time for open source is finite). I will do my best to release a new release either tomorrow or next week depending on how much time I can find.

MatthieuDartiailh, 2020-10-08:
https://github.com/pyvisa/pyvisa-sim/pull/53

MatthieuDartiailh, 2020-10-15:
The latest release of PyVISA-sim 0.4.0 supports PyVISA >= 1.11

closed 2020-10-15 by MatthieuDartiailh