pyvisa/pyvisa #554: OSError: Could not open VISA library risen after upgrade from 1.10.1 to 1.11.1

bgarbin, 2020-10-13:
I updated pyvisa using pip (pip install pyvisa --upgrade) and I now get the error:
OSError: Could not open VISA librar

To Reproduce
--------------
import pyvisa as p
v = p.ResourceManager()

**Output of `pyvisa-info`**
SOMEPATH/highlevel.py:2844: FutureWarning: Backends packages should use an _ rather than a - .Project can/should keep using a - (like pytest plugins).Support for backends with - will be removed in 1.12
  FutureWarning,
Machine Details:
   Platform ID:    Linux-4.19.0-11-amd64-x86_64-with-debian-10.6
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.7.3
   Compiler:       GCC 8.3.0
   Bits:           64bit
   Build:          Jul 25 2020 13:03:44 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.1

Backends:
   ivi:
      Version: 1.11.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'

MatthieuDartiailh, 2020-10-13:
You should also upgrade pyvisa-py. The latest version of each work well together.

bgarbin, 2020-10-14:
Many thanks. It does indeed work fine. 
Wouldn't it make sense to add pyvisa-py to the requirements at package installation/upgrade?

MatthieuDartiailh, 2020-10-14:
You do not need pyvisa-py to use pyvisa (if you install a VISA implementation, NI, Keysight, etc) so such a constraint would be odd. Also I usually do not know if the next release will break so stipulating an upper bound could also be problematic. I should follow semver more accurately and use upper bounds, but on a project moving so slowly I am always shy of tagging a version 2. Sorry for the inconvenience.

closed 2020-10-14 by MatthieuDartiailh