pyvisa/pyvisa #555: hantek dso4084b - read waveform problem

KKolin16, 2020-10-20:
Hello,

I try to catch waveform signal from oscilloscope with command:  

`oscilloscope.query(':WAVeform:DATA:ALL?')`

but what i receive is string:

`'#90000040290000040990000000994434334334444344333333333433443444334343343334433334333443443344443444333433433334333333433444443434343343333444444434444333334433344344443344435344344334334434434443443444443334343433434444\x02\x02\x02\x02\x01\x02\x01\x01\x00\x02\x02\x00\x02\x02\x01\x01\x00\x02\x00\x01\x00\x00\x00\x01\x00\x01\x02\x02\x01\x01\x00\x01\x01\x01\x01\x00\x00\x01\x00\x02\x01\x00\x01\x01\x01\x00\x01\x00\x01\x01\x00\x01\x01\x01\x00\x01\x01\x01\x02\x00\x02\x01\x00\x00\x01\x01\x01\x00\x01\x00\x01\x00\x00\x00\x01\x01\x01\x01\x01\x01\x02\x00\x01\x01\x01\x01\x00\x00\x02\x00\x00\x01\x02\x01\x01\x01\x00\x01\x00\x01\x02\x00\x01\x01\x01ÿ\x01\x01\x00\x01\x00\x01\x01\x02\x01\x01\x01\x01\x01\x01\x00\x02\x00\x01\x00\x01\x02\x01\x01\x00\x01\x01\x00\x01\x02\x01\x01\x00\x00\x01\x02\x00\x01\x00\x02\x02\x01\x00\x01\x00\x02\x01\x00\x01\x01\x02\x00\x01\x01ÿ\x01\x00\x02\x02\x00\x01\x01\x01\x02\x00\x02\x00\x02\x00\x01\x0 ... `

I notice periodicity  in this chain because it should be a square wave, but how to get proper values.

I tried this commands:

`data2 = oscilloscope.query_binary_values(':WAVeform:DATA:ALL?', datatype="b", chunk_size = 102400)`

`oscilloscope.query_binary_values(':WAVeform:DATA:ALL?', datatype="b", delay = 2)`

`oscilloscope.query_binary_values(':WAVeform:DATA:ALL?')`

but the only result I get is 

`VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.`

Did I ommit something? How those data should be transer in useful representation?

Regards

------------------
* Model: HANTEK DSO4084B
* Communication: USB
* Link to the documentation (if available):

Output of `pyvisa-info`
-----------------------

MatthieuDartiailh, 2020-10-22:
Your instrument is transmitting data in binary format, you should be able to find in your instrument manual what kind of number are transferred, which will allow you to set the datatype properly. But first you need to get all the data.

Since you can recover at least part of the data, I would suggest first to increase the timeout (your call to query is likely to return only a fraction of the data). Based on the header you expect 4029 bytes in addition to the 11 bytes of the header (#9XXXXXXXXX), you can try using `read_bytes` to read exactly that amount of data and then read one extra bytes, if the extra read fails, you should pass `expect_termination=False` to `query_binary_values`. You can also have a look at the following section of the documentation https://pyvisa.readthedocs.io/en/latest/introduction/rvalues.html

KKolin16, 2020-10-26:
Thank You for your answer . I think I receive all data. For example for step function I receive: `'#9000004029000004099000000099ÒÒÒÑÒÒÑÒÒÒÒÒÑÒÒÒÐÑÒÒÒÒÑÒÒÒÒÒÒÒÒÒÒÒÒÒÒÒÑÒÒÒÒÑÑÒÒÒÒÒÒÒÑÒÑÒÒÐÒÒÒÒÒÑÒÒÑÓÑÒÑÒÒÒÒÒÒÒÒÒÒÒÑÒÒÒÒÑÑÒÒÒÒÑÐÒÒÑÒÐÒÒÑÒÒÑÒÑÐÒÒÒÒÒÑÒÒÑÐÑÒÒÑÐÒÒÒÑÒÒÒÒÒÒÒÒÒÒÒÑÒÑÒÒÒÑÑÑÒÒÒÒÒÒÒÒÑÑÒÑÑÒÒÑÐÑÑÐÒÐÐÑÒÒÒÒÑÒÒÒÒÒÐÒÒÒÒÑÑÒÑÑÒÒÐÒÑÒÑÒÒÐÑÑÒÑÑÐÑÑÑÒÒÑÑÒÐÒÒÒÐÑÒÑÑÒÒÒÑÒÒÒÒÐÑÑÒÐÑÑÑÑÒÑÒÐÑÐÑÑÒÑÐÑÒÑÒÑÑÑÒÐÒÐÑÑÐÑÒÒÒÒÐÒÒÒÒÑÑÑÑÐÒÑÐÒÑÑÐÒÑÑÑÑÒÑÐÑÐÑÑÑÒÐÑÑÑÒÑÑÑÒÐÑÐÒÑÐÐÑÒÑÐÐÑÑÑÐÑÒÑÑÒÑÑÐÑÒÑÒÑÑÑÒÑÑÑÐÑÑÑÑÑÑÐÑÑÑÑÑÐÑÑÐÑÐÒÑÐÐÒÒÑÒÐÑÑÑÒÒÑÑÑÐÑÑÐÑÑÑÐÑÑÑÒÒÑÑÒÒÐÑÑÑÑÐÑÑÑÑÑÐÑÒÐÑÐÑÑÐÐÑÐÑÐÑÑÑÑÐÑÑÑÑÐÒÑÐÑÑÑÐÑÑÑÒÑÑÑÐÑÐÑÑÑÑÐÑÑÑÑÑÐÑÑÐÑÐÑÑÒÐÑÐÑÐÐÑÑÑÐÑÑÑÑÐÑÑÐÑÑÑÐÑÑÑÑÑÑÒÐÑÐÑÑÑÑÐÑÑÑÑÑÐÑÑÐÑÐÑÑÐÐÑÐÑÐÐÒÒÑÐÑÑÑÑÐÑÑÐÑÑÑÐÑÒÑÑÑÑÑÐÑÐÑÑÑÑÐÑÑÑÑÑÐÑÑÐÑÐÑÑÐÐÑÐÑÐÐÑÑÑÐÑÑÑÒÐÑÑÒÐÑÑÐÑÑÑÑÑÑÑÐÑÐÑÑÑÑÐÑÑÑÑÑÐÑÑÐÑÐÐÑÐÐÑÐÑÐÐÑÑÑÐÑÑÑÑÐÑÑÐÑÑÑÐÑÑÑÑÑÑÑÐÑÐÑÑÐÑÐÑÐÑÑÑÐÑÑÐÑÐÐÑÐÐÑÐÑÐÐÑÑÑÐÑÑÑÑÐÑÑÐÑÑÑÐÑÑÑÑÑÑÑÐÑÐÑÑÑÑÐÑÑÑÐÑÏÑÑÐÑÐÑÑÐÐÑÐÐÐÐÐÑÑÐÑÑÑÐÐÑÑÐÑÐÑÐÑÑÑÑÑÑÑÐÑÐÑÑÑÑÐÑÑÑÑÑÐÏÑÐÑÐÑÑÐÐÑÒÑÐÐÐÑÑÏÑÑÑÐÐÑÑÐÑÐÑÐÑÑÑÑÐÑÐÐÏÐÑÐÑÑÐÑÑÏÑÑÏÑÑÐÑÐÑÐÏÏÑÐÑÐÐÑÑÏÐÑÑÑÑÐÐÑÐÑÑÑÐÑÑÑÑÑÐÐÐÑÏÑÑÑÑÐÐÑÑÑÑÐÏÑÐÑÐÐÐÏÐÏÐÑÐÐÑÐÏÐÐÑÑÐÐÑÏÐÐÐÑÐÐÑÑÑÑÑÑÏÑÏÑÐÑÑÐÐÑÑÑÑÐÑÑÐÑÐÑÑÐÐÏÐÐÐÏÑÐÑÏÐÑÑÐÐÑÏÐÐÑÑÐÐÑÐÑÐÑÑÏÑÐÐÑÑÏÐÑÑÑÑÐÏÑÑÏÏÐÐÐÐÐÏÐÑÐÐÐÐÑÏÑÑÐÑÐÑÏÏÑÐÑÐÑÑÐÑÑÐÑÏÏÐÐÐÑÏÏÐÑÑÐÐÏÏÐÐÏÏÐÑÏÏÑÏÐÏÐÑÑÑÏÐÑÑÑÏÑÏÐÐÑÑÐÐÑÐÐÐÐÐÏÏÏÑÑÑÏÏÑÐÏÐÐÐÏÐÏÑÐÑÐÏÏÑÐÐÏÐÐÐÏÏÑÑÐÐÏÑÏÐÑÑÐÐÑÐÐÑÑÑÐÏÏÏÑÐÐÏÏÐÑÏÐÐÏÏÑÐÏÏÐÑÏÏÏÏÐÏÏÐÐÏÏÐÏÐÐÐÑÏÏÐÐÑÐÑÐÑÑÐÐÐÏÏÏÐÐÑÑÐÑÐÑÐÐÏÏÐÏÑÐÐÐÏÏÏÏÐÏÏÐÐÑÏÐÑÑÐÐÐÏÏÐÐÐÏÐÐÑÐÐÐÐÏÑÏÐÐÐÏÏÐÑÏÐÐÏÏÑÏÏÏÐÑÏÏÏÏÐÏÏÐÐÑÐÐÏÐÐÏÐÏÏÐÐÐÏÐÑÐÐÐÐÐÏÏÏÐÑÐÏÏÑÐÏÐÐÏÏÐÏÏÏÐÐÏÏÏÏÐÐÐÐÐÑÏÐÑÐÐÏÐÑÐÐÐÐÏÑÐÐÐÑÐÐÏÏÐÐÐÐÑÏÐÐÏÑÐÏÏÐÏÏÏÑÐÏÏÏÎÐÏÏÐÐÏÏÑÏÐÑÏÐÏÏÐÐÐÏÐÐÐÐÐÐÐÏÑÏÐÐÐÑÏÐÐÑÐÐÏÏÐÏÏÏÐÐÏÏÏÏÐÏÏÐÐÑÏÑÏÐÐÏÐÏÏÐÐÑÏÐÐÐÐÐÐÐÏÏÐÐÐÐÏÏÐÐÏÐÐÏÏÐÏÏÎÐÐÐÏÏÏÐÏÏÐÐÏÏÐÏÐÐÏÐÏÏÐÐÐÏÐÐÐÐÐÑÑÏÏÏÐÑÐÏÏÐÐÏÐÐÏÏÐÏÏÏÐÐÏÏÏÏÐÏÏÐÐÑÏÑÏÐÐÏÐÏÏÑÐÐÏÐÐÐÐÑÐÐÏÏÏÐÐÎÑÏÐÐÏÐÐÏÏÐÏÏÏÐÐÏÏÏÏÐÏÏÐÐÏÏÐÏÐÐÏÐÏÐÐÑÐÏÐÐÐÐÐÐÐÏÎÏÐÐÐÏÏÐÐÏÐÐÏÑÐÏÏÏÐÐÏÏÏÏÐÎÏÐÐÏÏÐÏÐÏÏÐÏÏÐÐÐÏÐÏÐÐÐÐÐÏÏÏÐÐÐÏÏÐÐÏÐÐÏÏÐÏÏÏÐÐÏÏÏÏÐÏÏÐÐÏÏÐÏÐÐÏÐÏÏÐÐÐÏÐÐÐÐÐÐÐÏÏÏÐÐÐÏÏÎÐÏÐÐÏÏÐÏÏÏÎÐÏÏÏÏÐÏÏÐÐÏÏÐÏÐÐÏÐÏÏÐÐÐÏÐÐÐÏÐÐÐÎÏÏÐÐÐÏÏÐÐÏÐÐÏÏÐÎÏÏÐÐÏÏÏÏÐÎÏÐÐÏÏÐÏÐÐÏÐÏÏÐÎÐÏÐÐÐÐÐÐÐÐÏÏÐÐÐÏÏÐÐÏÐÐÏÏÐÏÏÏÐÐÏÏÏÏÐÏÏÐÐÏÏÐÏÐÐÏÐÏÏÐÐÐÏÐÐÐÐÐÐÐÏÏÏÐÐÐÏÏÎÐÏÏÐÏÎÐÏÎÏÐÐÏÏÏÏÐÎÏÐÐÏÏÐÏÐÐÏÐÏÏÐÐÐÏÐÐÏÐÐÐÐÏÏÏÐÐÐÏÏÐÐÏÐÐÏÏÐÏÎÏÐÎÏÏÏÏÐÏÏÐÐÏÏÐÏÐÐÏÏÏÏÐÐÐÎÐÐÐÐÐÐÐÏÏÏÐÐÐÏÏÐÐÏÐÐÏÏÐÏÏÏÐÎÏÏÏÏÐÏÏÐÐÏÏÐÏÐÐÏÐÎÏÐÐÐÏÐÐÐÐÏÐÐÏÏÎÎÐÐÏÏÐÐÏÐÐÎÏÐÏÏÏÐÐÏÏÏÏÐÏÏÐÐÏÏÐÎÏÐÏÐÏÏÐÐÐÏÐÐÐÐÐÐÐÏÏÏÐÐÐÏÏÐÐÎÐÐÏÏÐÏÏÏÐÐÏÏÏÏÐÎÏÐÐÏÏÐÏÐ\x1b1323222222223222222222212221222221112222122222122111221121221122222122122112212222211221121222121112121121221112221122212221122221212212222212222122222123222122222222121221122222222222122222222222122222122222222122222222222222222222222222212222222222222322222222222222222222222222222322222222222222222222223222222222222222222222222222222222222222222222222222222222222222222222222222222222332223222222222223222222222222222222232223222222233222232222222223223322222223222222223232322332232232222222223222323222332223332323222232222223332222323233322332223233333222332323222323222333222233333223333332233323222333333232333333332223233333333333332322232333333233233333333233333233333333333332332323332323333333223323333333333333323333333333333332333333233333332333333333233333333333333333333333333333333333333333333333343333333233333333333333333333333333333333333333333334333333333333333333333333333333323333333333233334333433333433333333333333333333333333333333333333233343343333333433333333333433333433333433333333333333343333333433333333343334343334333333333333333333343333333434333333333334343333333333333334333333443333343333333433343334333334343334333333333433334333343334333433333344333433343344343444333434333333344343344333443334443344443433344334434434334343333334343434434444343334334334434334434444343434344434333334333444344334333443344434344443343444443434344434443433344434334434343444433434333434443434344444434333334444344333344434444444443334444344444433344434444444333434343444343444443334444444444344444444444444343444333344433444444444344444344434444344434444344444443434444434444443433444344444444433444444444444444444434444444444444344444444444444444434344444444444444434444444444444444444344444444444443444444444444444444444444444444444444444444444344444444444444445444444434444444444445444444444444444444444444444444444444444344444444444444444444444454434534444444444444444444464444444434544444444444444444444444'` But as I wrote before I don't know how to obtain readable values .
and documentation from hantek says:

_WAVeform commands
The list of commands：
 WAVeform:DATA:ALL?
WAVeform:DATA:ALL?
Syntax :WAVeform:DATA:ALL?
Description : Gets any stored data
Return : Returns the waveform packet containing the data header in
the form of a string
remark : For the first time, this command is parsed to data[x]
data[0]-data[1] (2 bits): #9
data[2]-data[10](9 bits): Represents the byte length of the
current packet
data[11]-data[19](9 bits): The total length of bytes
representing the amount of data
data[20]-data[28](9 bits): Represents the byte length of the
data that has been uploaded
data[29](1 bits)：Represents the current running state
data[30](1 bits)：Represents the state of the trigger
data[31]-data[34]（ 4 bits） : bias of ch1
data[35]-data[38]（ 4 bits） : bias of ch2
data[39]-data[42]（ 4 bits） : bias of ch3
data[43]-data[46]（ 4 bits） : bias of ch4
data[47]-data[53]（ 7 bits） : volt of ch1
data[54]-data[60]（ 7 bits） : volt of ch2
data[61]-data[67]（ 7 bits） : volt of ch3
data[68]-data[74]（ 7 bits） : volt of ch4
data[75]-data[78]（ 4 bits） : Represents the enabling of the
channel [1-4]
data[79]-data[87]（ 9 bits） : Indicated sampling rate
data[88]-data[93]（ 6 bits） : Sampling multiple
data[94]-data[102]（ 9 bits）: The current frame displays
trigger time
[... cut ...]
starting point of the data start point
data[112]-data[127]（16 bits）: Reserved bit
59
The data of waveform data[x] was resolved before the data
was read again
data[0]-data[1] (2 bits) : #9
data[2]-data[10](9 bits): Represents the byte length of the
current packet
data[11]-data[19](9bits): The total length of bytes
representing the amount of data
data[20]-data[28](9 bits): Represents the byte length of the
data that has been uploaded
data[29]-data[x]: Represents the waveform data corresponding to
the current data header_

KKolin16, 2020-10-27:
And below is the respond after sending command  `oscilloscope.query_binary_values(':WAVeform:DATA:ALL?')`

---------------------------------------------------------------------------
VisaIOError                               Traceback (most recent call last)
<ipython-input-496-3f0c930fc3f9> in <module>
----> 1 oscilloscope.query_binary_values(':WAVeform:DATA:ALL?')

C:\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py in query_binary_values(self, message, datatype, is_big_endian, container, delay, header_fmt, expect_termination, data_points, chunk_size)
    711             Container type to use for the output data. Possible values are: list,
    712             tuple, np.ndarray, etc, Default to list.
--> 713         delay : Optional[float], optional
    714             Delay in seconds between write and read operations. If None,
    715             defaults to self.query_delay.

C:\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py in read_binary_values(self, datatype, is_big_endian, container, header_fmt, expect_termination, data_points, chunk_size)
    517             If a str is given, data.split(separator) is used. Default to ",".
    518         container : Union[Type, Callable[[Iterable], Sequence]], optional
--> 519             Container type to use for the output data. Possible values are: list,
    520             tuple, np.ndarray, etc, Default to list.
    521 

C:\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py in read_bytes(self, count, chunk_size, break_on_termchar)
    354         success = constants.StatusCode.success
    355         termchar_read = constants.StatusCode.success_termination_character_read
--> 356 
    357         with self.ignore_warning(
    358             constants.StatusCode.success_device_not_present,

C:\Anaconda3\lib\site-packages\pyvisa\ctwrapper\functions.py in read(library, session, count)
   1582 
   1583 def move_in_32(library, session, space, offset, length, extended=False):
-> 1584     """Moves an 32-bit block of data to local memory.
   1585 
   1586     Corresponds to viMoveIn32* functions of the VISA library.

C:\Anaconda3\lib\site-packages\pyvisa\ctwrapper\highlevel.py in _return_handler(self, ret_value, func, arguments)
    191         # We store the error code per session
    192         session = None
--> 193         if func.__name__ not in ("viFindNext",):
    194             try:
    195                 session = arguments[0]

VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

MatthieuDartiailh, 2020-10-27:
You situation is more complex than I first assumed, if you can provide me with a link to the manual of the instrument it could help me provide you with a more concrete solutiuon. It is in particular unclear at this point how to know when you get the first kind of answer describe and when you get the second. Also the overall structure is more complex than for most instruments so `read_binary_values` won't work.

The first kind of response, which is what you apparently get (based on the length of the answer, see below) contains information about the configuration of the scope. I am however a bit unsure how to interpret them since 7 bytes cannot be decoded neither as a integer or as a float (an int64 is encoded on 8 bytes). Maybe you are just expected to pad those with a null byte but I wish the documentation was clearer. To convert bytes to number you can have a look at the `struct` module from the python standard library. Also note that since the message is using raw bytes you should use `read_raw` for reading the data in order to obtain bytes.

I think you get the first kind of answer because the system transmit 99 bytes after the header #9XXXXXXXXXYYYYYYYYY000000099, while the X portion says the packet is 4029 bytes long, the Y that the total data is 4099 bytes long but only 99 bytes have been uploaded which is the length of the data described in the first kind of answer. Once again having the full manual may help make thinks clearer.

KKolin16, 2020-10-28:
To be precise, first when I send command `oscilloscope.query(':WAVeform:DATA:ALL?')'
I receive:
`'#900000012800000409900000000010-0480000000000002.5e-3194.9e-3184.9e-3184.9e-31810005.000e+06000001+0.00e+00+0.00e+0000010759721'`
after I send second time the same command `oscilloscope.query(':WAVeform:DATA:ALL?')'
I receive:
`'#9000004029000004099000000099ÒÑÒÑÒÒÐÒÒÑÒÐÒÒÒÒÒÒÑÒÒÑÑÒÒÒÒÒÒÑÑÒÒÒÒÒÐÒÑÑÑÒÒÒÒÑÒÑÒÑÒÑÒÒÑÐÒÒÒÐÑÒÒÒÑÒÒÒÒÐÒÒÒÒÑÒÒÑÑÒÑÑÒÑÒÒÑÐÒÑÒÑÑÒÒÐÒÒÑÑÒÐÒÑÐÑÐÑÐÑÐÑÑÐÒÑÒÒÒÑÐÑÒÑÑÒÑÑÒÑÒÑÑÑÑÐÑÑÒÑÑÒÐÑÑÑÐÑÒÑÒÑÒÑÒÐÒÐÑÐÑÐÒÒÐÐÒÐÒÑÑÐÑÒÒÑÒÑÒÑÒÐÑÒÑÒÐÑÑÐÑÑÑÐÑÑÑÒÑÑÒÑÑÐÑÑÐÑÐÑÐÑÐÑÑÐÐÑÐÑÑÑÐÑÑÑÑÑÑÑÑÑÐÒÑÒÑÐÒÑÐÑÑÑÐÑÑÑÐÑÑÑÑÑÐÑÑÐÑÐÑÐÑÐÑÑÐÐÑÐÑÑÑÑÑÑÑÑÑÑÑÑÑÐÑÑÑÑÐÑÑÐÑÑÑÐÑÑÑÐÑÑÑÑÑÐÑÑÐÑÐÑÐÑÐÑÑÐÐÑÐÑÑÑÐÑÑÑÑÑÑÏÑÐÐÑÑÑÑÐÑÑÐÑÏÑÐÑÑÑÐÐÑÑÑÑÐÑÑÐÑÐÑÐÑÐÑÑÐÐÑÐÑÑÑÐÑÑÑÑÑÑÑÐÑÐÑÑÑÑÐÑÑÐÑÑÑÐÑÑÑÏÑÑÑÐÐÐÑÑÐÑÐÐÐÑÐÑÑÐÐÑÐÑÐÑÐÑÑÑÑÑÑÑÑÑÐÑÑÑÑÏÑÑÐÑÑÐÐÏÑÑÐÑÑÑÑÑÐÏÑÐÑÏÑÏÏÐÑÑÐÐÑÐÑÑÑÐÑÑÑÑÑÑÑÐÑÐÑÑÏÐÐÐÑÐÑÑÐÐÑÑÑÏÑÑÑÑÑÐÏÑÐÐÐÑÏÑÐÑÑÐÐÑÏÐÑÏÐÐÑÑÑÐÑÑÑÑÐÐÑÏÑÐÐÑÐÑÑÑÏÏÑÑÐÑÏÐÑÐÐÑÑÏÑÐÑÏÑÏÑÑÏÐÐÏÐÐÏÐÑÐÑÐÐÑÑÑÑÏÐÐÏÑÐÑÏÏÑÏÑÐÏÐÑÐÐÑÐÑÑÐÏÐÏÐÐÐÐÑÐÐÑÏÏÑÏÑÑÑÐÐÑÐÑÑÑÏÐÐÏÐÑÏÐÐÐÏÏÐÏÐÏÑÑÑÏÑÑÑÑÑÐÏÐÐÑÏÑÐÑÏÐÏÏÏÐÏÐÑÏÏÑÐÑÐÐÑÏÐÑÐÐÐÑÐÐÑÏÐÑÏÐÏÑÐÐÐÐÏÑÑÑÐÏÑÏÐÏÐÏÏÏÐÏÏÏÐÏÐÑÏÏÑÐÐÐÐÑÏÐÐÏÐÑÏÑÏÐÏÐÐÏÑÏÏÐÐÏÐÏÐÐÑÏÏÑÏÑÏÐÏÏÏÑÏÐÏÑÏÐÐÏÏÐÐÑÐÐÐÏÐÐÏÐÐÏÐÏÐÏÏÐÏÐÏÏÐÐÏÐÏÐÑÐÏÏÐÏÐÏÐÐÑÏÐÏÐÏÐÏÐÐÏÏÐÐÐÐÐÐÏÐÐÏÐÐÏÐÏÐÏÏÐÏÐÏÏÐÐÏÐÏÐÐÐÏÑÐÏÐÏÐÐÏÏÐÏÏÏÐÏÐÐÏÐÐÐÐÑÐÐÏÐÐÏÐÐÏÑÏÐÏÏÐÏÐÏÏÐÐÏÐÏÐÐÑÐÏÐÏÐÏÏÏÏÏÐÏÏÏÐÏÐÐÏÏÐÐÐÐÐÐÏÐÐÏÐÐÏÐÏÑÏÏÐÏÐÏÏÐÐÏÐÏÐÐÐÐÏÐÏÐÏÐÏÏÏÐÏÐÏÐÏÐÐÏÏÐÐÐÐÐÐÏÐÐÏÐÐÏÐÏÐÏÏÐÏÐÏÏÐÐÏÐÏÐÐÏÏÑÐÏÐÏÐÏÏÏÐÏÏÏÐÏÐÐÏÏÐÐÐÐÐÐÏÏÐÏÐÐÏÐÏÐÏÏÐÏÐÏÏÐÐÏÐÏÐÐÐÏÏÐÐÐÏÐÏÏÏÐÏÏÏÐÏÐÐÏÏÐÐÐÐÐÐÏÐÐÏÐÎÏÐÏÐÏÏÐÏÐÏÏÐÐÏÐÏÐÐÐÏÏÐÏÐÏÐÏÏÏÐÏÏÏÐÎÐÐÏÏÐÐÐÐÐÐÏÐÐÏÐÐÏÐÏÐÏÏÐÏÐÏÏÐÐÏÐÏÐÐÏÏÏÐÏÐÎÐÏÏÏÐÏÏÏÐÎÐÐÏÏÐÐÐÐÐÎÏÏÏÏÐÎÏÐÏÐÏÏÐÏÏÏÏÐÐÏÐÏÐÐÐÏÎÐÏÐÏÐÏÏÏÐÏÏÎÐÏÎÐÎÏÐÐÐÐÐÐÏÐÐÏÐÐÏÐÏÐÏÏÐÎÐÏÏÏÐÏÐÏÐÐÏÏÏÐÏÐÏÐÏÏÏÏÏÏÎÐÏÐÐÎÎÐÐÐÎÎÐÏÐÐÏÎÐÏÐÏÐÏÏÏÏÐÎÏÐÐÏÐÏÐÏÐÏÏÐÎÐÏÏÎÎÏÐÏÏÏÐÏÐÐÏÏÐÐÏÐÐÐÏÏÐÏÎÐÏÐÏÐÏÏÐÏÏÏÎÐÐÏÐÏÏÏÐÏÏÐÏÐÎÐÎÏÏÐÏÏÏÐÏÐÎÏÏÐÐÐÐÐÐÏÏÏÏÐÎÎÐÏÐÎÎÐÏÐÏÏÐÐÏÐÏÏÐÐÏÎÐÎÐÏÏÏÏÏÏÏÎÎÐÎÐÐÏÏÎÐÐÐÎÐÎÏÐÎÎÎÏÐÏÎÏÏÐÏÐÏÑÐÐÏÐÏÏÐÏÏÏÐÏÐÎÐÎÎÏÐÎÏÎÐÏÐÎÎÏÎÐÐÐÐÐÎÏÐÏÎÐÎÏÏÐÎÏÏÏÐÏÎÏÐÎÎÎÐÐÐÏÎÏÏÐÎÏÏÏÏÏÏÎÏÐÎÎÎÎÎÎÐÏÐÎÐÎÏÐÏÐÐÎÏÏÎÏÏÏÎÐÎÏÏÐÎÎÏÐÐÐÏÎÐÎÎÏÐÏÎÏÏÎÏÏÐÏÎÐÎÏÎÐÐÐÎÐÏÐÏÏÐÎÎÐÏÐÏÏÐÏÐÏÏÐÏÎÐÏÏÐÏÎÎÐÎÐÏÐÎÎÏÏÏÏÎÐÏÎÐÏÎÐÎÏÐÎÐÏÏÏÏÎÐÏÐÏÎÎÏÏÎÐÎÏÏÏÎÐÏÐÐÏÎÎÐÏÐÏÏÏÎÏÏÏÏÎÐÎÎÎÏÎÎÎÐÐÎÎÏÏÐÏÐÎÏÏÏÎÎÎÐÎÐÎÏÏÏÎÐÎÐÏÐÎÎÏÏÐÏÏÎÏÏÏÏÎÎÐÎÎÎÏÎÐÎÏÎÎÎÎÏÐÏÐÎÎÐÏÐÎÎÐÏÐÎÎÐÐÏÐÏÐÏÏÎÎÏÏÎÎÏÎÏÎÐÎÏÎÏÎÎÎÎÏÎÐÐÐÎÎÎÐÐÏÎÎÎÏÎÎÏÏÐÎÏÎÎÐÏÎÐÏÏÐÏÎÏÏÎÎÏÐÏÏÎÐÎÎÎÏÎÎÎÎÎÎÐÏÐÎÎÎÐÏÎÎÎÎÐÎÐÏÏÏÎÏÎÎÏÐÎÎÏÏÐÐÎÎÏÎÐÏÏÎÎÏÏÎÎÎÐÎÎÎÏÏÎÎÏÎÐÎÏÐÏÎÎÐÎÐÏÐÎÎÏÎÏÎÏÏÐÎÐÏÐÏÏÎÏÐÏÎÎÐÏÎÎÏÎÏÎÏÎÎÐÏÎÎÎÏÐÎÎÎÐÏÏÎÐÎÏÎÐÎÎÐÏÐÎÏÏÐÎÎÏÏÐÏÎÏÏÎÎÎÐÎÎÎÐÏÎÎÏÏÎÎÎÎÎÎÐÐÐÐÎÏÏÎÎÎÏÐÎÎÎÎÏÎÏÏÎÏÐÏÎÎÏÐÐÎÎÏÎÎÏÏÎÎÏÏÎÎÎÐÏÐÎÎÏÎÎÏÐÎÎ,00100100000010/0000000000/0000000.000000000000000/0/000000000.0/0/00000000000/000000./00000/0/000000000/001000000000/00/00000000000000000/00000000000000000100000000000/0000000000010000100000000000000000000000010000000010010010110000010000000001000110010101111010010010011000010100010100010110010000111010011110101110011110011100111101101010111110111110111011110010111111110111111110011110111011111111101111101110101111111111110111111111111111111111111111111111110111111111111111112111111111111111111111111111122111111111111111121111111111121111211212122212111111111111111111112111122112121112211212111212121122112112212112111112211121121222121112222121212211122222222212222122222211212112211221122122122212222222222222212222221222111222112221222222222222221222222221222222222122212221222222222222222222222211221222222222222221222222222222222222222222222222122122222122222222222122322222222122222232222222222222222222222222222222222222222222222222222222222222222222222222222222222221222222222222223222222332223322222222222223222222222222222222222222222222222223222222232222222222222222222222222322322322222322222222222222232222222222232222222223222232222223232223232322232232222323322222232322232233222222222223232222322232323223233332333222223233332222222322322222232223232222222323222323333333332232233322223232333322322232333332223332233233233233323222333233322333323333333232323332333332222232233233333232223333333332333232333233232333333333333333333333333333333333333333333233333333233333333233333232333333333333333333333333333233232333323333333333233233233233333333333332232233333233323332233332323233333333333333333333333333333332333333233333333323333333333333333333333333333333332233333333333333333333333233333333322333333333343333333333333333333333333333333333333333333333333333333343333333333333333333333333333333333333333333333333333333333433333333333333333333333333333333433333333333333333333333343333333334333434333333333'`

I was trying to change enconding but I get the same mess, only with different signs, now I am using "latin-1".

The only manuals are available here (in download section): http://www.hantek.com/products/detail/12167

MatthieuDartiailh, 2020-10-28:
First of all thanks for posting everything that helps figuring out what is going on. Here is what should allow you to convert your data:
```python
import struct
header = oscilloscope.query(":WAVeform:DATA:ALL?")
# This part is easy to parse and give you the offset and amplitude of each channel
# Request the instrument to send the real data
oscilloscope.write(":WAVeform:DATA:ALL?")
# Read just the beginning to know what to expect
h = oscilloscope.read_bytes(27)
data_length = int(h[2:11]) - 27 # length of the packet minus what we already read
data_raw = oscilloscope.read_bytes(data_length + 1) # +1 for the termchar
data = struct.unpack(f"<{data_length}b", data_raw)
```
The data you are going to get are encoded between -128 and 127 and need to be scaled by the header you get after the first command. Hopefully I got it right.

KKolin16, 2020-10-30:
Thanks a lot for Your help. After I had done some tests I modified Your code:

`oscilloscope.query(":WAVeform:DATA:ALL?")`
`# This part is easy to parse and give you the offset and amplitude of each channel`
`# Request the instrument to send the real data`
`oscilloscope.write(":WAVeform:DATA:ALL?")`
`# Read just the beginning to know what to expect`
`h = oscilloscope.read_bytes(29)`  **was 27**
`data_length = int(h[2:11])-29 # length of the packet minus what we already read` **was 27**
`data_raw = oscilloscope.read_bytes(data_length_1 )`  ~~+1 for the termchar~~
`data = struct.unpack(f"<{data_length}b", data_raw)`

I had problem to find scale value inside header. I don't know why, it shows strange values: 2.5e-318; 4,9e-318 or 9.9e-318 for active channel. 
 Anyway I exporded data directly from oscilloscope to excel and compared it with data received from pyton and finally figured out proper scale value.

Once again thanks for help

MatthieuDartiailh, 2020-10-30:
Happy you got it to work ! And sorry for the offset in bytes in number I forgot to count the first #9.

closed 2020-10-30 by MatthieuDartiailh