pyvisa/pyvisa #557: allow changing of baudrate in pyvisa-shell

hkennyv, 2020-10-23:
If I'm not mistaken, there doesn't seem to be a way to change the baudrate when using `pyvisa-shell`. I've narrowed it down to this function:

https://github.com/pyvisa/pyvisa/blob/af1f69566afa674ae2a186f1bbf77e477918dbc8/pyvisa/shell.py#L61-L98

I think that we can make this possible by changing line 82 to allow passing in other kwargs (such as baud_rate). I feel like this is not too difficult to do since pyvisa already uses argparse.

Happy to try to contribute this myself if there's interest :). BTW pyvisa-shell is a godsend for quick prototyping and testing, thank you for your work!

MatthieuDartiailh, 2020-10-23:
Technically you can change the baudrate using `attr VI_ATTR_ASRL_BAUD 9600` (and the attr name can be discovered using `attr` which list all values). I agree however that is not really ergonomic. I personally have little use for the shell but I will be happy to review any PR you may submit to improve open, or attr or both. 

For attr supporting the python name would be nice and also having access (and a way to list) the constants values stored in enum. I will be happy to provide some advice as you go if you want to give it a try.

hkennyv, 2020-10-23:
Thanks @MatthieuDartiailh, I wasn't aware you could change the baudrate like that. i feel a sufficient solution would consist of two things: 

1) adding an example/explanation in the [pyvisa shell](https://pyvisa.readthedocs.io/en/latest/introduction/shell.html#) docs (at least making a note that the default baud is 9600 and a demonstration of how to change an attribute)
2) adding support for using the python name

I'm not completely sure what you mean by having access to the constant values stored in enum but i hope once i dive into the code a bit more, it'll be clear

MatthieuDartiailh, 2020-10-24:
For the baudrate it does not apply but many attributes (such as parity) have enumerated values and a way to easily list possible values in the shell would be nice.