pyvisa/pyvisa #559: query_binary_values with datatype='s' no longer returns a bytes array in 1.11.1 (was ok in 1.10.1)

leriak, 2020-10-26:
I have the follwing code to get bitmaps from an Agilent oscilloscope and write to a file 

```python
data = osc.query_binary_values(":DISPlay:DATA? PNG", datatype='s', delay=0.7)
with open(file_name + '.png', "wb") as f:
        f.write(data[0])
```

This code works fine with pyvisa 1.10.1 but today I upgraded to 1.11.1 and `f.write(data[0])` raises an exception saying that `data[0]` is not a bytes array.

After some debugging it turns out that `query_binary_values` no longer returns a `bytes` array but an `int` array, even when `datatype='s'` is passed (it also doesn't work with `datatype='p'`)

I've downgraded back to 1.10.1 and it works fine again.

First time poster on github, sorry if I'm not following standard format.

MatthieuDartiailh, 2020-10-26:
Thanks for posting. 

Technically the `"s"` and `"p"` formats were broken in 1.10.1 since you got a list with a single bytes rather than just bytes. Inarticulate this was causing issues when writing to the instrument (see #532). The issue was addressed in #536 but I could have made the changelog more explicit. As per the expected documentation if you want to use bytes as container you should simply pass `container=bytes` and you will get a bytes object (not wrapped in a list as before).

leriak, 2020-10-30:
Thanks for your answer, I've tried code below calling `query_binary_values` like in the test performed in #536 by ABillBlakely 
In version 1.10.1 but get a `TypeError` exception
I haven't tested in 1.11.1 since right now I can't spare the time to upgrade and then downgrade, sorry.

```python
data = osc.query_binary_values(":DISPlay:DATA? PNG", datatype='s', container=bytes, delay=0.7)
Traceback (most recent call last):

  File "<ipython-input-36-a9e5444cae27>", line 1, in <module>
    data = osc.query_binary_values(":DISPlay:DATA? PNG", datatype='s', container=bytes, delay=0.7)

  File "C:\Users\fonellos\Anaconda3\envs\py38\lib\site-packages\pyvisa\resources\messagebased.py", line 713, in query_binary_values
    return self.read_binary_values(datatype, is_big_endian, container,

  File "C:\Users\fonellos\Anaconda3\envs\py38\lib\site-packages\pyvisa\resources\messagebased.py", line 542, in read_binary_values
    return util.from_binary_block(block, offset, data_length,

  File "C:\Users\fonellos\Anaconda3\envs\py38\lib\site-packages\pyvisa\util.py", line 473, in from_binary_block
    return container(struct.unpack_from(fullfmt, block, offset))

TypeError: 'bytes' object cannot be interpreted as an integer
```

MatthieuDartiailh, 2020-10-30:
Sadly no that will work only in 1.11.0 and  above.