pyvisa/pyvisa #560: Python Can't Find Pyvisa, but only when it is run from the command line

phanixis, 2020-10-29:
I wrote a Python script to operate a Keithley 2450 SourceMeter from a Raspberry Pi over the USB.  I coded this in Thonny Python IDE.  When run within Thonny, the script runs without any problems.  However, when I attempt to run the script from the command line using "sudo Python3 myscript.py", I get the error "No module named 'pyvisa'" when I call "import pyvisa as visa".  No idea what is wrong.  Can anyone help me correct this problem?

Thanks.

MatthieuDartiailh, 2020-10-29:
Thonny may be using a different python interpreter than the system one (which is a good idea to avoid corrupting the system one). Try to look where Thonny store the Python interpreter it uses.

kdkaydai, 2021-04-09:
You can use below command to check your library.
`pip list --local`

And using below python code to check the python version.
```
import sys
import os
version = sys.version
print("Python version: ")
print(str(sys.version))
print("\n")

print("Version info.")
print(str(sys.version_info))
```