pyvisa/pyvisa #561: [COM] Communication issue with MDO3034 using open_resource()

zhixing1996, 2020-10-30:
Description
------------------
Currently, I'm connecting my computer with MDO3034 oscilloscope. Here is the description.
* Setup
    The connection between MDO3034 and my PC is: PC-router-MDO3034, and the IP addresses of PC and oscilloscope are attributed by router automatically.
* Operation
 [1] Open Visual Studio Code
 [2] Enter python
[User] `python`
\>\>\> import visa
\>\>\> rm = visa.ResourceManager()
\>\>\> rm.list_resources()
('TCPIP0::169.254.5.249::inst0::INSTR', 'ASRL1::INSTR', 'ASRL3::INSTR', 'ASRL4::INSTR', 'ASRL5::INSTR', 'TCPIP0::192.168.239.191::inst0::INSTR', 'USB0::0x0699::0x0408::C028075::0::INSTR')
\>\>\> rm.open_resource('TCPIP0::169.254.5.249::inst0::INSTR')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\ProgramData\Anaconda3\lib\site-packages\pyvisa\highlevel.py", line 1655, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\ProgramData\Anaconda3\lib\site-packages\pyvisa\resources\resource.py", line 209, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "C:\ProgramData\Anaconda3\lib\site-packages\pyvisa\highlevel.py", line 1612, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "C:\ProgramData\Anaconda3\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1213, in open
    ret = library.viOpen(session, resource_name, access_mode, open_timeout, byref(out_session))
OSError: exception: access violation reading 0x000000000000100C

I've tried to search for the error reason of 'OSError: exception: access violation reading 0x000000000000100C', but I've failed to find a useful one for pyvisa. Also, I've done a test with NI-MAX, it seems the communication is successful. 

Instrument details
------------------
* Model: MDO3034
* Communication: TCPIP, GPIB: TCPIP
* Link to the documentation (if available): MDO4000-B-MSO-DPO4000B-and-MDO3000-Oscilloscope-Programmer-Manual-Rev-A.pdf

Output of `pyvisa-info`
-----------------------
* Machine Details:
   Platform ID:    Windows-10-10.0.17134-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 9, GenuineIntel

* Python:
   Implementation: CPython
   Executable:     c:\programdata\anaconda3\python.exe
   Version:        3.6.4
   Compiler:       MSC v.1900 64 bit (AMD64)
   Bits:           64bit
   Build:          Jan 16 2018 10:22:32 (#default)
   Unicode:        UCS4

* PyVISA Version: 1.9.0

* Backends:
   ni:
      Version: 1.9.0 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 5243905
         Spec. Version: 5243136
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 5243905
         Spec. Version: 5243136

zhixing1996, 2020-10-30:
Have solved this problem by installing python3.9.0 and NI-VISA 20.0, can't figure out why, could someone explain that? Thanks

MatthieuDartiailh, 2020-10-30:
I do not have a good explanation for your issue no. In general I would advise using a local network with fixed IP addresses since otherwise your instrument address can be randomly reassigned.

MatthieuDartiailh, 2020-10-30:
I closed since you got things ro work feel free to reopen if necessary.

closed 2020-10-30 by MatthieuDartiailh