pyvisa/pyvisa #562: VI_ERROR_NSUP_ATTR when trying to set timeout. Since pyvisa==1.11.1

buster3, 2020-11-09:
Since updating from pyvisa 1.10.1 to 1.11.1 I get an VI_ERROR_NSUP_ATTR error when trying to set a custom timeout. With pyvisa 1.10.1 everything works well.

To Reproduce
--------------
1.10.1, everything works fine
```>pyvisa-shell
c:\users\maierb_t\appdata\local\programs\python\python36-32\lib\site-packages\pyvisa\ctwrapper\functions.py:1227: VisaIOWarning: VI_WARN_CONFIG_NLOADED (1073676407): The specified configuration either does not exist or could not be loaded. VISA-specified defaults will be used.
  ret = library.viOpenDefaultRM(byref(session))

Welcome to the VISA shell. Type help or ? to list commands.

(visa) open TCPIP::10.215.2.23::INSTR
TCPIP::10.215.2.23::INSTR has been opened.
You can talk to the device using "write", "read" or "query".
The default end of message is added to each message.
(open) timeout 5000
Done
(open) exit
```

update to 1.11.1
```
python -m pip install -U pyvisa
Installing collected packages: pyvisa
  Attempting uninstall: pyvisa
    Found existing installation: PyVISA 1.10.1
    Uninstalling PyVISA-1.10.1:
      Successfully uninstalled PyVISA-1.10.1
Successfully installed pyvisa-1.11.1
```

Issue
```
>pyvisa-shell
c:\users\maierb_t\appdata\local\programs\python\python36-32\lib\site-packages\pyvisa\ctwrapper\highlevel.py:222: VisaIOWarning: VI_WARN_CONFIG_NLOADED (1073676407): The specified configuration either does not exist or could not be loaded. VISA-specified defaults will be used.
  return self.handle_return_value(session, ret_value)  # type: ignore

Welcome to the VISA shell. Type help or ? to list commands.

(visa) open TCPIP::10.215.2.23::INSTR
TCPIP::10.215.2.23::INSTR has been opened.
You can talk to the device using "write", "read" or "query".
The default end of message is added to each message.
(open) timeout 5000
VI_ERROR_NSUP_ATTR (-1073807331): The specified attribute is not defined or supported by the referenced object.
```

**Output of `pyvisa-info`**
```
>pyvisa-info
c:\users\maierb_t\appdata\local\programs\python\python36-32\lib\site-packages\pyvisa\ctwrapper\highlevel.py:222: VisaIOWarning: VI_WARN_CONFIG_NLOADED (1073676407): The specified configuration either does not exist or could not be loaded. VISA-specified defaults will be used.
  return self.handle_return_value(session, ret_value)  # type: ignore
Machine Details:
   Platform ID:    Windows-10-10.0.17763-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\users\maierb_t\appdata\local\programs\python\python36-32\python.exe
   Version:        3.6.3
   Compiler:       MSC v.1900 32 bit (Intel)
   Bits:           32bit
   Build:          Oct  3 2017 17:26:49 (#v3.6.3:2c5fed8)
   Unicode:        UCS4

PyVISA Version: 1.11.1

Backends:
   ivi:
      Version: 1.11.1 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
   py:
      Version: 0.5.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
   sim:
      Version: 0.4.0
      Spec version: 1.1

```

MatthieuDartiailh, 2020-11-09:
This is very weird reproduce locally and the testsuite running against the Keysight testing tools do not show that either. Do you observe this behavior only in the pyvisa shell ?

buster3, 2020-11-10:
Thanks for your replay. No, I get this behavior also in scripts

This code works fine also with pyvisa==1.11.1:
```
    rm = pyvisa.ResourceManager()
    inst = rm.open_resource('TCPIP::10.215.2.23::INSTR')
    print(inst.query("*IDN?"))
```
Here i get an error:
```
    rm = pyvisa.ResourceManager()
    inst = rm.open_resource('TCPIP::10.215.2.23::INSTR', read_termination='\n', timeout=80000)
    print(inst.query("*IDN?"))
```

Stack trace:
```
C:\Users\maierb_t\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\ctwrapper\highlevel.py:222: VisaIOWarning: VI_WARN_CONFIG_NLOADED (1073676407): The specified configuration either does not exist or could not be loaded. VISA-specified defaults will be used.
  return self.handle_return_value(session, ret_value)  # type: ignore
Traceback (most recent call last):
  File "C:/git/systemtests/sim_test.py", line 16, in <module>
    inst = rm.open_resource('TCPIP::10.215.2.23::INSTR', timeout=80000)
  File "C:\Users\maierb_t\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\highlevel.py", line 3307, in open_resource
    setattr(res, key, value)
  File "C:\Users\maierb_t\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\attributes.py", line 186, in __set__
    self.pre_set(value),
  File "C:\Users\maierb_t\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\resources\resource.py", line 393, in set_visa_attribute
    return self.visalib.set_attribute(self.session, name, state)
  File "C:\Users\maierb_t\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2417, in set_attribute
    return library.viSetAttribute(session, attribute, attribute_state)
  File "C:\Users\maierb_t\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\maierb_t\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_NSUP_ATTR (-1073807331): The specified attribute is not defined or supported by the referenced object.
Exception ignored in: <bound method Resource.__del__ of <'TCPIPInstrument'('TCPIP0::10.215.2.23::inst0::INSTR')>>
Traceback (most recent call last):
  File "C:\Users\maierb_t\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\resources\resource.py", line 199, in __del__
    self.close()
  File "C:\Users\maierb_t\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\resources\resource.py", line 333, in close
    self.before_close()
  File "C:\Users\maierb_t\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\resources\resource.py", line 327, in before_close
    self.__switch_events_off()
  File "C:\Users\maierb_t\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\resources\resource.py", line 346, in __switch_events_off
    constants.EventType.all_enabled, constants.EventMechanism.all
  File "C:\Users\maierb_t\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\resources\resource.py", line 495, in disable_event
    self.visalib.disable_event(self.session, event_type, mechanism)
  File "C:\Users\maierb_t\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\ctwrapper\functions.py", line 662, in disable_event
    return library.viDisableEvent(session, event_type, mechanism)
  File "C:\Users\maierb_t\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\maierb_t\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_INV_OBJECT (-1073807346): The given session or object reference is invalid.
```

MatthieuDartiailh, 2020-11-10:
This is really confusing since I cannot reproduce this issue (neither on master, nor for the published release). Can you check the type of the resource returned by `open_resource` ? Can you perhaps also try to set up a clean environment ?

buster3, 2020-11-11:
An python update and reinstall of pyvisa fixed the issue for me. I now use python 3.8.6, 64 Bit

MatthieuDartiailh, 2020-11-11:
I will close then. Thanks for investigating this.

closed 2020-11-11 by MatthieuDartiailh