pyvisa/pyvisa #564: Freezing when executing `read_raw` in TCPIP instrument

mshrn, 2020-11-11:
Hello,
I am trying to communicate with a Keysight voltage source (E36102B) via an TPCIP SOCKET. 

I am able to connect to the device and write to the device, but not able to read.
I have read Issue#518, and did 
```
write_termination="\n"
read_termination="\n"
```

I also confirmed that `read_raw` raises TimeoutError.

Could you tell me how to fix it?

MatthieuDartiailh, 2020-11-11:
Please post the output of `pyvisa-info` and the if possible a link to your instrument manual.

mshrn, 2020-11-12:
```
$ pyvisa-info

Machine Details:
   Platform ID:    Windows-10-10.0.18362-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 10, GenuineIntel
Python:
   Implementation: CPython
   Executable:     \.venv\scripts\python.exe
   Version:        3.7.6
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Dec 19 2019 00:42:30 (#tags/v3.7.6:43364a7ae0)
   Unicode:        UCS4
PyVISA Version: 1.10.1
Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17827072
         Spec. Version: 5244928
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17827072
         Spec. Version: 5244928
```
Link is [here](https://literature.cdn.keysight.com/litweb/pdf/E36100-90002.pdf?id=2960816).

MatthieuDartiailh, 2020-11-12:
Looking at your instrument manual it appears to support the VXI11 protocol (TCPIP INSTR) not raw socket (TCPIP SOCKET). Which one have you been trying ? Can you post your code so far ?

mshrn, 2020-11-12:
I execute both `read` and `read_raw`:

```
device = dev.rm.open_resource(resource_key)
device
> <'TCPIPInstrument'(resource_key)>

device.write_termination = "\n"
device.read_termination = "\n"
device.query("*IDN?")
> VisaIOError                               Traceback (most recent call last)
\pyvisa\resources\messagebased.py in read(self, termination, encoding)
    425         if termination is None:
    426             termination = self._read_termination
--> 427             message = self._read_raw().decode(enco)
    428         else:
    429             with self.read_termination_context(termination):
\pyvisa\resources\messagebased.py in _read_raw(self, size)
    398                     logger.debug('%s - reading %d bytes (last status %r)',
    399                                  self._resource_name, size, status)
--> 400                     chunk, status = self.visalib.read(self.session, size)
    401                     ret.extend(chunk)
    402             except errors.VisaIOError as e:
\pyvisa\ctwrapper\functions.py in read(library, session, count)
   1582     buffer = create_string_buffer(count)
   1583     return_count = ViUInt32()
-> 1584     ret = library.viRead(session, buffer, count, byref(return_count))
   1585     return buffer.raw[:return_count.value], ret
   1586 
\pyvisa\ctwrapper\highlevel.py in _return_handler(self, ret_value, func, arguments)
    191 
    192         if ret_value < 0:
--> 193             raise errors.VisaIOError(ret_value)
    194 
    195         if ret_value in self.issue_warning_on:
VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```

```
device = dev.rm.open_resource(resource_key)
device
> <'TCPIPInstrument'(resource_key)>

device.write_termination = "\n"
device.read_termination = "\n"

device.write("*IDN?")
device.read_raw()
> ---------------------------------------------------------------------------
VisaIOError                               Traceback (most recent call last)
<ipython-input-20-2c56e3afe1c7> in <module>
----> 1 chunk, status = temp.visalib.read(temp.session, size)
\pyvisa\ctwrapper\functions.py in read(library, session, count)
   1582     buffer = create_string_buffer(count)
   1583     return_count = ViUInt32()
-> 1584     ret = library.viRead(session, buffer, count, byref(return_count))
   1585     return buffer.raw[:return_count.value], ret
   1586 
\pyvisa\ctwrapper\highlevel.py in _return_handler(self, ret_value, func, arguments)
    191 
    192         if ret_value < 0:
--> 193             raise errors.VisaIOError(ret_value)
    194 
    195         if ret_value in self.issue_warning_on:
VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```

MatthieuDartiailh, 2020-11-12:
That is odd in particular since it works through NI. So the best I can offer is going through the debugging procedure described in the manual. First try to use `read_bytes` to read just one character. If it fails it means somehow the instrument did not understand the command and did no answer. In that case try to use  the command`"DISP 0"` to turn the display off and try different termination character till you get the instrument to actually turn off its display.

mshrn, 2020-11-12:
I appreciate your suggestions. 
I have tried to "DISP 0" and "DISP 1", and they were successful.

However, `read_bytes(1)` raises Invalid Protocol Error: "The protocol specified is invalid".
2020-11-12 16:26:17,513 - pyvisa - DEBUG - TCPIP0::hoge::inst0::INSTR - reading 1 bytes (last status None)
2020-11-12 16:26:17,514 - pyvisa - DEBUG - viRead(1, <ctypes.c_char_Array_1 object at 0x000001F1382CA5C8>, 1, 'c_ulong(0)') -> -1073807239
2020-11-12 16:26:17,515 - pyvisa - DEBUG - TCPIP0::hoge::inst0::INSTR - exception while reading: VI_ERROR_INV_PROT (-1073807239): The protocol specified is invalid.
Buffer content: bytearray(b'')

On NI MAX, a query of "*IDN?\n" returns a string which terminated with "\n". 
I think the read_termination should be "\n", is it right?

MatthieuDartiailh, 2020-11-12:
That is getting more and more confusing. You really should not get a Protocol error. When you performed your test did you close all other connections to the instrument ?

Actually re-checking your instrument manual you can use raw SOCKET on your instrument on port 5025, could you give that a try also to see if you run into the same issue ?

mshrn, 2020-11-16:
The manual in p. 29 says that it is possible to use raw socket. I will try it later (I need to write my PhD paper, whose deadline is coming up.) I greatly appreciate for your help.

closed 2020-11-16 by mshrn