pyvisa/pyvisa #565: list_resources() freezes with PyVISA-py backend if a GPIB adapter is connected to an off instrument

CharLee674, 2020-11-11:
Hello, I am currently trying to set up PyVISA and am hoping to rely on the PyVISA-py backend in a Linux environment. I've recently encountered an issue with list_resources() where it seems to run into an infinite loop if I have it attached to a GPIB adapter that is not connected to an instrument/connected to an instrument that is off.

My use case will involve connecting to multiple VISA interfaces and I would like to be able to still see other connections under list_resources() even if the GPIB adapter is still connected when the GPIB instrument is off.

For reference, I am using an Agilent 82357B adapter, and the function normally works if the instrument is on:
```Python
import pyvisa
rm = pyvisa.ResourceManager('@py')
rm.list_resources()
# Output: ('GPIB0::7::INSTR',)
```

When the instrument is off/the adapter is not connected:
```Python
import pyvisa
rm = pyvisa.ResourceManager('@py')
rm.list_resources() # gets stuck here if the adapter is not connected to an instrument/the instrument is not on; seems to infinitely wait for a response from the adapter specifically
```

Output of `pyvisa-info`:
```
Machine Details:
   Platform ID:    Linux-5.4.51-v7l+-armv7l-with-debian-10.4
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.7.3
   Compiler:       GCC 8.3.0
   Bits:           32bit
   Build:          Jul 25 2020 13:03:44 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.0

Backends:
   ivi:
      Version: 1.11.0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.1.0). Backend: libusb1
      USB RAW: Available via PyUSB (1.1.0). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR: Available via Linux GPIB (b'4.3.3 r[1916]')
      GPIB INTFC: Available via Linux GPIB (b'4.3.3 r[1916]')
```

I understand this is kind of an extraneous use case, but I was wondering if there would be a fix for it? On a non-PyVISA-py backend, it would just ignore the adapter, so I was hoping it would be the same here.

MatthieuDartiailh, 2020-11-12:
Can you enable logging ? (using `pyvisa.log_to_screen`) I would like to know which call blocks.

CharLee674, 2020-11-12:
Here's the logger output
```Python
2020-11-11 18:26:58,194 - pyvisa - DEBUG - No visa library specified, trying to find alternatives.
2020-11-11 18:26:58,194 - pyvisa - DEBUG - Environment variable PYVISA_LIBRARY is unset.
2020-11-11 18:26:58,200 - pyvisa - DEBUG - Not loading dll_extra_paths because we are not on Windows or Python < 3.8
2020-11-11 18:26:58,587 - pyvisa - DEBUG - Automatically found library files: [None, None, None, None, None]
2020-11-11 18:26:58,588 - pyvisa - DEBUG - No user defined library files
2020-11-11 18:26:58,589 - pyvisa - DEBUG - Did not find IVI binary
2020-11-11 18:26:58,609 - pyvisa - DEBUG - SerialSession was correctly imported.
2020-11-11 18:26:58,661 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2020-11-11 18:26:58,671 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2020-11-11 18:26:58,688 - pyvisa - DEBUG - GPIBSession was correctly imported.
2020-11-11 18:26:58,689 - pyvisa - DEBUG - pyvisa-py is available.
2020-11-11 18:26:58,690 - pyvisa - DEBUG - Created library wrapper for py
2020-11-11 18:26:58,690 - pyvisa - DEBUG - Created ResourceManager with session 3980855
```

MatthieuDartiailh, 2020-11-12:
Thanks. I will have to dive a bit in the docs to see how to avoid this deadlock so it may take me some time. Do not hesitate to ping me if you do hear from me next week.

MatthieuDartiailh, 2020-11-12:
Reading the documentation from linux-gpib it does not seem pyvisa-py is doing anything really wrong here (not that I can see). It may be worth reporting upstream to linux-gpib. Note that I am not sure that they officially support Agilent adapters.

Out of curiosity did you check that you can get thing to work using a different backend ? Just asking cause I know that one bad actor in a GPIB setup can crash the whole bus.

CharLee674, 2020-11-12:
When I try the same approach on a Windows machine with NI-VISA back-end (so I can confirm it works), what happens is that the code simply ignores the interface:

```Python
# Before turning on the instrument
In [3]: a = rm.list_resources()
In [4]: print(a)
('ASRL6::INSTR', 'USB0::0x1AB1::0x0E11::DP8B185050162::0::INSTR', 'TCPIP0::169.254.1.5::inst0::INSTR')

# After turning on the instrument
In [5]: a = rm.list_resources()
In [6]: print(a)
('ASRL6::INSTR', 'USB0::0x1AB1::0x0E11::DP8B185050162::0::INSTR', 'GPIB0::7::INSTR', 'TCPIP0::169.254.1.5::inst0::INSTR')
```

I was hoping that there would be a way to have PyVISA-Py exhibit similar behaviour. What seems to be the hold-up when I stepped into the code is that when it scans for GPIB devices it would continuously wait for a response from the adapter (which, of course, would not return it since the instrument is not on).

MatthieuDartiailh, 2020-11-12:
Once again this happen inside linux-gpib so there is nothing really I can do on pyvisa-py side. Keysight provides VISA on linux free of charge but only for some kernel so I am not sure it is an option for you.

CharLee674, 2020-11-12:
Understandable, I will look into that. Thank you for the help!

closed 2020-11-12 by CharLee674

CharLee674, 2020-11-20:
I've reported to `linux-gpib` and they've added a fix through a software-side timeout for specifically the Agilent adapter so that it no longer 'hangs': https://sourceforge.net/p/linux-gpib/bugs/75/

Might be important to note in case you want to handle that new implementation.

MatthieuDartiailh, 2020-11-21:
Thanks. If you want to submit a PR with a change in that direction you are welcome, otherwise I will try to care of it at some point (not sure when though).

MatthieuDartiailh, 2020-11-21:
Tracking issue https://github.com/pyvisa/pyvisa-py/issues/288