pyvisa/pyvisa #567: Register based instrument move_in numpy array output

gmnewma, 2020-12-01:
On move_in call would it be possible to provide a return value of numpy through a container keyword argument?

Query based devices allow for 'When using numpy.array or numpy.ndarray, PyVISA will use numpy routines to optimize the conversion by avoiding the use of an intermediate representation.'

MatthieuDartiailh, 2020-12-01:
I have honestly never encountered register based devices so I would have issues testing such a change. However I would be happy to review a PR in that direction.

MatthieuDartiailh, 2020-12-01:
Let me know if you need any guidance.

gmnewma, 2020-12-01:
Great, it sounds like a change like this would be acceptable.  Thank you for the offer of guidance!

IntegralWorks, 2020-12-27:
> 
> 
> I have honestly never encountered register based devices so I would have issues testing such a change. However I would be happy to review a PR in that direction.

For Tektronix equipment, advanced VISA applicatiosn can involve the usage of registers: https://github.com/pyvisa/pyvisa/files/4919729/MDO4000-B-MSO-DPO4000B-and-MDO3000-Oscilloscope-Programmer-Manual-Rev-A.pdf; section "Status and Events"