pyvisa/pyvisa #568: [COM] Communication issue with Agilent 33250A using pyvisa

HaoLiuwide, 2020-12-05:
I am using pyvisa to connect to 33250A, however it works not well. The code shown below worked well for Agilent 2700.

```
import pyvisa
rm = pyvisa.ResourceManager()
print(rm.list_resources())
inst = rm.open_resource(u'GPIB0::10::INSTR')
print(inst.query("*IDN?"))
```

The error is occurred line 4 ,

```
rm.open_resource(u'GPIB0::10::INSTR')
```

The detail is shown below,

```
Traceback (most recent call last):
  File "C:/Users/56942/PycharmProjects/NI-instruments/main.py", line 4, in <module>
    inst = rm.open_resource(u'GPIB0::10::INSTR')
  File "C:\Users\56942\PycharmProjects\NI-instruments\venv\lib\site-packages\pyvisa\highlevel.py", line 3271, in open_resource
    info = self.resource_info(resource_name, extended=True)
  File "C:\Users\56942\PycharmProjects\NI-instruments\venv\lib\site-packages\pyvisa\highlevel.py", line 3197, in resource_info
    ret, err = self.visalib.parse_resource_extended(self.session, resource_name)
  File "C:\Users\56942\PycharmProjects\NI-instruments\venv\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2089, in parse_resource_extended
    for val in (resource_class, unaliased_expanded_resource_name, alias_if_exists)
  File "C:\Users\56942\PycharmProjects\NI-instruments\venv\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2089, in <listcomp>
    for val in (resource_class, unaliased_expanded_resource_name, alias_if_exists)
  File "C:\Users\56942\PycharmProjects\NI-instruments\venv\lib\site-packages\pyvisa\ctwrapper\types.py", line 74, in buffer_to_text
    return buf.value.decode("ascii")
    
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe6 in position 0: ordinal not in range(128)
('GPIB0::10::INSTR',)

Process finished with exit code 1
```

and it works well by using connect expert

[image]

MatthieuDartiailh, 2020-12-05:
Can you please post the output of `pyvisa-info`

HaoLiuwide, 2020-12-06:
the version of my pyvisa is 1.11.3

MatthieuDartiailh, 2020-12-06:
This issue is odd. I noticed that you have both NI and Keysight VISA installed. Could you try uninstalling one and see if the problem persist ? If you are using a GPIB adapter you should use the VISA library matching your adapter.

HaoLiuwide, 2020-12-09:
Thank you for your advice. The problem is related to the two type of VISA. I solved the problem by 

```
import pyvisa as visa

# rm = visa.ResourceManager()
rm = visa.ResourceManager('C:/Program Files (x86)/IVI Foundation/VISA/WinNT/agvisa/agbin/visa32.dll')

print(rm.list_resources())
my_2612A = rm.open_resource('GPIB0::10::INSTR')  
print(my_2612A.query('*IDN?'))
```

Although the problem was solved, I have some questions. If I have only NI-VISA installed, my computer could not recognize GPIB devices in the Measurement & Automation Explorer. Only if both NI-VISA and Keysight VISA were installed, my computer worked well.

[image]
[image]

MatthieuDartiailh, 2020-12-09:
I assume you are using a Keysight USB to GPIB adapter. This kind of adapter can only be used in conjunction with the adapter manufacturer VISA implementation. You could also try to use the Keysight VISA conflict manager (In connection expert look under Tools, and specify to use Keysight VISA for GPIB resources, it may solve your issue without having to explicitly pass the path to the dll.

closed 2020-12-09 by HaoLiuwide