pyvisa/pyvisa #573: pyvisa always return error.

tbago, 2021-01-04:
I use macos with python 2.7. And I have already install the NI-VISA_19.0.0.dmg. I can use Xcode with Objective C to call viOpenDefaultRM, and it always return success.  But when use pyvisa it always return error and never sucess.
This is the output log:

```
>>> import pyvisa
>>> rm = pyvisa.ResourceManager()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/tbago/Library/Python/2.7/lib/python/site-packages/pyvisa/highlevel.py", line 1562, in __new__
    obj.session, err = visa_library.open_default_resource_manager()
  File "/Users/tbago/Library/Python/2.7/lib/python/site-packages/pyvisa/ctwrapper/functions.py", line 1227, in open_default_resource_manager
    ret = library.viOpenDefaultRM(byref(session))
  File "/Users/tbago/Library/Python/2.7/lib/python/site-packages/pyvisa/ctwrapper/highlevel.py", line 193, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_SYSTEM_ERROR (-1073807360): Unknown system error (miscellaneous error).
```

MatthieuDartiailh, 2021-01-04:
First thing, Python 2.7 is not supported anymore. Second could you please post the output of `python -m visa info` so that we have a bit more context ?

tbago, 2021-01-04:
It is about the visa version issue. I installed the old pyvisa library. And it works well.

closed 2021-01-04 by tbago

shitou0417, 2022-08-10:
so which version of pyvisa library can solve the problem?   i had the same error as yours.