pyvisa/pyvisa #575: What is the generic way to regain control of an instrument trapped in an error?

kitmonisit, 2021-01-12:
Here I use a Keithley 2400 SourceMeter. Below are basic operations for it.

``` python
import pyvisa
rm = pyvisa.ResourceManager()
k0 = rm.open_resource("GPIB0::3::INSTR", timeout=2000)
k0.write(':source:clear:auto off')

# All well and good
k0.query_ascii_values(':measure?')

try:
    k0.write(':output:state 0')
    # This next line causes an error
    k0.read(':measure?')
except Exception as err:
    print(err)

# Any of the following 3 lines (used individually) cannot get around the instrument error
k0.write('**RST')
k0.write('*CLS')
k0.clear()

# Trying to read out the errors individually (so that they clear out) does no good either
while True:
    err = k0.query(':system:error:next?')
    print(err, end='')
    if err.startswith('0'):
        break
```

What is the generic way to brute-force clear errors?
1. Remotely
2. Without restarting the Python interpreter
2. Without being instrument-specific?
 
I've been looking around the low-level GPIB commands but I don't understand them and haven't been able to send them through the line either.

MatthieuDartiailh, 2021-01-12:
There is no fully generic way to clear errors. I have seen some instruments needing nothing else than a full restart to recover from a bad communication. The usual one would indeed be `.clear()`.

However I am not sure what you are trying to attempt here since `k0.read(":measure?")` will try to read in a situation where the instrument is not expected to answer and then use a non-sensical encoding. Do you mean to use `query` ?

IntegralWorks, 2021-01-13:
Try this:

```
import pyvisa

def init():
    rm = pyvisa.ResourceManager()
    k0 = rm.open_resource("GPIB0::3::INSTR", timeout=2000)
    k0.write(':source:clear:auto off')
    return rm, k0, False

# All well and good
k0.query_ascii_values(':measure?')

while True:
    while not someCatastrophicState:
        try:
            k0.write(':output:state 0')
            # This next line causes an error
            k0.read(':measure?')
        except Exception as err:
            print(err)
            del rm
            del k0
            rm, k0, someCatastrophicState = init()
            k0.write('**RST')
            k0.write('*CLS')

# # Any of the following 3 lines (used individually) cannot get around the instrument error
# k0.write('**RST')
# k0.write('*CLS')
# k0.clear()

# # Trying to read out the errors individually (so that they clear out) does no good either
# while True:
#     err = k0.query(':system:error:next?')
#     print(err, end='')
#     if err.startswith('0'):
#         break
```