pyvisa/pyvisa #576: PyVisa-Py on Windows7 with Hantek6022BE not listing any devices

mbobinger, 2021-01-12:
Hi Guys,

I have already posted this issue on stackoverflow already and will update the threads to avoid double posts and wasting someone
s time:
https://stackoverflow.com/questions/65681166/pyvisa-py-on-windows7-with-hantek6022be-not-listing-any-devices

The problem is that I can not access my PC oscope (Hantek6022BE) using PyVisa-Py despite Backend and packages installed. Please see the link above for the detailed error report.
Thank you for your help!

MatthieuDartiailh, 2021-01-12:
I do not know this specific instrument but nothing from the manual suggests that it can communicate through the VISA standard. Most USB only compact instrument with a dedicated software do not. You may be able to use ctypes to access the shared library of the driver.  I have done similar things to access Signal analyzers from Signal Hound.

MatthieuDartiailh, 2021-01-12:
Feel free to re-open if relevant but I do not thing we can do anything on PyVISA side.

closed 2021-01-12 by MatthieuDartiailh