pyvisa/pyvisa #578: pyvisa installation points to 1.10.1 when using pip

Chasenbeck, 2021-01-31:
Sorry, I dont know what I am talking about but I think the installation with pip `pip install PyVISA` is pointing to an old version 1.10.1. From what I understand  from the release notes, it should be installing 1.11.3. 

``
pi@raspberrypi:~$ pip install pyvisa
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pyvisa
Requirement already satisfied: enum34 in /usr/lib/python2.7/dist-packages (from pyvisa) (1.1.6)
Installing collected packages: pyvisa
  The scripts pyvisa-info and pyvisa-shell are installed in '/home/pi/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed pyvisa-1.10.1
``

When trying to install pyvisa `sudo pip install pyvisa-py`, I get the following error because it is looking for pyvisa 1.11.0 or greater:

`pi@raspberrypi:~$ pip install pyvisa-py
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pyvisa-py
  Using cached https://files.pythonhosted.org/packages/d9/cf/12bd2373bdf31d4bcc4630434561253794af43c52a4452b23c86ab1c3f24/PyVISA-py-0.5.1.tar.gz
  Installing build dependencies ... done
Collecting pyvisa>=1.11.0 (from pyvisa-py)
  Could not find a version that satisfies the requirement pyvisa>=1.11.0 (from pyvisa-py) (from versions: 1.3, 1.4, 1.5, 1.6, 1.6.1, 1.6.2, 1.6.3, 1.7, 1.8, 1.9.0, 1.9.1, 1.10.0, 1.10.1)
No matching distribution found for pyvisa>=1.11.0 (from pyvisa-py)
`

MatthieuDartiailh, 2021-01-31:
It appears you are running Python 2.7 which is not supported anymore. Try upgrading to Python >= 3.6, it should fix your issue.

Chasenbeck, 2021-01-31:
Ok. Thanks! @MatthieuDartiailh. That fixed it.

 I wonder if there is a way too foolproof the install with pyvisa-py so that there isn’t incompatibilities. Perhaps pyvisa-py only installs versions that is 2.7 compatible instead of just baulk’n that the correct version of pyvisa isn’t installed.

MatthieuDartiailh, 2021-01-31:
Apparently I forgot to add a python_requires entry in the setup configuration which is why pip picked up a too recent version of pyvisa-py. I will fix it and publish a bug fix release next week.

MatthieuDartiailh, 2021-02-04:
This should now be addressed. 0.5.2 has the proper safety in placed and I yanked 0.5.0 and 0.5.1 which means that on Python 2 it should fall back on 0.4.1.

closed 2021-02-04 by MatthieuDartiailh