pyvisa/pyvisa #579: [COM] Communication issue with R&S HMC 8012 using USB

MarcAntoineCRUE, 2021-02-01:
Instrument details
------------------
* Model: HMC 8012 DIGITAL MULTIMETER
* Communication: USB
* Link to the documentation (if available): [https://scdn.rohde-schwarz.com/ur/pws/dl_downloads/dl_common_library/dl_manuals/gb_1/h/hmc8012_1/HMC8012_SCPI_ProgrammersManual_en_01.pdf](url)

Output of `pyvisa-info`
-----------------------

```
Machine Details:
   Platform ID:    Windows-10-10.0.17763-SP0
   Processor:      Intel64 Family 6 Model 142 Stepping 12, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\users\crm1mo\appdata\local\programs\python\python39\python.exe
   Version:        3.9.0
   Compiler:       MSC v.1927 64 bit (AMD64)
   Bits:           64bit
   Build:          Oct  5 2020 15:34:40 (#tags/v3.9.0:9cf6752)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: Rohde & Schwarz GmbH
         Impl. Version: Rohde & Schwarz GmbH
         Spec. Version: Rohde & Schwarz GmbH
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: Rohde & Schwarz GmbH
         Impl. Version: Rohde & Schwarz GmbH
         Spec. Version: Rohde & Schwarz GmbH
   py:
      Version: 0.5.1
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'

```
Issue description
-----------------------

The device is used into an automatic test procedure where it should get voltage or amperage. I need to implement the resistance measurement.
The device need to use the following command to be configured for resistance measurement:
```
CONFigure:RESistance [{<Range>|AUTO|MIN|MAX|DEF}]
```
where the accepted range values are : `400 Ω, 4 kΩ, 40 kΩ, 400 kΩ, 4 MΩ, 40 MΩ, 250 MΩ `

The code 
```
command = str("CONFigure:RESistance " + "4 kΩ")
self.multimeter.write(command)
```

gives me the following result : UnicodeEncodeError: 'ascii' codec can't encode character '\u03a9' in position 24: ordinal not in range(128)

if i try to use the commands :
```
command = str("CONFigure:RESistance " + "4 kΩ")
self.multimeter.write(command.encode("utf-8"))
```
I have the following messages : TypeError: endswith first arg must be bytes or a tuple of bytes, not str

Is it possible to sent non utf-8 elements like the symbol Ω, how should I deal with it?

MarcAntoineCRUE, 2021-02-01:
Issue can be closed
I tried with Ohms, ohms, but need to be implemented using Ohm without s

closed 2021-02-01 by MarcAntoineCRUE

MatthieuDartiailh, 2021-02-01:
As a note, you should pass a str to `write` not bytes. You can specify a specific encoding using the `encoding` keyword.