pyvisa/pyvisa #586: Query(":WAV:DATA?") only gets 9 ASCii numbers

lsi8, 2021-03-02:
I am running the latest version of pyvisa on raspberry pi 4 trying to save waveform data from DS1104Z oscilloscope using the following code:

```
import numpy
import matplotlib.pyplot as plot
import sys
import time
import pyvisa as visa

resources = visa.ResourceManager('@py')
usbDevices = list(filter(lambda x: 'USB' in x, resources.list_resources()))
if len(usbDevices) == 0:
    print("no usb devices found")
    sys.exit(-1)
print(usbDevices[0])
scope = resources.open_resource(usbDevices[0])
print(scope.query('*IDN?')) #Return the Rigol’s ID string to tell us it’s there

scope.timeout = 600
scope.write("*RST")
time.sleep(8)

scope.write(":CHAN1:PROB 1")
scope.write(":CHAN1:SCAL 0.1")
scope.write(":TIM:MAIN:SCAL 0.0002")
time.sleep(2)
scope.write(":STOP")
time.sleep(8)

timescale = float(scope.query(":TIM:SCAL?"))
print("Time scale",timescale)
timeoffset = float(scope.query(":TIM:OFFS?"))
print("Time offset =",timeoffset)
voltscale = float(scope.query(':CHAN1:SCAL?'))
print("Volt scale =", voltscale)
voltoffset = float(scope.query(":CHAN1:OFFS?"))
print("Volt offset =",voltoffset)
sample_rate = float(scope.query(':ACQ:SRAT?'))
print("Sample rate =", sample_rate)
memoryDepth = sample_rate*timescale*12
print("Memory depth =",memoryDepth)

scope.write(":WAV:SOUR CHAN1")
scope.write(":WAV:MODE RAW")
scope.write(":WAV:FORM ASCii")
scope.write(":WAV:STAR 1")
scope.write(":WAV:STOP 100")
time.sleep(1)
rawdata = scope.query(":WAV:DATA?")

time.sleep(8)
print(rawdata[0:60])
rawdata = rawdata[11:len(rawdata)]
print(rawdata[0:60])

# Start data acquisition again, and put the scope back in local mode:
scope.write(":RUN")
scope.write(":KEY:FORCE")
scope.close()

splitData=rawdata.split(',')
data=[0]*len(splitData)
for i in range(0,len(splitData)):
    try:
        data[i]=float(splitData[i])
    except:
        data[i]=0
print('Raw data size =', len(splitData))
print(data[len(splitData) - 1])
```

I expect 100 ASCii numbers because I set waveform stop to 100. I think the pyvisa query method is cutting me off at only 9 numbers:

```
USB0::6833::1230::DS1ZD223400795::0::INSTR
/usr/local/lib/python3.7/dist-packages/pyvisa_py/protocols/usbtmc.py:116: UserWarning: Unexpected MsgID format. Consider updating the device's firmware. See https://github.com/pyvisa/pyvisa-py/issues/20
  "Unexpected MsgID format. Consider updating the device's firmware. See https://github.com/pyvisa/pyvisa-py/issues/20"
RIGOL TECHNOLOGIES,DS1104Z Plus,DS1ZD223400795,00.04.04.SP4

Time scale 0.0002
Time offset = 0.0
Volt scale = 0.1
Volt offset = 0.0
Sample rate = 1000000000.0
Memory depth = 2400000.0
#90000012995.297277e-03,4.023314e-07,5.297277e-03,5.297277e-
5.297277e-03,4.023314e-07,5.297277e-03,5.297277e-03,5.297277
Raw data size = 9
5.0
```

MatthieuDartiailh, 2021-03-02:
The header of your raw data looks like the header I would expect for binary data transfer. Did you configure your oscilloscope to use ASCII data transfer or binary ? Also oscilloscope rarely returns floating points, they usually return integer that you need to convert based on the scale and and offset, what does the manual say about it?

Try to go through https://pyvisa.readthedocs.io/en/latest/introduction/rvalues.html, it contains a step by step procedure to get your data transfer right.

lsi8, 2021-03-02:
I  configured scope for ascii on the line where I say...
```
scope.write(":WAV:FORM ASCii")
```

The manual says that if you choose ASCii then it gives the actual voltage values, of which I have printed the first few...
```
5.297277e-03,4.023314e-07,5.297277e-03,5.297277e-03,5.297277
```

My problem is the query method line I think:
```
rawdata = scope.query(":WAV:DATA?")
```

I believe the query method is cutting off my data stream before it is finished. If I set number of data points to anything over 9 then the stream gets cut off prematurely.

How do I fix the query method?

DavidLutton, 2021-03-02:
Try `rawdata = inst.query_ascii_values(":WAV:DATA?", container=numpy.array)`

MatthieuDartiailh, 2021-03-02:
Sorry I missed the setting to ASCII. Out of curiosity do you have access to a system where you can install either NI or Keysight VISA ? If you do, it would be great to test it there to check if that is a pyvisa-py issue (far from impossible on USB and Rigol instruments) or something else.

emorell96, 2023-08-08:
I recently had this issue with a Keysight DSO-X 2004A and I was able to figure out why.

The Keysight in ASCII mode will not simply send a list but it will include a header before the list separated with a space. The header indicates how many bites are to be read (which is superfluous info since the ascii list terminates anyways with its termination character).

I haven't been able to tell the oscilloscope to not send that header and the manual says that it will send it anyways, so I had to use read_raw() and do my own parsing:

```python
def __read_ascii(self) -> numpy.array:
        self.visa_resource.write(AgilentScopeCommands.data_read)
        data = self.visa_resource.read_raw().decode("utf-8")
        data = data.split(", ")
        data[0] = data[0].split(" ")[-1]
        return numpy.array(data, dtype=float)
```

I do this inside my own class wrapping pyvisa so it's as simple as just calling this method, but you could also just create a function if you are not using classes.

Hope this helps anyone who lands in this.

OscarS568, 2023-12-22:
Has there been a solution for this?

DavidLutton, 2023-12-24:
Later firmware versions **_may_** have fixed ASCII transfer behaviour (no length header)  

Otherwise try setting to binary format and then `query_binary_values` 
See the transfer section in [labtoolkit/Oscilloscope/KeysightInfiniiVisionX.py](https://github.com/DavidLutton/LabToolkit/blob/main/labtoolkit/Oscilloscope/KeysightInfiniiVisionX.py#L301-L322) for setting oscope to binary format