pyvisa/pyvisa #587: [COM] Communication issue with 5 Series MSO54 using Pyvisa1.11.3

DevtaD, 2021-03-04:
So I was working around with Tektronix MSO54 with Python 3.9.1 using the pyvisa library and backend ivi 1.11.3 (bundled with pyvisa) and I was just trying basic communication of reading data from it and write some functions to adjust values in it. tl;dr query works but write does not for channels.

Here is the code 

import pyvisa as visa
from time import sleep

rm = visa.ResourceManager()
mso = rm.open_resource('TCPIP0::192.168.1.10::inst0::INSTR') # Yes this is correct got this from *IDN?

# Set Channel 1 at 20Mhz Bandwidth, Offset 0, Couplling to DC & Vertical Scale at 2V/div (200e-2)
print(mso.query('CH1?')) # Calling for the Channel stats
sleep(1)
# Writing commands to change to the values on MSO's Channel
mso.write('CH1: BANDwidth 20; OFFset 0; COUPling DC; SCAle 2')
# Calling the info again to see if values have changed
print(mso.query('CH1?'))

I can't see values changing here but if I do 
mso.write('HORizontal:MODE:SCAle 2e-9')
I do see things changing in MSO

not sure what is happening here...
[image]

Instrument details
------------------
* Model: Tektronix MSO54
* Communication: TCPIP
* Link to the documentation (if available): https://in.tek.com/manual/oscilloscope/4-5-6-series-mso-5-series-mso-mixed-signal-oscilloscope-077130513

MatthieuDartiailh, 2021-03-04:
Did you check for errors after sending your command with the settings? I am surprised to see spaces after : and after ; so I am wondering if your command is properly formatted.

DevtaD, 2021-03-04:
> 
> 
> Did you check for errors after sending your command with the settings? I am surprised to see spaces after : and after ; so I am wondering if your command is properly formatted.

Oh my god, I feel like a total noob now....................

closed 2021-03-04 by DevtaD

MatthieuDartiailh, 2021-03-04:
Don't worry about it. Happy I could help.