pyvisa/pyvisa #588: Reading more than 4096 bytes in a single transaction fails intermittently

edmundsj, 2021-03-18:
I am trying to stream data from a measurement device (using a Teensy 4.1) over USB to my computer. It works great when I am transmitting less than 4093 bytes, but when I try to transmit 4096 bytes or more, I am not able to read the full number of bytes. I confirmed the device is working as expected and actually sending the correct number of bytes (this same example works just fine with pyserial and over the Arduino IDE, but with pyvisa it does not). This issue is intermittent - sometimes I get a timeout error when sending 4096 bytes and sometimes I do not, but it becomes increasingly likely as the transaction size gets larger.

The code giving me the problem is the following:

```
import pyvisa

rm = pyvisa.ResourceManager()
dev = rm.open_resource('MY_DEVICE_NAME')
dev.write_line('MEASURE?')
dev.read_bytes(4096) # This fails with a timeout error

```
And my error:
```
 pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```

**Output of `pyvisa-info`**
```
Machine Details:
   Platform ID:    Darwin-20.3.0-x86_64-i386-64bit
   Processor:      i386

Python:
   Implementation: CPython
   Executable:     /Users/jordan.e/opt/miniconda3/bin/python
   Version:        3.7.6
   Compiler:       Clang 4.0.1 (tags/RELEASE_401/final)
   Bits:           64bit
   Build:          Jan  8 2020 13:42:34 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: /Library/Frameworks/visa.framework/visa:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
```

edmundsj, 2021-03-18:
I have also confirmed that the device is in fact only *sending* data during the specified period because I can see the communication LED on the Teensy light up only for 1 second. The pyvisa interface just doesn't appear to be receiving all of it.

MatthieuDartiailh, 2021-03-18:
Since you are running on the ivi backend (using NI VISA), and the error is only intermittent I am not sure what we can do on pyvisa side. Since things works over pyserial you could try running through pyvisa-py that will use pyserial as the backend for serial resources and see if you can reproduce the issue.