pyvisa/pyvisa #589: How to set timeout for rm.open_resource() when trying to connect to an instrument

ThomasGund, 2021-03-19:
Hi,
This is maybe not really an issue, but rather a question. 

I have made a short script to check if instruments are connected to a network. I know that the instruments will answer within 100 msec if it they are connected. Unfortunately, if it they are not connected I have to wait for the rm.open_resource(‘TCPIP_INST_ADDR’) to timeout. I have measured this timeout to ~7 sec. Quite a long time to wait when/if many of the instruments in my checklist are not connected to the network. Is there a way I can set the timeout for rm.open_resource()?

My code:
```python
import pyvisa

instr_list = ['TCPIP0::IPADDR_01::inst0::INSTR',
              'TCPIP0::IPADDR_02::inst0::INSTR',
              ...,
              'TCPIP0::IPADDR_35::inst0::INSTR']

rm = pyvisa.ResourceManager()

for i in range(len(instr_list)):
    print('\nCheck connection to: ', instr_list[i])
    try:
        instrument = rm.open_resource(instr_list[i])
        query = instrument.query('*IDN?')
        print('>', instr_list[i], '\tConnected.\n> *IDN? >', query, end =' ')
    except:
        print('>', instr_list[i], '\tNot connected')```

MatthieuDartiailh, 2021-03-23:
The VISA standard provides an open_timeout which is exposed by PyVISA however it is only related to the acquisition of a VISA lock and will not achieve your purpose. 

If you are interested only in TCPIP resources you could try to directly open a socket with their address on port 111 which is supposed to accept request for port query for the vxi11 protocol used by for TCPIP::INSTR resources (I assume you do not have hislip resources).

ThomasGund, 2021-03-30:
Thank you for the answer.
Unfortunately, there could be USB resources in the checklist as well. And as you mention, the open_timeout is related to the the VISA lock. It seems like I have to build a separate timeout function in Python and use that. I hoped that there would be a parameter that I could set rather than making the code more complex with extra functions. The code is quite simple at the moment and (almost) everyone could understand and run it without me having to support them. 

Is the lack of a of connection timeout parameter related to the VISA driver?

MatthieuDartiailh, 2021-03-30:
The VISA specification (available at https://www.ivifoundation.org/specifications/) does not expose this capability, sadly. 

You could add an argument to `open_resource` (not sure what name to use though) that if present would try to open the connection in a Thread and stop waiting for the thread if nothing happen (but we would need to join the Thread later...). Actually it may be better to have it as a separate helper class, to check connection. I am not sure how generic it would be but I would consider reviewing a PR in that sense.

ThomasGund, 2021-03-31:
I have done some testing related to timeout. 

My observations:
From what I see it seems like the timeout/open_timeout parameter in open_resource() is what I am looking for. From looking at traffic in RsVisaTraceTool it seems like default value for timeout is 2000 msecs. When I run my script it takes approx. 14 secs before timeout. It is independent of which VISA driver I am using. Unfortunately, it seems like timeout parameter can not be set to values lower than 2000 msecs. The long timeout seems to be only related to TCPIP resources. For USB resources the connection test will respond fast if the resource is not connected, within 0,5 - 2 msecs.

What I have tested:
1. I have run my script with only one VISA resource to see how long I have to wait before I get an error message if the resource isn’t connected

1. Tried with TCPIP and USB resources

1. Tested different VISA drivers:
•	R&S VISA 5.12.3
•	Keysight 18.2.26526.0 
•	NI 20.0

1. Used tools from R&S to test connecting to a VISA resource and see VISA traffic/commands:
•	RsVisaTester 5.12.3
•	RsVisaTraceTool 5.12.3

RsVisaTester 5.12.3
RsVisaTester has a connection timeout parameter. It is set to 2000 ms as default. When trying to connect to TCPIP0::10.71.30.31::inst0::INSTR in RsVisaTester; RsVisaTester will return VI_ERROR_RSRC_NFOUND after 2 secs if it can't find the resource. I have tried to set the timeout to 4 secs, it will then return the VI_ERROR_RSRC_NFOUND after 4 secs.

RsVisaTraceTool 5.12.3
Running my script and using RsVisaTraceTool to see traffic/commands. When running my script it takes approx. 14 secs before I get  error message: pyvisa.errors.VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343. But RsVistaTraceTool shows timeout error (-1073807343 VI_ERROR_RSRC_NFOUND) after 2 secs. Have to wait 12 extra seconds before getting error message in Python.

**Logs from RsVisaTraceTool:**

**Test: Running script, using R&S VISA driver.**
RsVisaTraceTool:

	Line		timestamp					Adress					Duration		Status		Command
	0	0	13:49:35.000	1224	12348	1							0.0002783	1	0		viOpenDefaultRM(pvi(0000016BC0211B68)=1)
	1	1	13:49:35.059	1224	12348	1		DefaultRM				0.0003682	1	0		viParseRsrcEx(sesn=1,rsrc="TCPIP0::10.71.30.31::inst0::INSTR",intfType(0000016BC09EAE08)=6,intfNum(0000016BC0A29808)=0,class="INSTR",exp="TCPIP0::10.71.30.31::inst0::INSTR",alias="")
	2	2	13:49:35.059	1224	12348	1		DefaultRM				0.0001916	1	0		viParseRsrcEx(sesn=1,rsrc="TCPIP0::10.71.30.31::inst0::INSTR",intfType(00000014B55EC920)=6,intfNum(00000014B55EC924)=0,class="INSTR",exp="TCPIP0::10.71.30.31::inst0::INSTR",alias="")
	3	3	13:49:35.061	1224	12348	3224215984	DefaultRM				2.01027		1	-1073807343	viOpen(sesn=1,rsrc="TCPIP0::10.71.30.31::inst0::INSTR",mode=0,timeout=0,vi(00000014B55EC9D8)=3224215984)
	4	4	13:49:37.072	1224	12348	3224215984	TCPIP0::10.71.30.31::inst0::INSTR	2.15e-05	1	-1073807346	viClose(vi=3224215984)

Observation: We can sett from the log that we get timeout connection error after 2.01027 seconds; error message -1073807343. Then I think it is a bit strange that I have to wait 12 more seconds when running my script.

**Test: Using RsVisaTester to connect to TCPIP0::10.71.30.31::inst0::INSTR, which is not connected.**
RsVisaTraceTool:

	Line		timestamp					Adress					Duration		Status		Command
	0	6	13:53:10.958	14104	11392	0		DefaultRM				2.00691		1	-1073807343	viOpen(sesn=1,rsrc="TCPIP0::10.71.30.31::inst0::INSTR",mode=0,timeout=2000,vi(000002E3677EE228)=0)

Observation: The log only have one line, viOpen. The RsVisaTester GUI has a setting for timeout. It is set to 2000 msec as default. Timeout = 2000 in viOpen and from the log we can see it takes 2,00691 secs before we get timeout message.

**Test: Trying to change the timeout parameter in RsVisatester GUI, timeout = 3000.**
Using RsVisaTester to connect to TCPIP0::10.71.30.31::inst0::INSTR, which is not connected:
RsVisaTraceTool:

	Line		timestamp					Adress					Duration		Status		Command
	0	8	14:03:54.699	14720	9100	0		DefaultRM				3.0078		1	-1073807343	viOpen(sesn=1,rsrc="TCPIP0::10.71.30.31::inst0::INSTR",mode=0,timeout=3000,vi(000001AF21EA5228)=0)

Observation: We see that timeout = 3000 in viOpen and timeout before getting timeout message is now 3.0078 seconds. 

**Test: Trying with different values for timeout.**
RsVisaTraceTool:

	Line		timestamp					Adress					Duration		Status		Command
	0	11	14:10:23.609	14720	9100	0		DefaultRM				2.51506		1	-1073807343	viOpen(sesn=1,rsrc="TCPIP0::10.71.30.31::inst0::INSTR",mode=0,timeout=2500,vi(000001AF21EA5228)=0)
	2	13	14:10:52.224	14720	9100	0		DefaultRM				3.75913		1	-1073807343	viOpen(sesn=1,rsrc="TCPIP0::10.71.30.31::inst0::INSTR",mode=0,timeout=3750,vi(000001AF21EA5228)=0)
	0	14	14:27:46.734	14720	9100	0		DefaultRM				2.01417		1	-1073807343	viOpen(sesn=1,rsrc="TCPIP0::10.71.30.31::inst0::INSTR",mode=0,timeout=1000,vi(000001AF21EA5228)=0)

Observation: It seems like timeout has to be >= 2000 msec. When setting timeout=1000; still have to wait 2 secs before getting timeout message in RsVisaTraceTool.

**Test: Trying to use instrument = rm.open_resource(instr_list[i], open_timeout = 3000) in my script and running it.**
RsVisaTraceTool:

	Line		timestamp					Adress					Duration		Status		Command
[... 45 lines cut ...]
```python
import pyvisa
import datetime

instr_list = ['TCPIP0::10.71.30.31::inst0::INSTR']

rm = pyvisa.ResourceManager()

for i in range(len(instr_list)):
    print('\nCheck connection to: ', instr_list[i])
    start = datetime.datetime.now()
    try:
        instrument = rm.open_resource(instr_list[i], open_timeout = 1000)
        stop = datetime.datetime.now()
        query = instrument.query('*IDN?')
        print('>', instr_list[i], '\tConnected.\n> *IDN? >', query, end =' ')
    except:
        stop = datetime.datetime.now()
        print('>', instr_list[i], '\tNot connected')

diff = stop - start
print(start)
print(stop)
print('Diff:', diff) 
```