pyvisa/pyvisa #590: No USB or Ethernet devices found

yuanhzha, 2021-03-30:
I have been successfully communicating with a GPIB or RS232 interface before. But I have never seen the USB or Ethernet devices listed by the resources. Since I have a USB webcam on my desktop, and USB-interfaced mouse and keyboard, I think I should see these devices listed as resources. But the pyvisa cannot see it. What's wrong???

I installed [NI-VISA as a backend ](https://www.ni.com/en-us/support/downloads/drivers/download.ni-visa.html#346210). 
```
In [1]: import pyvisa

In [2]: rm = pyvisa.ResourceManager()

In [3]: rm.list_resources()
Out[3]: ('ASRL1::INSTR', 'ASRL3::INSTR')
```
--------------------------------------------------
Output of `pyvisa-info`
----------------------------------------------------

(base) C:\Users\zhang>pyvisa-info
Machine Details:
   Platform ID:    Windows-10-10.0.19041-SP0
   Processor:      Intel64 Family 6 Model 58 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\users\zhang\anaconda3\python.exe
   Version:        3.8.3
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Jul  2 2020 17:30:36 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
   py:
      Version: 0.5.1
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'

yuanhzha, 2021-03-30:
I then tried to use the PyVISA-Py backend, and installed pyUSB with pip. I was notified the pyusb itself also need a backend. I download libusb and unzip it on the computer. But what's the next step?? I still got the same error

====================================================
(base) C:\Users\zhang>pyvisa-info
Machine Details:
   Platform ID:    Windows-10-10.0.19041-SP0
   Processor:      Intel64 Family 6 Model 58 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\users\zhang\anaconda3\python.exe
   Version:        3.8.3
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Jul  2 2020 17:30:36 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
   py:
      Version: 0.5.1
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      USB RAW:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'

MatthieuDartiailh, 2021-03-31:
VISA is meant to communicate with test and measurement devices (oscilloscope, voltmeters, etc) and uses on USB the USBTMC protocol (Universal Serial Bus Test and Measurement Class). It is not meant to communicate with random pieces of equipment on the USB bus. Mouse and keyboard are definitively not supported. 

If you need to interact with such peripherals, you can try to have a look as using directly pyusb, but neither VISA, nor pyvisa and pyvisa-py are the tools you need.

closed 2021-03-31 by MatthieuDartiailh

yuanhzha, 2021-03-31:
@MatthieuDartiailh  Thank you for clarifying this :)  What about the second problem? I already installed `libusb` by pip, why it still says the `PyUSB` needs a backend?

MatthieuDartiailh, 2021-03-31:
As suggested in the report from pyvisa-info I would suggest you check the pyusb documentation, I do not remember of teh top of my head the procedure for all platforms.

yuanhzha, 2021-04-01:
I see. Thank you so much! Really helpful