pyvisa/pyvisa #591: [COM] Communication issue with XXX using XXX

sjlukacs, 2021-04-05:
hello one and all,
i'm porting my communication with my incubator from python 2 to python 3.7 and i'm trying to not use the serial import but only pyvisa for it recognizes my USB connection to the incubator when i list_resources.  so, i'm trying to only rely on pyvisa for all of my communications.

the example code below works with the serial connection but not when i switch to the pyvisa, and it gives the same error under pyvisa no matter i use "visa = pyvisa.ResourceManager()" or "visa = pyvisa.ResourceManager("@py")" for that line in the example code.  the error it throws when i use pyvisa is on the "db.write(a)" which is:

  File "/usr/local/lib/python3.7/dist-packages/pyvisa/resources/messagebased.py", line 190, in write
    if message.endswith(term):
TypeError: endswith first arg must be bytes or a tuple of bytes, not str

the funny thing is i'm passing bytes to bd.write(a), through variable "a".

#!/usr/bin/env python3
# -*- coding: utf-8 -*-
import serial
import binascii, struct
import pyvisa
visa = pyvisa.ResourceManager("@py")
#bd = serial.Serial('/dev/ttyUSB%i' % 0, timeout=3)
bd = visa.open_resource('ASRL/dev/ttyUSB0::INSTR')
a = bytes.fromhex('0103 0000 0002 C40B')
print(type(a), a, len(a))
print("written:", bd.write(a)) ###<<---this is where the above exception is thrown but not when "bd = serial.Serial..."
x = bd.read(9)
print(3, type(x), x, len(x))
b = binascii.b2a_hex(x)
print(3, type(b), b, len(b))
print("---")
x = b[10:14]+b[6:10]
print(3, type(x), x, len(x))
#n = struct.unpack('!f', x.decode('hex'))[0]
n = struct.unpack('!f', bytes.fromhex(x.decode('utf-8')))[0]
print(3, type(n), n, round(n, 3))
exit()

here is the output when using serial:
<class 'bytes'> b'\x01\x03\x00\x00\x00\x02\xc4\x0b' 8
8
3 <class 'bytes'> b'\x01\x03\x04d7A\xef%\x11' 9
3 <class 'bytes'> b'010304643741ef2511' 18
---
3 <class 'bytes'> b'41ef6437' 8
3 <class 'float'> 29.923933029174805 29.924

AND, here is the output when using pyvisa:
<class 'bytes'> b'\x01\x03\x00\x00\x00\x02\xc4\x0b' 8
Traceback (most recent call last):
  File "./prac5_visa_test.py", line 24, in <module>
    print("written:", bd.write(a))
  File "/usr/local/lib/python3.7/dist-packages/pyvisa/resources/messagebased.py", line 190, in write
    if message.endswith(term):
TypeError: endswith first arg must be bytes or a tuple of bytes, not str

WHERE you'll see i'm passing the same exact bytes to both serial and pyvisa.  that bytes is basically the command to the incubator to tell it to report the temperature back to the comm port.  in which 9 bytes are read and then translated to human readable temperature.

thank you in advance and hopefully and didn't give too much information.  lucas

MatthieuDartiailh, 2021-04-05:
On Python 3, PyVISA expects that if you pass text, it is indeed unicode. In your case, I am not sure how relevant VISA is to you, in particular you do not seem to really care about termination character handling. If it is indeed the case you can use the lower level function `write_raw` which expects bytes and will not add or attempt to add a termination character.

sjlukacs, 2021-04-05:
that helped a bit.  i believe it wrote the characters to write_raw but then when i tried to read it, it bugged.  so then i tried with read_raw(9) and it bugged again with the following:

<class 'bytes'> b'\x01\x03\x00\x00\x00\x02\xc4\x0b' 8
written: 8
Traceback (most recent call last):
  File "./prac5_visa_test.py", line 29, in <module>
    x = bd.read_raw(9)
  File "/usr/local/lib/python3.7/dist-packages/pyvisa/resources/messagebased.py", line 405, in read_raw
    return bytes(self._read_raw(size))
  File "/usr/local/lib/python3.7/dist-packages/pyvisa/resources/messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/usr/local/lib/python3.7/dist-packages/pyvisa_py/highlevel.py", line 519, in read
    return data, self.handle_return_value(session, status_code)
  File "/usr/local/lib/python3.7/dist-packages/pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

MatthieuDartiailh, 2021-04-05:
Once again you do not seem to rely on termination character as is usually done to identify end of messages in VISA. If you need to read a fixed number of bytes, use `read_bytes`, not `read_raw`.

sjlukacs, 2021-04-05:
ok, and yes, I get what you're saying.  I don't add any prefix leading or suffix trailing characters when I use serial to get smooth communications.  so is serial prefixing or suffixing bytes or characters that I don't know about?  or is there a way to see the comm trails?

sjlukacs, 2021-04-05:
ahhh, that did it perfectly.

```
bd.write_raw(a)
x = bd.read_bytes(9)

```
thank you for all of your help and guidance.  lucas

closed 2021-04-05 by sjlukacs