pyvisa/pyvisa #592: [COM] Communication issue with Keysight MSOS404A using USB connection

hgetter, 2021-04-06:
I am running a python script to obtain the analog waveform. The commands I used are pasted below. I have tried lowering the waveform points to only 16 data points. I have tried changing the format to ascii, binary and word formats. I have tried running query_ascii_values() and query_binary_values(). Part of the issue is that when I run a binary query I receive a "#0" at the beginning of data dump, this means the bus is reading 0 bytes, correct?

When I run it like this, it times out even with a 250000 ms time-out:
[image]
  

This is the trace for all the commands I ran to get to the position I was in. The trigger event and acquisition was completed before running this script.
[image]

When I run query_ascii_values() and change the format ascii, I can see the trace pick up this data: 

viRead (USB0::0x2A8D::0x904D::_____::0::INSTR (0x00000001), "1.983825E-01,1.974604E-01,1.986460E-01,1.999632E-01,1.971970E-01,1.977239E-01,1.966701E-01,1.970652E-01,2.019391E-01,1.983825E-01,1.953528E-01,2.020708E-01,2.019391E-01,1.966701E-01,1.979873E-01,1.990411E-01,.", 20480 (0x5000), 209 (0xD1))
Process ID: 0x00012C14         Thread ID: 0x0000000106308DC0
Start Time: 14:25:30.8572      Call Duration 00:00:00.0005
Status: 0 (VI_SUCCESS)

My code ends with this error: 
[image]

This shows that there is data to be obtained and the "#0" is incorrect. 

I've talked to Keysight extensively and they are certain it is not an issue with the Oscilloscope, leading me to believe it is a scripting issue or an Pyvisa issue. Here is my whole script for reference. What do you believe could be causing this issue?

[Infiniium_Basic_Analog_Waveform_Grab_Python-3.5.py.zip](https://github.com/pyvisa/pyvisa/files/6267696/Infiniium_Basic_Analog_Waveform_Grab_Python-3.5.py.zip)

Instrument details
------------------
* Model: Keysight MSOS404A

* Communication: USB

Output of `pyvisa-info`
-----------------------
[image]

MatthieuDartiailh, 2021-04-07:
Hi @hgetter,

First thanks for the very detailed issue report.

So starting with the binary transfer, the fact that the header is `#0` signifies that the instrument does not specify how much data it is going to transfer, which some manufacturer decide to do when you are expected to already know this information. In your case you do and you properly used the `points` argument to the `query_binary_values`. 

Looking at your script, you appear to pass a datatype of `f` while you properly indicates in your comments below it should be h for WORD data, so you need to fix this. 

Also even in the case of binary transfer where you know how many points to expect, there is one ambiguity remaining which has to do with the termination character used in VISA communication. In some cases, this character will be appended at the end of the binary block, and in some other cases it will be omitted. I have seen both behavior in different Keysight instrument. You can control this behavior using the `expect_termination` argument. If True you will expect the termination character to be appended to the block.

Looking at your logs the one thing bothering me, is that you expect 16 WORD but you read only 16 bytes in addition of the header, which is very weird. However it can be that the last byte is `\n` causing the first read to terminate and the next one fails because it looked for 12*4 bytes because you indicated a format of `f`  and hence 4 bytes per point but you have only 2 bytes per point and the buffer hence contains only 16 bytes. In any case, I recommend you give a look at https://pyvisa.readthedocs.io/en/latest/introduction/rvalues.html#when-things-are-not-what-they-should-be that discuss in details how to trouble shoot that exact kind of issues.

Regarding the ASCII transfer, it appears that Keysight made the weird choice of appending a `,` after the last value (which is just plain weird). This causes the conversion to crash. You can either do a conventional query and handle the conversion yourself or you can try to patch PyVISA by adding the following in utils.py below line 348:
```python
        if not data[-1]:
            data = data[:-1]
```
If this works feel free to open a PR against PyVISA. 

Hope this helps and do not hesitate to come back to me if you have any further questions.

ThomasGund, 2021-04-08:
Hi hgetter,
I have an MSO oscilloscope and I have also recently been working on a script do download waveform data. I tried to run your script. And ‘luckily’, I get the same error message for query_ascii_values. I don,t know why, but MatthieuDartiailh is probably correct when says that “appending a , after the last value” might be the problem.

In my script, I use sData = instrument.query(":WAVeform:DATA?") for ASCII waveform data. I have tried to change your code from
Data_Ch1 = np.array(KsInfiniium.query_ascii_values('WAVeform:DATA?'))
print(Data_Ch1)
To:
Data_Ch1 = KsInfiniium.query(':WAVeform:SOURce CHANnel1;DATA?')
print(Data_Ch1)
I didn’t include the np.array. I was just interesting in the data.

And it works. You now get the waveform data. But it seems like you will get a new problem at line 443:
Traceback (most recent call last):
  File "G:\Python\MSOS104A\Infiniium_Basic_Analog_Waveform_Grab_Python-3.5.py\Infiniium_Basic_Analog_Waveform_Grab_Python-3.5.py", line 443, in <module>
    Data_Ch1 = (Data_Ch1*Y_INCrement_Ch1)+Y_ORIGin_Ch1
numpy.core._exceptions._UFuncNoLoopError: ufunc 'multiply' did not contain a loop with signature matching types (dtype('<U236742'), dtype('<U236742')) -> dtype('<U236742')

I haven’t tried to run your script with BYTE or WORD format. I will look into that later 😊

MatthieuDartiailh, 2021-04-08:
The error you see @ThomasGund is likely due to the fact that you do not convert the string you retrieve from the instrument into an array (on which the operation causing an error would make sense). Could you test my suggested patch and make a PR if it works ?

ThomasGund, 2021-04-08:
@MatthieuDartiailh 
what I wrote in my reply to @hgetter was a bit wrong.
First I tried:  Data_Ch1 = KsInfiniium.query(':WAVeform:SOURce CHANnel1;DATA?')
Then I tried: Data_Ch1 = np.array(KsInfiniium.query(':WAVeform:SOURce CHANnel1;DATA?'))

Both of them will give an error message at line 443, two different error messages. The error message I wrote in my reply was for Data_Ch1 = np.array(KsInfiniium.query(':WAVeform:SOURce CHANnel1;DATA?'))

MatthieuDartiailh, 2021-04-08:
Simply calling np.array on a string won't convert into an array of floating point numbers. You need to call `np.fromstring(Data_Ch1, "f", sep=",")`. Numpy will properly handle the trailing comma, so actually calling `query_ascii_values` with `container=np.ndarray` should work.

ThomasGund, 2021-04-08:
@hgetter 
It seems like your problem with binary data is related to :WAVeform:STReaming. :WAVeform:STReaming is set to ON in your script. I tried with
```
:WAVeform:FORMat BYTE
```
and 
```
:WAVeform:FORMat WORD
```
both got stuck at.
```python 
Data_Ch1 = np.array(KsInfiniium.query_binary_values(':WAVeform:SOURce CHANnel1;DATA?', datatype = "f", is_big_endian = False, data_points = POINTS))
```
It could be that the script is waiting some kind of termination. I am not familiar with using  :WAVeform:STReaming ON

If setting streaming = OFF, then I got data:
```
:WAVeform:STReaming OFF
:WAVeform:FORMat BYTE
```
and/or
```
:WAVeform:STReaming OFF
:WAVeform:FORMat WORD
```

It seems like you will get a new error message when writing the data to file:
```
Traceback (most recent call last):
File "G:\Python\MSOS104A\Infiniium_Basic_Analog_Waveform_Grab_Python-3.5.py\Infiniium_Basic_Analog_Waveform_Grab_Python-3.5.py", line 515, in <module>
    np.savetxt(filehandle, np.vstack((DataTime,Data_Ch1)).T, delimiter=',')
  File "<__array_function__ internals>", line 5, in vstack
  File "C:\Users\thgu\AppData\Roaming\Python\Python39\site-packages\numpy\core\shape_base.py", line 283, in vstack
    return _nx.concatenate(arrs, 0)
  File "<__array_function__ internals>", line 5, in concatenate
ValueError: all the input array dimensions for the concatenation axis must match exactly, but along dimension 1, the array at index 0 has size 20003 and the array at index 1 has size 5000
```

I have a slightly different approach when getting data. I have basically taken code from the examples in the Infiniium Programmer's Guide; corrected some errors in their code and extended the functionality. Their original example only supported BYTE format.
``` python
waveform_format_query = instrument.query(":WAVeform:FORMat?")
waveform_byteorder_query = instrument.query(":WAVeform:BYTeorder?")

# Get the waveform data.
instrument.write(":WAVeform:STReaming OFF")

if waveform_format_query.upper()[0:3] != 'ASC':
    sData = instrument.query_binary_values(":WAVeform:DATA?", datatype='B', container=bytearray)
    #print(sData)
else:
    sData = instrument.query(":WAVeform:DATA?")
    #print(len(sData))
    #print('sData_list', len(sData_list))
    #print(sData_list)
    
# Unpack data
if waveform_format_query.upper()[0:3] == 'ASC':
    print('Format: ASCII')
    sData_list = sData.split(',')
    sData_len = len(sData_list)-1
elif waveform_format_query.upper()[0:4] == 'WORD':
    print('Format: WORD')
    sData_len = len(sData)/2
    #print('Number of values received:', sData_len)
    # Endianess, MSB or LSB. Read from :WAVeform:BYTeorder?
    if waveform_byteorder_query.upper()[0:4] == 'MSBF':
        values = struct.unpack(">%dh" % sData_len, sData)   # MSB, double, short
    else:
        values = struct.unpack("<%dh" % sData_len, sData)   # LSB, double, short
elif waveform_format_query.upper()[0:4] == 'BYTE':
    print('Format: BYTE')
    sData_len = len(sData)
    #print('Number of values received:', sData_len)
    values = struct.unpack("%db" % sData_len, sData)
else:
    print('Not a recognized format')
```

ThomasGund, 2021-04-09:
@hgetter 
Hi Harrison,
it seems like the problem is that datatype is set to 'f' in KsInfiniium.query_binary_values()

```
Data_Ch1 = KsInfiniium.query_binary_values(':WAVeform:SOURce CHANnel1;DATA?', datatype = "f", is_big_endian = False, data_points = POINTS))
```
I tried with "B" instead when waveform:format BYTE
```
Data_Ch1 = KsInfiniium.query_binary_values(':WAVeform:SOURce CHANnel1;DATA?', datatype = "B", is_big_endian = False, data_points = POINTS)
```
The script doesn't get stuck at query_binary_value when using B. 

When waveform:format WORD, datatype should be "h"
```
Data_Ch1 = np.array(KsInfiniium.query_binary_values(':WAVeform:SOURce CHANnel1;DATA?', datatype = "h", is_big_endian = False, data_points = POINTS))
```

By correcting those two things, I was able to run your script without any errors. 
I have compared the CSV files from your script to the CSV files from my script and the CSV files when saved directly on the oscilloscope. The values looks fine, but the values in your CSV file have a lot of decimals. Even more decimals than the oscilloscope reports. That may make your CSV unnecessary big; it may take a long time to open and process the files. It depends on how many datapoints you are transferring. There are some other minor thing you could consider to correct in your script as well, but it works now.

I hope that helped you.

hgetter, 2021-04-12:
Thank you everyone for the assistance! Changing the datatype to "h" and turning streaming off allowed me to obtain the data from the scope. Here is an updated version of my code. I still don't fully understand why streaming has to be turned off to obtain data. Can someone explain this to me?

Querying without streaming puts a cap of the data able to be acquired to 1 GB, so ideally I would not have this cap.

[
[Infiniium_Basic_Analog_Waveform_Grab_Python-3.5.py.zip](https://github.com/pyvisa/pyvisa/files/6300286/Infiniium_Basic_Analog_Waveform_Grab_Python-3.5.py.zip)
](url)

ThomasGund, 2021-04-13:
@hgetter 
Hi Harrison,
Nice to hear that it helped. 

I tried to run your updated script and it works with both streaming ON and OFF when I run it.
From my Windows command window:
[image]

I am running on Windows though, and it seems like you are using MacOS. I don't know whether that may have anything to say. Which version of PyVISA are you using?

A short comment:
It will take a long time to write a lot of waveform points. Getting them from the scope is quite fast, but writing them to CSV takes time. I tried to get 100 Mpts from my oscilloscope; transferring them took less than 5 secs, but saving to CSV took almost 30 minutes. It is recommended that you save CSV directly at your computer. Saving on a network drive makes it even slower.

I have made some small updates to your script. I just wanted to make it easier to keep track of settings and progress:
[Infiniium_Basic_Analog_Waveform_Grab_Python-3-5_updated.zip](https://github.com/pyvisa/pyvisa/files/6302661/Infiniium_Basic_Analog_Waveform_Grab_Python-3-5_updated.zip)

hgetter, 2021-04-15:
@ThomasGund Hi Thomas,

this is my py-visa info.

Machine Details:
   Platform ID:    macOS-10.15.7-x86_64-i386-64bit
   Processor:      i386

Python:
   Implementation: CPython
   Executable:     /Users/harrisongetter/env/bin/python3
   Version:        3.8.2
   Compiler:       Clang 12.0.0 (clang-1200.0.32.2)
   Bits:           64bit
   Build:          Aug 25 2020 09:23:57 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.1

Backends:
   ivi:
      Version: 1.11.1 (bundled with PyVISA)
      #1: /Library/Frameworks/visa.framework/visa:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
   py:
      Version: 0.5.1
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.1.0). Backend: libusb1
      USB RAW: Available via PyUSB (1.1.0). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'

ThomasGund, 2021-04-16:
@hgetter
Hi Harrison,
I am out of suggestions. 

The differences I can see is that I am using Windows, Python 3.9.2 and PyVISA 1.11.3. I don't know whether it makes any difference.

pyvisa-info
Machine Details:
   Platform ID:    Windows-10-10.0.19041-SP0
   Processor:      AMD64 Family 23 Model 96 Stepping 1, AuthenticAMD

Python:
   Implementation: CPython
   Executable:     c:\program files\python39\python.exe
   Version:        3.9.2
   Compiler:       MSC v.1928 64 bit (AMD64)
   Bits:           64bit
   Build:          Feb 19 2021 13:44:55 (#tags/v3.9.2:1a79785)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: Rohde & Schwarz GmbH
         Impl. Version: Rohde & Schwarz GmbH
         Spec. Version: Rohde & Schwarz GmbH
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: Rohde & Schwarz GmbH
         Impl. Version: Rohde & Schwarz GmbH
         Spec. Version: Rohde & Schwarz GmbH