pyvisa/pyvisa #593: [COM] Communication issue with XXX using XXX

mnemocron, 2021-04-11:
This is a low-level question not directly affecting the pyvisa software. However, I hope someone here can point me at low-level documentation or source files required to debug this issue. I am having issues connecting to my custom hardware / firmware using python. 

**Keysight Connection Expert and Telnet from the console both work flawlessly.**

My question is, how does the pyvisa library differ from a telnet connection on the low-level side. Is there a diagram, state machine, documentation ... ?

### Telnet

```bash
$ > telnet 192.168.1.179 5025
Trying 192.168.1.179...
Connected to 192.168.1.179.
Escape character is '^]'.
*IDN?
ACME INC.,VXI11 DEVICE 9000,V1.0.0,2021-03-14
:TEST:TEXT "hello"
TEXT: ***hello***
```

> works: STM32 replies with correct messages

### Keysight Connection Expert

The Keysight Connection Expert correctly identifies my hardware assigning the address: `'TCPIP0::192.168.1.179::5025::SOCKET'` 

I can open the console there and the communication works perfectly.

### pyvisa (in Spyder)

```python
import pyvisa as visa
import vxi11 as vxi
from time import sleep

rm = visa.ResourceManager()
print(rm.list_resources())
# ('ASRL9::INSTR', 'GPIB0::20::INSTR', 'USB0::0x2A8D::0x0101::MY54505614::0::INSTR')

dut = rm.open_resource('TCPIP0::192.168.1.179::5025::SOCKET')
dut.timeout = 1000;

pause = 0.5
sleep(pause)

print(dut.query('*IDN?', pause))

VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

dut.close()
```

> timeout occurs. On my debug console from the STM32 I can see that the "*IDN?" string is correctly received and handled accordingly. But the python file never receives anything (timout error) no matter what timeout I set.

Instrument details
------------------
* Model: STM32f746zg Nucleo-144
* Communication: TCPIP (LwIP based using scpi-parser library)
* Link to the documentation (if available): [mnemocron/STM32F7_LXI_Device](https://github.com/mnemocron/STM32F7_LXI_Device)

Output of `pyvisa-info`
-----------------------

```
Machine Details:
   Platform ID:    Windows-10-10.0.19041-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\programdata\anaconda3\python.exe
   Version:        3.8.8
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Feb 20 2021 15:50:08 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: Keysight Technologies
         Spec. Version: Keysight Technologies
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: Keysight Technologies
         Spec. Version: Keysight Technologies
```

MatthieuDartiailh, 2021-04-11:
You probably need to set the termination character. In your case Pyvisa is using Keysight so you mort likelysimply

mnemocron, 2021-04-11:
Haha, that was exactly it! 😄 
Definitely the pointer that I needed. Have a lovely Sunday :)

for anyone curious:

```python
dut = rm.open_resource('TCPIP0::192.168.1.179::5025::SOCKET')  # Visa Drivers over GPIB-USB Adapter
dut.timeout = 1000;
dut.read_termination = '\n'

pause = 0.1
sleep(pause)

print(dut.query('*IDN?'))

dut.close()
```
`ACME INC.,VXI11 DEVICE 9000,V1.0.0,2021-03-14`

closed 2021-04-11 by mnemocron

MatthieuDartiailh, 2021-04-11:
Happy to help