pyvisa/pyvisa #594: pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

unknownfox71, 2021-04-25:
Instrument details
------------------
* Model:SIM900, SIM960
* Communication: RS232 USB  'ASRL3::INSTR'
* Link to the documentation (if available): https://www.thinksrs.com/downloads/pdfs/manuals/SIM900m.pdf; https://www.thinksrs.com/downloads/pdfs/manuals/SIM960m.pdf
* 

Output of `pyvisa-info`
-----------------------

import pyvisa as visa

SIM900=visa.ResourceManager().open_resource('ASRL3::INSTR',baud_rate=9600, write_termination="\n", read_termination="\n")
SIM900.write('CONN 5')  # sim960 is on the slot 5
SIM900.write('SNDT 5,"GAIN 100"') #  set the P(gain) = 100
print(SIM900.query("*IND?"))

### i use the sim900 mainframe to connect the sim960(PID).
### when i write the information into the sim900, it worked, but once i "read",  it shows the following error. 

-------------------------------the terminal information-----------------------------------------------------------
Traceback (most recent call last):
  File "c:\Users\Oliver\Desktop\sim960.py", line 13, in <module>
    print(SIM900.query("*IND?"))
  File "C:\Users\Oliver\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 644, in query
    return self.read()
  File "C:\Users\Oliver\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "C:\Users\Oliver\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\Oliver\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\Oliver\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\Oliver\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

MatthieuDartiailh, 2021-04-25:
First I would advise against throwing the resource manager since when it will be garbage collected it will wreak havoc in your code.

Looking at the manual of your instrument you expect a baud rate of 9600 (which is the default), no parity (again pyvisa default), but a hardware based control flow, which is not the default. To properly configure the connection try the following:
```python
import pyvisa
from pyvisa.constants import ControlFlow

rm = pyvisa.ResourceManager()
instr = rm.open_resource(
    'ASRL3::INSTR',
    baud_rate=9600, 
    write_termination="\n", 
    read_termination="\n", 
    flow_control=ControlFlow.rts_cts
)
```

unknownfox71, 2021-04-26:
Thank you, i add that on.
And, now, i find, i can query other command, 
```
SIM900.write('SNDT 5,"MMON?"')   # Query the Measure input voltage on port 5(SIM900)
for i in range(3):
    print("the value:",SIM900.query('GETN? 5,40'))    # get the meg form port 5.  up to 40 bytes.
```
the output is 
```
the value #3024-00.008053   #  3(rule) 024(the number of bytes) -00.00805 (the value i need)
the value -00.008140
the value
```
### so, i just thought, maybe,  
```
SIM900.write("*IND?")          # or SIM900.write('SNDT 5,"*IND?"'), the results are exactly the same, with no output at all.
for i in range(3):
    print("the value:",SIM900.query('GETN? 5,40'))
# the output
the value #3000
the value #3000
the value #3000
```
i don't know why.  Or just ignore it?

MatthieuDartiailh, 2021-04-26:
Not sure for your instrument but typically you should be able to query IDN*? and I think that GETN is reserved to read the results of some particular orders related to measurements. Since IDN does not do any this is likely why you get an empty response.

ThomasGund, 2021-04-26:
@unknownfox71 
I see that you write *IND? in the examples here, not *IDN? May that be the reason why you aren't getting any answer from the mainframe?

@MatthieuDartiailh 
May I ask what you mean by (I would just like to learn and understand):
"First I would advise against throwing the resource manager since when it will be garbage collected it will wreak havoc in your code. "

unknownfox71, 2021-04-26:
@ThomasGund 
yes, i was my fault, just a typo. 
and
```
print(SIM900.query("*IDN?"))
# i get the information of  mainframe SIM900, #  research_Systems,SIM900,s/n152126,ver3.6
SIM900.write('SNDT 5,"*IDN?"')
print(SIM900.query('GETN? 5,80'))
# i get the SIM960, ford_Research_Systems,SIM960,s/n024632,ver2.17
```
thank you so much  @MatthieuDartiailh and @ThomasGund

MatthieuDartiailh, 2021-04-26:
If you do `instr = pyvisa.ResourceManager().open_resource("ASRL10::INSTR")` you do not keep a reference to the resource manager. And the only live reference is the one the resource has. When the resource is garbage collected, the resource manager is too. It used to cause some issues in the past and I took the habit to discourage throwing away the rm an not closing the resource. Technically, it should not happen anymore and all resources are closed before disposing of the ResourceManager, so I should update my advice.

Happy you got everything to work @unknownfox71

closed 2021-04-26 by MatthieuDartiailh