pyvisa/pyvisa #595: Get control over Thorlabs PM100USB from python

dilrukshi924, 2021-04-26:
I have Thorlabs PM100USB power meter connected to the computer and I want to get the current reading from python. From the following code, the measurement type shows "TEMP", which gives the current temperature from a temperature sensor. I printed out the list of resources and this is what I'm getting

`rm.list_resources()
Out[7]: ('USB0::0x1313::0x80F8::M00495436::INSTR', 'ASRL1::INSTR', 'ASRL10::INSTR')`

and I used it in my code

`import visa
import pyvisa
from ThorlabsPM100 import ThorlabsPM100
rm = visa.ResourceManager()
rm.list_resources()
inst = rm.open_resource('USB0::0x1313::0x80F8::M00495436::INSTR')
power_meter = ThorlabsPM100(inst=inst)
inst.timeout = None

print("Measurement type :", power_meter.getconfigure)
print("Current value    :", power_meter.read)

print(inst.query("*IDN?"))
print(rm)
print(inst)`

This is the output I'm getting

Measurement type : TEMP
Current value    : 24.350668
Thorlabs,TSP01,M00495436,1.2.0

Resource Manager of Visa Library at C:\WINDOWS\system32\visa64.dll
USBInstrument at USB0::0x1313::0x80F8::M00495436::INSTR

I want to get the power reading, not the temperature reading. Any lead would be appreciated

MatthieuDartiailh, 2021-04-26:
Reading the manual of your instrument (available at https://www.thorlabs.com/_sd.cfm?fileName=12271-D02.pdf&partNumber=PM100) it appears you want the POWER? command

dilrukshi924, 2021-04-26:
Using that command I get an error `VisaIOError: VI_ERROR_INP_PROT_VIOL (-1073807305): Device reported an input protocol error during transfer.`
Why `rm.list_resources()` only lists out the temperature sensor, but not the power sensor?

MatthieuDartiailh, 2021-04-26:
According to the documentation, your instrument use a serial emulation. So try the ASRL10.

Excerpt of the documenation:
```
This device provides an interface which is similar to the IEEE1174 “Serial Interface for  Programmable  Instrumentation” standard which provides a way to emulate the common IEEE488 interface over a serial link using SCPI-style commands. Refer to the SCPI specification for further information. The device is considered to be a Data Terminal Equipment (DTE). To connect it with a control PC use the 9-pole null-modem cable that is provided with the optical power meter system. Plug one end to the DB9 socket in the bottom plate of the PM100, the other end to a free COM port of the PC. 
```

dilrukshi924, 2021-04-26:
you mean `inst = rm.open_resource('ASRL10::INSTR')` ?

MatthieuDartiailh, 2021-04-26:
Yes

dilrukshi924, 2021-04-26:
It does not give any output

MatthieuDartiailh, 2021-04-26:
The manual does not specify VISA explicitely but some custom Thorlas driver so we may not be able to achieve what you want and you may have better luck contacting Thorlabs. However let's try the following first, since serial instrument can require more configuration. First check that if you disconnect the power meter, it disappear from the listed instrument (that will allow to confirm the port number). Next try the following snippet based on the manual information and post any error you may get:
```python
import pyvisa
from pyvisa.constants import ControlFlow
rm = pyvisa.ResourceManager()
instr = rm.open_resource(
    "ASRL10::INSTR", 
    write_termination="\n", 
    read_termination="\r\n", 
    flow_control=ControlFlow.rts_cts
)
instr.query("*IDN?")
```

 If it does not work you can also try to use NI-MAX, Keysight Connection Expert or whatever program came with your VISA installation and see if you can figure the configuration, or directly contact Thorlabs. FYI I never worked with Thorlabs instruments.

LongnoseRob, 2021-04-26:
did you have a look at pymeasure? they have a class for the Thorlabs 100 USB sensor 
https://github.com/pymeasure/pymeasure/blob/f9bab2cd4c3418d8e8d7d7f12e4a26a6c17efe3c/pymeasure/instruments/thorlabs/thorlabspm100usb.py#L32

dilrukshi924, 2021-04-26:
> The manual does not specify VISA explicitely but some custom Thorlas driver so we may not be able to achieve what you want and you may have better luck contacting Thorlabs. However let's try the [...]

Using the code snipped I got this
`ImportError: cannot import name 'ControlFlow' from 'pyvisa.constants' (C:\Users\TJ_Research_i3_2\Anaconda3\lib\site-packages\pyvisa\constants.py)`

dilrukshi924, 2021-04-26:
> did you have a look at pymeasure? they have a class for the Thorlabs 100 USB sensor
> https://github.com/pymeasure/pymeasure/blob/f9bab2cd4c3418d8e8d7d7f12e4a26a6c17efe3c/pymeasure/instruments/thorlabs/thorlabspm100usb.py#L32

Would you be able to provide a small example of identifying the device based on that class? That would be really helpful

MatthieuDartiailh, 2021-04-26:
Can you try updating pyvisa ? Or at least post the output of pyvisa-info so that I know what version you are using and update the snippet.

dilrukshi924, 2021-04-27:
this is `pyvisa-info`
`Machine Details:
   Platform ID:    Windows-10-10.0.19041-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 11, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\TJ_Research_i3_2\Anaconda3\python.exe
   Version:        3.7.3
   Compiler:       MSC v.1915 64 bit (AMD64)
   Bits:           64bit
   Build:          Mar 27 2019 17:13:21 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 20971520
         Spec. Version: 5244928
      #2: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 20971520
         Spec. Version: 5244928
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'`

dilrukshi924, 2021-04-27:
Also, I ran `rm.list_resources()` with `@py` backend, such that `rm = visa.ResourceManager('@py')`  and I get the following
`Found a device whose serial number cannot be read. The partial VISA resource name is: USB0::4883::32882::???::0::INSTR`

Yes, this is my device. But I can't access it because the serial number is not showing

LongnoseRob, 2021-04-27:
> Would you be able to provide a small example of identifying the device based on that class? That would be really helpful

Unfortunately not, I just was working on adding some Multimeter to pymeasure and notice the name Thorlabs.

MatthieuDartiailh, 2021-04-27:
I would only have limited confidence in pyvisa-py ability to list USB devices. On your version of pyvisa you can try, since I added the ControlFlow enum later.

```python
import pyvisa
from pyvisa.constants import VI_ASRL_FLOW_RTS_CTS
rm = pyvisa.ResourceManager()
instr = rm.open_resource(
    "ASRL10::INSTR", 
    write_termination="\n", 
    read_termination="\r\n", 
    flow_control=VI_ASRL_FLOW_RTS_CTS
)
instr.query("*IDN?")
```

dilrukshi924, 2021-04-28:
https://knowledge.ni.com/KnowledgeArticleDetails?id=kA00Z0000019La2SAE&l=en-CA

So NI-MAX was showing my device as a serial device, I guess that's why I couldn't access it with `pyvisa`. Then I reinstalled NI-VISA and after that the device appeared as a USB connection on NI-MAX. Now I am able to access the device.
Thanks all for the support!

MatthieuDartiailh, 2021-04-28:
That was unexpected... Thanks for letting us know.

closed 2021-04-28 by MatthieuDartiailh

desula, 2021-12-09:
I read the information file and found out that there's example python code installed when you install the Thorlabs software. It was in the folder C:\Program Files (x86)\IVI Foundation\VISA\WinNT\TLPM\Example\Python for me. The example program (PowermeterSample.py) read the power on the first try for me. It also includes a file with all relevant functions (TLPM.py). Hope this helps.

unycorn, 2024-06-05:
The solution to this that works for me is using the Thorlabs Driver Switcher to change to the older PM100D (NI-VISA) driver