pyvisa/pyvisa #596: [COM] Try...Except pyvisa.errors.VisaIOError doesn't work

IntegralWorks, 2021-04-26:
Instrument details
------------------
* Model: Tektronix MDO 3024
* Communication: TCPIP
* Link to the documentation (if available): https://www.naic.edu/~phil/hardware/oscilloscopes/MDO4000-B-MSO-DPO4000B-and-MDO3000-Oscilloscope-Programmer-Manual-Rev-A.pdf

Output of `pyvisa-info`
-----------------------
Machine Details:
   Platform ID:    Windows-10-10.0.18362-SP0
   Processor:      Intel64 Family 6 Model 58 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\users\redacted\appdata\local\programs\python\python38\python.exe
   Version:        3.8.2
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Feb 25 2020 23:03:10 (#tags/v3.8.2:7b3ab59)
   Unicode:        UCS4

PyVISA Version: 1.11.1

Backends:
   ivi:
      Version: 1.11.1 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments

Problem Description
-----------------------
My whole code is too complicated to share here, as it relies a lot on the library `PySimpleGUI`. But basically the gist of my issue is the following design pattern doesn't work:

```
try:
    while True:
        doImportantTasks() #such as inst.query('*IDN?')
except pyvisa.errors.VisaIOError:
    del rm
    del inst
    rm = pyvisa.ResourceManager()
    inst = rm.open_resource(IP_ADDRESS)
```

I can make a code sample if needed, but has anyone encountered this issue? Essentially, how do I reconnect a device after it has been intermittently disconnected?

MatthieuDartiailh, 2021-04-26:
First at least your snippet appears to have a typo, since the while loop should be outside the try except to do what you expect. Could you elaborate a bit more on the behavior you observe ?

IntegralWorks, 2021-04-26:
@MatthieuDartiailh I know I am not asking my question very well, but I really don't know how to explain the issue without just posting the code wholesale. Because I didn't use an object-oriented architecture, all the globals have caught me into this scope creep that can be hard to get out of. 

https://pastebin.com/WhHjZU0k

Only if you have time, maybe you can review Lines 673-677, Lines 430-438, and the function definitions `selectScope`, `initializeScope`, and `reconnect`. If you don't have time or don't want to, I will properly write some example code that is concise.

In my power electronics work, sometimes a phenomenon involving electrical transients causes NI-VISA to drop the connection intermittently. I just need to know how to destroy a TCPIP connection and restore it while still running a python script.

IntegralWorks, 2021-04-27:
@MatthieuDartiailh I have written proper example code that works. If you don't want to install PySimpleGUI, please tell me so I can remake this CLI-only.

```
import pyvisa as visa
import PySimpleGUI as sg
import re, sys, os

rm = visa.ResourceManager()
def selectScope():
    entries = []
    newl_char = "\n"
    success = False
    for n,i in enumerate(rm.list_resources()):
        if re.search('TCPIP',i):
            entries.append(f'instrument:{i} index: {n}') 
    selection = sg.popup_get_text(f'{newl_char.join(entries)}','select a scope')
    while success == False:
        try:
            scope = rm.open_resource(rm.list_resources()[int(selection)])
            success = True
        except ValueError:
            selection = sg.popup_get_text(f'ERROR: Must type one of the indices listed here:\n{newl_char.join(entries)}','select a scope')
    sg.popup(f'Selected: {scope.query("*IDN?")}')
    return scope

scope = selectScope()

def initializeScope(scope=scope):
    scope.timeout = 20000 # ms
    scope.encoding = 'latin_1'
    scope.read_termination = '\n'
    scope.write_termination = None
    scope.write('*cls') # clear ESR
    scope.write('header OFF') # disable attribute echo in replies
initializeScope()

def checkErrorStatus():
    scope.query('*ESR?')
    try:
        return errors[int(scope.query('EVENT?'))]
    except:
        print("error code not found")

query_rgx = re.compile(r'(.*)\?')
def executeCommand(inst, s, rgx):
    if re.search(rgx, s):
        print('Query Command')
        print(inst.query(s))
    else:
        print('Write Command')
        inst.write(s)
    

layout =  [[]]
layout += [[sg.Combo(['*IDN?', 'clearmenu'], k = 'Command_Menu', enable_events=True)]]
window = sg.Window('Window Title', layout, finalize=True)

while True:
    try:
        event, values = window.read()
        if event in (None, 'Cancel'):   # Output:if user closes window or clicks cancel
            break
        if event == 'Command_Menu':
            executeCommand(scope, values['Command_Menu'], query_rgx)
    except visa.errors.VisaIOError:
        del rm
        del scope
        rm = visa.ResourceManager()
        scope = selectScope()   
        initializeScope()     
[... 46 lines cut ...]
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "C:\Users\redacted\AppData\Local\Programs\Python\Python38\lib\site-packages\pyvisa\highlevel.py", line 3096, in close
    resource.close()
  File "C:\Users\redacted\AppData\Local\Programs\Python\Python38\lib\site-packages\pyvisa\resources\resource.py", line 334, in close
    self.visalib.close(self.session)
  File "C:\Users\redacted\AppData\Local\Programs\Python\Python38\lib\site-packages\pyvisa\ctwrapper\functions.py", line 637, in close
    return library.viClose(session)
  File "C:\Users\redacted\AppData\Local\Programs\Python\Python38\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\redacted\AppData\Local\Programs\Python\Python38\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_CLOSING_FAILED (-1073807338): The VISA driver failed to properly close the session or object reference. This might be due to an error freeing internal or OS resources, a failed network connection, or a lower-level driver or OS error.
Exception ignored in: <function Resource.__del__ at 0x0000019FBBA171F0>
Traceback (most recent call last):
  File "C:\Users\redacted\AppData\Local\Programs\Python\Python38\lib\site-packages\pyvisa\resources\resource.py", line 199, in __del__
  File "C:\Users\redacted\AppData\Local\Programs\Python\Python38\lib\site-packages\pyvisa\resources\resource.py", line 333, in close
  File "C:\Users\redacted\AppData\Local\Programs\Python\Python38\lib\site-packages\pyvisa\resources\resource.py", line 327, in before_close
  File "C:\Users\redacted\AppData\Local\Programs\Python\Python38\lib\site-packages\pyvisa\resources\resource.py", line 345, in __switch_events_off
  File "C:\Users\redacted\AppData\Local\Programs\Python\Python38\lib\site-packages\pyvisa\resources\resource.py", line 495, in disable_event
  File "C:\Users\redacted\AppData\Local\Programs\Python\Python38\lib\site-packages\pyvisa\ctwrapper\functions.py", line 662, in disable_event
  File "C:\Users\redacted\AppData\Local\Programs\Python\Python38\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
  File "C:\Users\redacted\AppData\Local\Programs\Python\Python38\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
pyvisa.errors.VisaIOError: VI_ERROR_INV_OBJECT (-1073807346): The given session or object reference is invalid.
```

MatthieuDartiailh, 2021-04-27:
Thanks for the abridged version. It looks like you are faced with more than just a random issue causing an unexpected timeout (which I assumed at first). Recovering from a connection lost may be more complicated and what appears here is that even after re-opening the connection, the new connection gets corrupted inside your except block and fail. I would advise the following but I cannot guarantee it will work:
- first manually close the resource and the resource manager rather than relying on the garbage collector doing it for you
- after closing maybe sleep for a second or so, it may give the OS time to cleanup the mess
- put the closing reopening logic itself into a while/try except to be able to retry that part too (but put some limit on the number of retries to avoid an infinite loop)

Let me know how it goes.

IntegralWorks, 2021-04-27:
I appreciate the sound advice @MatthieuDartiailh. I went ahead and did the following modification:

```
reconnection_attempts = 0
while True:
    try:
        event, values = window.read()
        if event in (None, 'Cancel'):   # if user closes window or clicks cancel
            break
        if event == 'Command_Menu':
            executeCommand(scope, values['Command_Menu'], query_rgx)
    except visa.errors.VisaIOError:
        while reconnection_attempts != 10:
            try:
                reconnection_attempts += 1
                print(f'Reconnection attempt no.{reconnection_attempts}...')
                scope.close()
                rm.close()
                del scope
                del rm
                print('objects deleted. reinitializing (Please ensure physical connection(s) is/are nominal again)...')
                rm = visa.ResourceManager()
                scope = selectScope()
                initializeScope()
            except visa.errors.VisaIOError:
                print('failure')
```

I get strange behavior.
```
C:\Users\jsalazar\Documents\git_repos\Autoscope>py connectionTest.py
Write Command
Reconnection attempt no.1...
failure
Reconnection attempt no.2...
failure
Reconnection attempt no.3...
failure
Reconnection attempt no.4...
failure
Reconnection attempt no.5...
failure
Reconnection attempt no.6...
failure
Reconnection attempt no.7...
failure
Reconnection attempt no.8...
failure
Reconnection attempt no.9...
failure
Reconnection attempt no.10...
failure
```

After that last `failure` I can use my `Query Command` or `Write Command` but it just silently fails. I don't get the rest of the traceback unless I exit the program:

```
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "C:\Users\jsalazar\AppData\Local\Programs\Python\Python38\lib\site-packages\pyvisa\highlevel.py", line 3096, in close
    resource.close()
  File "C:\Users\jsalazar\AppData\Local\Programs\Python\Python38\lib\site-packages\pyvisa\resources\resource.py", line 333, in close
    self.before_close()
  File "C:\Users\jsalazar\AppData\Local\Programs\Python\Python38\lib\site-packages\pyvisa\resources\resource.py", line 327, in before_close
    self.__switch_events_off()
  File "C:\Users\jsalazar\AppData\Local\Programs\Python\Python38\lib\site-packages\pyvisa\resources\resource.py", line 345, in __switch_events_off
    self.disable_event(
  File "C:\Users\jsalazar\AppData\Local\Programs\Python\Python38\lib\site-packages\pyvisa\resources\resource.py", line 495, in disable_event
    self.visalib.disable_event(self.session, event_type, mechanism)
  File "C:\Users\jsalazar\AppData\Local\Programs\Python\Python38\lib\site-packages\pyvisa\ctwrapper\functions.py", line 662, in disable_event
    return library.viDisableEvent(session, event_type, mechanism)
  File "C:\Users\jsalazar\AppData\Local\Programs\Python\Python38\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\jsalazar\AppData\Local\Programs\Python\Python38\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_INV_OBJECT (-1073807346): The given session or object reference is invalid.
```

Perhaps
```
Error in atexit._run_exitfuncs
``` 
might be of interest?

MatthieuDartiailh, 2021-04-27:
I would still suggest trying to sleep after cleaning up resources, it might help. Also I would suggest to print the exception causing to try again. Finally since you just need TCPIP resource, you could look into using pyvisa-py since it may give you more control into how the sockets are cleaned up.