pyvisa/pyvisa #601: pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

jbehura, 2021-05-17:
Timeout error when running the code below. I suspect it is related to the termination character. However, I have tried different combinations of termination characters to no avail.

To Reproduce
--------------
```
import pyvisa as visa

ipaddress = "192.168.1.1"

visa.ResourceManager('@py')

rm = visa.ResourceManager()
scope = rm.open_resource("TCPIP::" + ipaddress + "::INSTR")
print(scope)
print(scope.query('*IDN?'))
print(scope.read_bytes(1))

```
Output:

> /usr/bin/python3.8 /home/jbehura/Work/Projects/Wild/Code/acquireTekData3.py
> TCPIPInstrument at TCPIP0::192.168.1.1::inst0::INSTR
> TEKTRONIX,TDS 3014B,0,CF:91.1CT FV:v3.41 TDS3FFT:v1.00 TDS3TRG:v1.00
> 
> Traceback (most recent call last):
>   File "/home/jbehura/Work/Projects/Wild/Code/acquireTekData3.py", line 12, in <module>
>     print(scope.read_bytes(1))
>   File "/usr/local/lib/python3.8/dist-packages/pyvisa/resources/messagebased.py", line 371, in read_bytes
>     chunk, status = self.visalib.read(self.session, size)
>   File "/usr/local/lib/python3.8/dist-packages/pyvisa_py/highlevel.py", line 519, in read
>     return data, self.handle_return_value(session, status_code)
>   File "/usr/local/lib/python3.8/dist-packages/pyvisa/highlevel.py", line 251, in handle_return_value
>     raise errors.VisaIOError(rv)
> pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

**Output of `pyvisa-info`**

Machine Details:
   Platform ID:    Linux-5.4.0-66-generic-x86_64-with-glibc2.29
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3.8
   Version:        3.8.5
   Compiler:       GCC 9.3.0
   Bits:           64bit
   Build:          Jan 27 2021 15:41:15 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.2
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'

MatthieuDartiailh, 2021-05-17:
What do you esxpect to read ? You already read the answer to the IDN in the query so the read_bytes has nothing to read. Everything looks fine.

If you were using write instead of query though it would make sense.

jbehura, 2021-05-18:
Hi Matthieu,

I appreciate the quick response. I am trying to read the time-series corresponding to channel 1:
```
import pyvisa as visa
import numpy
import time

ipaddress = "192.168.1.1"

visa.ResourceManager('@py')

rm = visa.ResourceManager()
scope = rm.open_resource("TCPIP::" + ipaddress + "::INSTR")

# Read Channel
scope.write('ACQUIRE:STATE STOP')
scope.write('ACQUIRE:STATE RUN')
scope.write('ACQUIRE:STOPAFTER SEQUENCE')

scope.write('*WAI')
scope.write(":DATa:SOUrce CH" + str(1))
time.sleep(20)

voltage = scope.query_ascii_values('CURVe?', container=numpy.array)

```
Not sure where I am going wrong:

> Traceback (most recent call last):
>   File "/home/jbehura/Work/Projects/Wild/Code/acquireTekData2.py", line 11, in <module>
>     print(pyTekLib.get4TraceAcqTime(scope=scope, waitTime=0, trueValue=False))
>   File "/home/jbehura/Work/Software/PyLib/pyTekLib.py", line 551, in get4TraceAcqTime
>     _,_,_,_ = get4ChannelData(scope, waitTime, trueValue=trueValue)
>   File "/home/jbehura/Work/Software/PyLib/pyTekLib.py", line 562, in get4ChannelData
>     trace1 = scope.query_ascii_values('CURVe?', container=numpy.array)
>   File "/usr/local/lib/python3.8/dist-packages/pyvisa/resources/messagebased.py", line 685, in query_ascii_values
>     return self.read_ascii_values(converter, separator, container)
>   File "/usr/local/lib/python3.8/dist-packages/pyvisa/resources/messagebased.py", line 529, in read_ascii_values
>     block = self.read()
>   File "/usr/local/lib/python3.8/dist-packages/pyvisa/resources/messagebased.py", line 486, in read
>     message = self._read_raw().decode(enco)
>   File "/usr/local/lib/python3.8/dist-packages/pyvisa/resources/messagebased.py", line 442, in _read_raw
>     chunk, status = self.visalib.read(self.session, size)
>   File "/usr/local/lib/python3.8/dist-packages/pyvisa_py/highlevel.py", line 519, in read
>     return data, self.handle_return_value(session, status_code)
>   File "/usr/local/lib/python3.8/dist-packages/pyvisa/highlevel.py", line 251, in handle_return_value
>     raise errors.VisaIOError(rv)
> pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

MatthieuDartiailh, 2021-05-18:
When reading binary values a number of things can go wrong. I suggest you give a quick look to the relevant part of the documentation (https://pyvisa.readthedocs.io/en/latest/introduction/rvalues.html#reading-binary-values). If you still struggle after that feel free to ask more questions (and in that case provide the results of the small tests suggested in the docs).

jbehura, 2021-05-18:
I am reading ASCII values using `query_ascii_values`.

MatthieuDartiailh, 2021-05-18:
Sorry I read too fast. The documentation also has a section regarding ASCII transfer (https://pyvisa.readthedocs.io/en/latest/introduction/rvalues.html#reading-ascii-values). Note also that when transferring large amout of data you may need to increase the default timeout simply because there is a lot of data to transfer.

jbehura, 2021-05-18:
Hi Matthieu,

That hint helped me solve the issue. It was a case of "stupid user syndrome". Turns out the trigger was not working properly and so there was no data to be read.
On fixing the trigger, everything is working beautifully.

Cheers,
Jyoti

MatthieuDartiailh, 2021-05-18:
Happy to help

closed 2021-05-18 by MatthieuDartiailh

Sarada67, 2024-06-01:
@MatthieuDartiailh Hi sir , I need some help.
I am also getting same error . I am using python and Pyvisa to read measurement values from Tektronix MSO58 oscilloscope . I am trying to read delay measurement value . But everytime i am getting the same timeout error . I am sending query command , it is throwing error . what to do sir .
Even if i am simply trying to send query *IDN? command  , it is throwing timeout error.

Sarada67, 2024-06-01:
@MatthieuDartiailh  Hi sir , i am also struggling with timeout error . I am using python and pyvisa to communicate with Tektronix MSO58 scope . I am trying to read scope measurement , but it is throwing timeout error everytime . How to fix it , please help me sir , i am struggling