pyvisa/pyvisa #602: [COM] Communication issue with Newport ESP302 Controller using pyvisa 1.11.3

chr0203, 2021-05-19:
Instrument details
------------------
* Model: Newport ESP302 Controller with BGS80PP Goniometric stage
* Communication: TCPIP
* Link to the documentation (if available): https://www.newport.com/medias/sys_master/images/images/h22/hae/9540544725022/EDH0414En1012-ESP302-Programmer-s-Manual-.pdf

Output of `pyvisa-info`
-----------------------

-----------------------
Hi! This is my first time asking a question on github! Please let me know if the format/question is not adequate, if there is any lack of information etc. Thank you very much! 

I have an issue getting some data points from my Newport ESP302 Controller. Im using an ascii command 'DG' to retrieve the data points but I get a timeout error. I have tried a lot of different things they mostly result in timeout errors. I have communication with the instrument and all other commands works fine. I suspect that it has something to do with the format that the data points have to be read as.  

This is my code: 

_import pyvisa
import time
pyvisa.log_to_screen()
rm = pyvisa.ResourceManager()
ESP302 = 'TCPIP0::ESP-562C::5001::SOCKET'
inst = rm.open_resource(ESP302)

inst.read_termination = '\r\n'
inst.write_termination = '\r\n'
values1 = inst.write('1MO')
time.sleep(2)
values2 = inst.write('DC1,1,0,1,1,20')
time.sleep(2)
values3 = inst.write('DE1')
time.sleep(2)
values4 = inst.write('1TP')
time.sleep(2)
values5 = inst.read()
print(values5)
time.sleep(2)
values6 = inst.write('1PA10')
time.sleep(2)
values7 = inst.write('1WS500')
time.sleep(15)
values8 = inst.write('DD')
time.sleep(2)
values9 = inst.read()
print(values9)
time.sleep(2)
values10 = inst.write('DE0')
time.sleep(2)
values11 = inst.write('DF')
time.sleep(2)
values12 = inst.read()
print(values12)
time.sleep(2)
values13 = inst.write('DG')
time.sleep(20)
values14 = inst.read()
print(values14)_

I can get correct outputs from all the other reads, just not the 'DG' command. I have tried inst.read_ascii_values(), inst.read_binary_values() with a bunch of different parameters with no luck. An example on the use of the data aquisition scheme, including the type of output expected from the 'DG' command can be found in link for documentation at the bottom (page 41). 

I get the following error when using the code above and other types of read functions read_raw(), read_bytes() etc.: 

_2021-05-19 15:22:50,150 - pyvisa - DEBUG - viRead(1, <ctypes.c_char_Array_20480 object at 0x000002A9F450C7C0>, 20480, 'c_ulong(0)') -> -1073807339
2021-05-19 15:22:50,150 - pyvisa - DEBUG - TCPIP0::ESP-562C::5001::SOCKET - exception while reading: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
Buffer content: bytearray(b'')
Traceback (most recent call last):
  File "C:\Users\Christoffer.Soerense\PycharmProjects\NWPRTLib\main.py", line 41, in <module>
    values14 = inst.read()
  File "C:\Users\Christoffer.Soerense\PycharmProjects\NWPRTLib\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "C:\Users\Christoffer.Soerense\PycharmProjects\NWPRTLib\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\Christoffer.Soerense\PycharmProjects\NWPRTLib\venv\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\Christoffer.Soerense\PycharmProjects\NWPRTLib\venv\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\Christoffer.Soerense\PycharmProjects\NWPRTLib\venv\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed._

[ESP302-Controller-Programmer-s-Manual-.pdf](https://github.com/pyvisa/pyvisa/files/6509206/ESP302-Controller-Programmer-s-Manual-.pdf)

MatthieuDartiailh, 2021-05-19:
According to the manual for DG, data are returned as binary with no header so it is very possible that  the termination character is omitted. I suggest you first try using `read_bytes` to read exactly the number of bytes you expect then try to read an extra one. Given what you have seen I would expect the second read to fail. 

If you observe this behavior, you should be able to use the following command (with N the number of points you expect) to recover your data and convert to integers:
`read_binary_values(header_fmt="empty", is_big_endian=True, datatype="I", expect_termination=False, data_points=N)`

Let me know if this helps.

chr0203, 2021-05-19:
Thank you for the reply. I tried using inst.read_bytes() with 80 bytes since I have 20 points (20*4=80, assuming the termination character has been omitted), instead of inst.read(). Unfortunately that did not work (timeout error again). I also tried the with the other line you wrote (inst.read_binary_values()). And got a timeout error as well. Did I understand your suggestions correctly? (replacing my read() line with one of the two options (read_bytes() or read_binary_values()). Cheers.

MatthieuDartiailh, 2021-05-19:
Yes you understood correctly. Try reading a single byte `read_bytes(1)`, if this fails or return \0 it likely means your instrument did not perform any measurement. What response do you get to DF ?

chr0203, 2021-05-19:
Okay I’ll try that next time I work on the stage. I get a 20 from the DF and a 1 from DD so there should be some data points somewhere.

MatthieuDartiailh, 2021-05-19:
Indeed

chr0203, 2021-05-26:
I tried the reading a single byte read_bytes(1) and got the same timeout error... I am a bit lost as what to try next..

MatthieuDartiailh, 2021-05-26:
I am running out of ideas too. It looks like the instrument is just not answering. Pyvisa did not succeed in reading anything which is why the buffer is empty. In some cases you can get a timeout because you could not find the termination character but still read something which is why we log the already received bits.

The fact that all the other order work and produced the expected response, but not this one is really confusing. However, giving a second look at your code maybe try reading the results before disabling acquisition (DE0). This would be more in line with the manual example and it is possible the acquired data are wiped when disabling acquisition.

chr0203, 2021-05-26:
Okay. I will keep looking for a solution. I tried moving the lines below the DE0 command like this: 

(The previous lines are as before) 
...
values11 = inst.write('DF')
time.sleep(2)
values12 = inst.read()
print(values12)
time.sleep(2)
values13 = inst.write('DG')
time.sleep(20)
values14 = inst.read_bytes(1)
time.sleep(2)
values10 = inst.write('DE0')
print(values14) 

I got the same old timeout error however. I have also written to Newports tech support in hope of getting some answers. Especially the sentence "N N N N N N N N | Collected data in binary format (use an adapted software to get data)" was a bit intriguing to me. I asked them to give me an example of such an adapted software. Hopefully I will get some insights from that.

LongnoseRob, 2021-05-26:
As this is is a TCP connection, why not try to capture the traffic with tcpdump or wiresharkl?
You could also try the same commands manually by telnet to the unit and see what the responses are.

Another option might be to try the RS232 or USB connection.

chr0203, 2021-06-28:
@LongnoseRob Thank you for the reply. Ill try to see if I can catch the traffic with tcpdump or wiresharkl (once I have figured out how they work as I am still fairly new to this type of instrument communication). And as you mention, a R232 or USB might also be a good option! 

Cheers, 

Christoffer

Antyos, 2022-11-18:
Any updates on this?

I'm running into a similar issue running an ESP302 on a VISA through a TCP/IP socket, but in LabView. I am finding that if I send a lot of commands through the VISA in rapid succession, an error will eventually occur when attempting a query. The error does not appear at the same time, but I can consistently reproduce it.

I double checked all the termination characters, and they seem to be fine. I thought the issue might be commands getting mixed together, so I checked in Wireshark, but each command is sent as its own packet and the data all seems to look correct.

When the error occurs, I have noticed that there an error shows up on the ESP302 itself, saying `COMMAND NOT ALLOWED`. I don't think there is a way for it to show what command it is receiving on its end, but it would potentially help.

Is it possible that there is an internal buffer in the ESP302 that is being exceeded? I have no clue what the issue might be aside from that.

My hunch is that this is an issue with the ESP302, not Pyvisa, especially as I am getting the same issue with a VISA in LabView, but I could be wrong.