pyvisa/pyvisa #603: OSError: Could not open VISA library: function 'viOpen' not found

kienanmooney, 2021-05-21:
I'm trying to just execute example programs for pyvisa provided by Keysight. I've installed the Keysight VISA via the Keysight IO Suite installer, but it seems the backend is broken and it can't find my VISA installation. When executing the FindDevices.py program (link: https://www.keysight.com/us/en/lib/software-detail/programming-examples/io-libraries-example-programs-2798637.html) I receive the following output: 

C:\Python\Scripts\thirdpole>python FindDevices.py
C:\Python\lib\site-packages\visa.py:13: FutureWarning: The visa module provided by PyVISA is being deprecated. You can replace `import visa` by `import pyvisa as visa` to achieve the same effect.

The reason for the deprecation is the possible conflict with the visa package provided by the https://github.com/visa-sdk/visa-python which can result in hard to debug situations.
  warnings.warn(
Find all devices and interfaces:

Traceback (most recent call last):
  File "FindDevices.py", line 31, in <module>
    find('?*')
  File "FindDevices.py", line 16, in find
    resourceManager = visa.ResourceManager()
  File "C:\Python\lib\site-packages\pyvisa\highlevel.py", line 3015, in __new__
    visa_library = open_visa_library(visa_library)
  File "C:\Python\lib\site-packages\pyvisa\highlevel.py", line 2929, in open_visa_library
    return cls(argument)
  File "C:\Python\lib\site-packages\pyvisa\highlevel.py", line 175, in __new__
    raise OSError("Could not open VISA library:\n" + "\n".join(errs))
OSError: Could not open VISA library:
function 'viOpen' not found

I've tried following the directions from pyvisa regarding configuring the IVI backend (https://pyvisa.readthedocs.io/en/latest/introduction/configuring.html#configuring-the-ivi-backend) but none of the provided instructions solve the problem.

MatthieuDartiailh, 2021-05-22:
Since you installed Keysight IO libraries, I would suggest you first check your installation is functional by using Keysight Connection Expert. If this works, you are likely seeing the issue describe in this entry of the FAQ. Can you confirm you did try the fix suggested ?

kienanmooney, 2021-05-24:
The installation is functional and the instrument is recognized with the connection expert. I confirmed I could control the device remotely using the interface provided in connection expert. I did try the fix suggested (manually pointing the resourcemanager call to the dll location) and was not successful.

MatthieuDartiailh, 2021-05-24:
This is not exactly what the fix suggest (also sorry I forgot to copy the FAQ entry: https://pyvisa.readthedocs.io/en/latest/faq/faq.html#oserror-could-not-open-visa-library-function-viopen-not-found). You need to add the folder in which the dll lives using `os.add_dll_directory` (Typically C:\Program Files\Keysight\IO Libraries Suite), equivalently using dll_extra_paths in .pyvisarc.

kienanmooney, 2021-05-24:
I did try the dll_extra_paths addition in .pyvisarc - it produced a different error. I'll have to try again and post the output - perhaps I pointed to an incorrect directory. 

how can I be sure I'm pointing to the correct dll location?