pyvisa/pyvisa #604: Installing the RSVisa backend (from Rohde&Schwarz)

dklein9500, 2021-05-27:
Hello,

I need to use the RsVISA backend instead of the pyvisa backend. For this reason I've installed it from [their website]( https://www.rohde-schwarz.com/de/applikationen/r-s-visa-application-note_56280-148812.html).  
My OS is Ubuntu 20.04.2 LTS.
In the Pyvisa Doc it says that I need to add the path to the "library" when using the command `rm = ResourceManager('Path to library')` . 

How does this library look like? With the installation I've found a folder called "RsVisa", with a few .dll in it. However, none is called anything close to visa either.

Does anyone know what or where I need to look for when adding the backend? 

Thank you in advance!

MatthieuDartiailh, 2021-05-27:
If you have no other VISA library installed than RS VISA, by default pyvisa will pick it. From pyvisa point of view all IVI conformant implementations of VISA can be used with the IVI backend. To see what backend (ivi, py, sim, etc) and dll are detected by pyvisa you can simply run `pyvisa-info`.

If you have suggestion to improve the docs, they are welcome.

dklein9500, 2021-06-10:
Sorry for the late reply. 

When I type in pyvisa-info, I receive this result: 

Machine Details:
   Platform ID:    Linux-5.8.0-55-generic-x86_64-with-Ubuntu-20.04-focal
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python2
   Version:        2.7.18
   Compiler:       GCC 9.3.0
   Bits:           64bit
   Build:          Mar  8 2021 13:02:45 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      Binary library: Not found

I use Python 3.8.5 though, don't know why it states Python 2.7

Anyway, the reason for me needing to use another backend is that I get timeout errors with my currently selected backend when receiving data from the oscilloscope RTO 2044 from R&S. 

I already wrote with their support, and they told me that it should work if I use their visa backend.

MatthieuDartiailh, 2021-06-10:
The fact that it shows Python 2.7 may indicate you did not install pyvisa with the interpreter you think. First check if pyvisa appears in `pip3 list`. If it does not then install it, otherwise I would still suggest you update pyvisa to the latest release.

Somehow I missed that you were on Ubuntu. Third party libraries sometimes install under opt/ try to see if RS did that and then look for a visa.so (or something similar), you will have to indicate the path to this file to the ResourceManager.

Out of curiosity which hardware connection are you using (USB, TCPIP, ...) ? for TCPIP pyvisa-py should mostly do a reasonable work but be sure you properly configure the termination charaters (both read and write)

dklein9500, 2021-06-10:
I receive
PyVISA                 1.10.1              
PyVISA-py              0.3.1 
, when typing pip3 list. So I guess pyvisa is properly installed. The question now is, how do I find the backend that the current interpreter is using? 

The opt/ folder is sadly empty. When searching the folders for "visa.so" I found "/usr/lib/librsvisa.so" instead, which seems to be the correct file. However, when I use `osc_res = pyvisa.ResourceManager('/usr/lib/librsvisa.so')`, I receive the error message `pyvisa.errors.VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.`

To come back to the connection type, I use the TCPIP connection and most tasks do work fine. I just cannot transfer the data. 
I also have a similar oscilloscope from R&S, the RTM 3004, which uses very similar / identical SCPI commands. With the RTM everything works fine, including the data transfer.

I am not sure to what termination character I would need to configure it to.

MatthieuDartiailh, 2021-06-10:
The error message suggest that PyVISA fails to create the default resource manager which is weird can you post the full traceback ? 

Regarding pyvisa-py, do you encounter the issue with binary transfer ? If yes I would suggest you give a look at https://pyvisa.readthedocs.io/en/latest/introduction/rvalues.html#reading-binary-values if you have not already done so. In particular the right value for expect_termination may be different between your 2 instruments.

Miloslav-RS, 2022-09-06:
@dklein9500, I see the discussion took place some time ago, is your problem with RTM data transfer still remaining?

Cheers from Rohde & Schwarz,
Milo

GuidoKiener, 2023-03-06:
Here is an example how you can use the R&S VISA with pyvisa:

```
#!/usr/bin/env python3
import pyvisa as visa

# Open VISA Resource-Manager
rm = visa.ResourceManager("/usr/lib/librsvisa.so@ni") #use this for Ubuntu
#rm = visa.ResourceManager("/usr/lib64/librsvisa.so@ni") #use this for CentOS
#rm = visa.ResourceManager() #use this for Windows
print(rm)

# Show available resources
list = rm.list_resources()
print(list)

dev = rm.open_resource('TCPIP::example.com::hislip0')
#dev.write_termination = '\n' #use this for raw socket connections
#dev.read_termination = '\n'  #e.g. TCPIP::example.com::5025::SOCKET
dev.write("*IDN?")
idn = dev.read()
print("IDN:", idn)
dev.close()
```
Newer pyvisa requires the backend name @ivi.
```
rm = visa.ResourceManager("/usr/lib/librsvisa.so@ivi")
```
I hope this helps.