pyvisa/pyvisa #605: Query_binary_values - Fetch data from multiple channels separated by comma. Keysight N6705B

ThomasGund, 2021-06-03:
Hi,
I am working on a N6705B Power analyzer from Keysight. I am logging two channels and are using the command <:FETCh:ELOG? 10000,(@1,2)> to get data. I am asking for max 10 000 data points and data from Ch1 and Ch2. The N6705B will return data from both channels separated by a comma. The values are 32-bit floats. I have also used a USB analyzer from Ellisys to see the actual traffic from the instrument.

This is what I have tried to fetch data:

```Python
sData = keysight_N6705B.query_binary_values(":FETCh:ELOG? 10000,(@1,2)", datatype = 'B', container = bytearray)
print('sData', sData)
```
```
USB traffic logged by USB analyzer, hex:
23 31 34 B3 80 98 A7 2C 23 31 34 B3 15 75 56 0A
ASCII decoded:
#14....,#14..uV.

Result of print('sData', sData):
bytearray(b'\xb3\x80\x98\xa7')
```
From the decoded USB traffic, we can see that we have a start character #. 1 says that we have one byte which are telling the payload length/size, the 4 says that the payload contains four bytes of data. Then we have a comma and then we have the same structure for channel 2. 

From the print('sData', sData) I get:
bytearray(b'\xb3\x80\x98\xa7')

Only the four bytes for Ch1 is read.

My question: is it possible to use query_binary_values to read multiple channels separated by comma like I am getting from N6705B. If yes, how?
Or is the only (or easiest) solution to use:
``` Python
keysight_N6705B.write(':FETCh:ELOG? 10000,(@1,2)')
sData = keysight_N6705B.read_raw()
```
and then manually decode sData. Or any other suggestions how to read the data from N6705B by using :FETCh:ELOG? 10000,(@1,2)?

MatthieuDartiailh, 2021-06-03:
There is no built in way to achieve what you want no (this situation is quite rare). However you can use the following (modulo any typo since I cannot test):

```python
keysight_N6705B.write(':FETCh:ELOG? 10000,(@1,2)')
# This can stop early if the data contain the termination character
block = keysight_N6705B.read_raw()

offset, data_length = pyvisa.util.parse_ieee_block_header(block)
expected_length = offset + data_length + 1  # for the comma
# read the rest  of the first channel if any
if (leftover := (expected_length - len(block)) > 0):
    block.extend(self.instr.read_bytes(leftover, chunk_size=chunk_size))
# You did not specify the endianness so adapt as relevant
ch1_data = data = pyvisa.util.from_binary_block(block, offset, datatype="f", is_big_endian=False, container=np.ndarray)

# We read less than the first channel to start with and just completed the data so read more
if leftover >= 0:
    block = keysight_N6705B.read_raw()
else:
    # Otherwise discard the part we already dealt with
    block = block[expected_length:]

# Proceed as with channel 1

offset, data_length = pyvisa.util.parse_ieee_block_header(block)
expected_length = offset + data_length + len(self.read_termination)  # Since your instrument appear to include it at the end of the message.
# read the rest  of the first channel if any
if (leftover := (expected_length - len(block)) > 0):
    block.extend(self.instr.read_bytes(leftover, chunk_size=chunk_size))
# You did not specify the endianness so adapt as relevant
ch2_data = data = pyvisa.util.from_binary_block(block, offset, datatype="f", is_big_endian=False, container=np.ndarray)

```

Hope this helps.

ThomasGund, 2021-06-03:
@MatthieuDartiailh 
Yes, it was very helpful. It definitely saved me a lot of time. I haven't fully understood how it works yet, but I am working on it. I had to do some minor corrections to get it working, but there are still some issues. So, I have some basic questions.  

**read_raw()**
your comment: This can stop early if the data contain the termination character

Do I understand it correctly that this command reads data until it sees a termination character? And the stream of data can contain the termination character even though it is not the end of the data; this is the reason why you have added the logic for reading leftover data? What if there are no termination character, will it then read until timeout? You are right about my instrument using a termination character when returning values, 0x0A (LF). but when I ask for the read_termination I get None: 

``` Python
print('keysight_N6705B.read_termination:', keysight_N6705B.read_termination) 
>keysight_N6705B.read_termination: None
``` 

This caused a small problem with len(self.read_termination) since None has no length. I changed it to +1, I don’t know whether that is good enough.

**leftover:**
``` Python
if (leftover := (expected_length - len(block)) > 0): 
```
here leftover will be of type bool. This caused a problem with:
``` Python
if leftover >= 0:
    block = keysight_N6705B.read_raw()
```
This will always be True. I assume that you want leftover to be an int.
For channel 2 I have added extra parentheses: 
``` Python
if ((leftover := (expected_length - len(block))) > 0): 
``` 
This is probably what you wanted, right?

**Ch1_data:**
This is maybe the main issue with the code. 
For:
``` Python
ch1_data = data = pyvisa.util.from_binary_block(block, offset, datatype="f", is_big_endian=True, container=np.ndarray)
```
the output is: 
``` 
[-5.6187584e-08 -5.5812784e-08  2.3191015e-12 -2.8244930e-08
 -2.8630749e-08]
``` 
It has read the whole block of data and converted it to floats, not just ch1. I assume that is because block is used as an argument for the number of bytes to read and block is 24 bytes. I have tried to create a dummy parameter block_1 and used that instead:
``` Python
block_1 = block[:expected_length]
ch1_data = data = pyvisa.util.from_binary_block(block_1, offset, datatype="f", is_big_endian=True, container=np.ndarray)
```
the output is: 
``` 
[-5.6187584e-08 -5.5812784e-08]
``` 
Is this a solution that can be used and/or are there any pitfalls I should be aware of?

**Self and chunk_size:**
``` Python
if (leftover := (expected_length - len(block)) > 0):        # leftover is bool.
    block.extend(self.instr.read_bytes(leftover, chunk_size=chunk_size)
``` 
should self in self.instr.read_bytes() be replaced by keysight_N6705B? and where do I get chunk_size from?

This is the code I have tried. I have added a lot of print() in order to understand what is happening
``` Python
print('keysight_N6705B.read_termination:', keysight_N6705B.read_termination)
keysight_N6705B.write(':FETCh:ELOG? 10000,(@1,2)')
# This can stop early if the data contain the termination character
block = keysight_N6705B.read_raw()
print('block:', block)
print('len(block):', len(block))

offset, data_length = pyvisa.util.parse_ieee_block_header(block)
[... 53 lines cut ...]
block: b'#18\xb3qR\xe7\xb3o\xb6\xce,#18\xb2\xf2\x9fB\xb2\xf5\xef\xaf\n'
len(block): 24
offset: 3
data_length: 8
expected length: 12
type(leftover): <class 'bool'>
leftover: False
block_1: b'#18\xb3qR\xe7\xb3o\xb6\xce,'
len(block_1): 12

Channel 1
ch1_data using block_1: [-5.6187584e-08 -5.5812784e-08]
ch1_data using block: [-5.6187584e-08 -5.5812784e-08  2.3191015e-12 -2.8244930e-08
 -2.8630749e-08]

Channel 2
block: b'#18\xb2\xf2\x9fB\xb2\xf5\xef\xaf\n'
len(block): 12
offset: 3
data_length: 8
expected length: 12
type(leftover): <class 'int'>
leftover: 0
ch2_data: [-2.824493e-08 -2.863075e-08]
```

MatthieuDartiailh, 2021-06-03:
I will try to answer in order but it looks like you caught all my mistakes.

**read_raw**
`read_termination` can be None indeed if you are not relying on it for identifying the end of a message (but use some low level hardware signal, or protocol specific signal). It may mean that you do not need the safeguards I put.
 
**leftover**
I am still getting used to this syntax and I did get it wrong indeed and your fix is correct

**ch1_data**
I forgot but you can pass `data_length` to `from_binary_block` if which case you won't go over read the other bytes:
`pyvisa.util.from_binary_block(block_1, offset, data_length, datatype="f", is_big_endian=True, container=np.ndarray)`

**self and chunk_size**
This is bad copy paste indeed. self should be your instrument, and you can just ignore chunk_size.

Happy my not quite right example helped. Let me know if you have any other questions.