pyvisa/pyvisa #606: GPIB write timeout

msmttchr, 2021-06-08:
I have the following scenario in my test:

- I am using an RF signal generator connected via GPIO or TCPIP
- I am loading a waveform that at the end of the day is a fairly big file loaded using a dedicated SCPI command (using binary mode).
During the loading of data, you could have a timeout depending on the length of the data transfer.

At the moment, I manage this situation by simply incrementing the timeout to something like 20 seconds, to make sure I can complete the transfer.

It is also difficult to estimate an accurate value for such a timeout.

I wonder if there is a possibility to avoid such a timeout, when doing a "write" transfer to the instrument and keep the timeout for the instrument "responses". I am not very familiar with GPIB protocol, but at the end of the day when I write to instrument, I think I am in charge of the operation so I am not sure what is the purpose of the timeout in that context.
Is it possible that the "write" transaction hangs indefinitely due to some flow control or any other issue ?

Do you have any suggestion on how to deal with this scenario ?

MatthieuDartiailh, 2021-06-08:
I do not remember exactly but I believe that yes a write can hang. If you do not want to fiddle with values you can set timeout to None which basically disables timeout checks. So something like the following would work:

```python
old = rsc.timeout
rsc.timeout = None
try:
    rsc.write_binary_values(...)
finally:
    rsc.timeout = old
```

We could add a timeout context manager to pyvisa to make this easier, feel free to open a PR if you feel so inclined.

msmttchr, 2021-06-08:
Thanks for the fast reply. I think your solution is a good compromise. I am not in the position to make a PR with my level of knowledge, but if I manage to get more involved, I will try. BTW, thank you very much for maintaining pyvisa, it is an immensely helpful package.