pyvisa/pyvisa #607: [COM] Communication issue with Keysight DSO1052B using Raspberry Pi, python3.  Timeout errors.

fallingin2infinity, 2021-06-09:
Hello,

I've been using pyvisa for a while, on my Windows machine.  Works like a treat - really impressed! :)

However, I'd like to use it on my Raspberry Pi, but I'm running into some timeout errors.

My code is:
------------------

`import pyvisa

rm = pyvisa.ResourceManager()
inst = rm.open_resource('ASRL/dev/ttyAMA0::INSTR')

inst.read_termination = '\n'
inst.write_termination = '\n'

inst.query('*IDN?')`

This exact code works in Windows, but on the Pi, I get:

"pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed."

I get the same error if I use inst.write(*IDN?) - as recommended in the pyvisa documentation.  I also tried extending the timeout, to no avail.

I'm not quite sure what to look for in terms of the configuration differences between my pi and windows machine.  I know that I don't have NI-VISA installed, because it's not compatible with the Pi OS.

I'd be extremely grateful for any help that can be offered!

Instrument details
------------------
* Model: Keysight DSO1052B
* Communication: TCPIP, GPIB, ....
* Link to the documentation (if available):  None available.

Output of `pyvisa-info`
-----------------------
Machine Details:
   Platform ID:    Linux-5.4.79-v7l+-armv7l-with-debian-10.6
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.7.3
   Compiler:       GCC 8.3.0
   Bits:           32bit
   Build:          Jul 25 2020 13:03:44 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.2
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'

MatthieuDartiailh, 2021-06-09:
There may be some discrepancies between the defaults serial parameters in NI-VISA and pyvisa-py causing the issue. Could you run the following on both your systems ?
```python
import pyvisa

rm = pyvisa.ResourceManager()
inst = rm.open_resource('ASRL/dev/ttyAMA0::INSTR')

print(inst.baud_rate)
print(instr.data_bits)
print(instr.parity)
print(instr.stop_bits)
print(instr.flow_control)
```

fallingin2infinity, 2021-06-09:
Hi Matthieu,

Thanks for your reply.  Running that on my Windows system results in:

"AttributeError: 'USBInstrument' object has no attribute 'baud_rate'", for each of the print lines.

On the Pi:

baud:  9600
Data:  8
Parity: none
stop:   one
flow:   none

I can manually define the parameters on the Windows machine, and then the above commands will read them back.  However, it seems I can change them to whatever I like, and pyvisa still talks to the device.

fallingin2infinity, 2021-06-09:
> n

MatthieuDartiailh, 2021-06-09:
Wait your device is seen as USB on Windows ? In that case it means it is a USBTMC device and you should not try to access it as a serial resource from linux. USBTMC is a different protocol. Does pyvisa list any USB device on your Pi ? If not try to have a look at https://pyvisa.readthedocs.io/projects/pyvisa-py/en/latest/installation.html#usb-resources-usb-instr-raw since you may need to edit udev rules.

fallingin2infinity, 2021-06-09:
Thanks for this... so yes, it turns out that there is no USB device showing on the Pi.  I'm working through the link you sent, plus python-usbtmc, but haven't had any luck yet.  Nevertheless, it looks likely that my issue is not related to pyvisa.